Ysurac/openmptcprouter #836: NAT 1:1 or DMZ host

lirees, 2020-02-13:
## Expected Behavior

It's possible to redirect all the traffic form the public IP of the vps ( OVH ) to a lan ip of OMR ? like a nat 1:1 or DMZ host 

i have tried to flag the option "Redirects all ports from server to this router" but don't works so i must redirect all the ports that i need to the firewall rules section 

## Specifications

  - OpenMPTCProuter version: (0.53.3)
  - OpenMPTCProuter VPS version: (0.1008)
  - OpenMPTCProuter platform: (raspberry pi 3)

dabuide, 2020-02-13:
Hello, if you want to do NAT, and access your equipment from the web, you must activate "Redirects all ports from server to this router" in the MPTCP options, then
`Network > Firewall > Port Forwards > New with source zone "VPN" / Protocol (TCP or UDP) / internal zone "LAN" / IP server / and port you want to open`
Hoping to have helped you

lirees, 2020-02-17:
Thanks for you reply

i have does some test,  if i create the firewall rules without activate the "Redirects all ports from server to this router" the rules still work ( also after reboot ) it's like that the flag "Redirects all ports from server to this router" as no effect 

this is not a problem for me, but i would like manage ports forwarding from my firewall and i would redirect all the traffic form IP of the VPS ( all port for any protocol ) to internal ip... simply nat 1:1 or dmz host as you prefer, without do it for a every single rules 

Thanks

Ysurac, 2020-02-17:
Redirect all ports from server to this router, really redirect all ports from 0 to 64999 to the router. You can redirect all ports in the same way to your firewall, but it's another NAT.
Else only ports defined with VPN as source are redirected.
For the VPS, all ports redirection are in ```/etc/shorewall/rules```.

closed 2020-09-14 by github-actions[bot]