Ysurac/openmptcprouter #837: key exchange failed

CoSysopK, 2020-02-14:
## Actual Behavior

From time to time, I have a connection problem.

**Log of router:**
Fri Feb 14 02:00:00 2020 cron.err crond[3198]: USER root pid 30903 cmd /etc/init.d/vnstat_backup backup
Fri Feb 14 02:00:00 2020 cron.err crond[3198]: USER root pid 30904 cmd /etc/init.d/omr-bypass bypass_asn
Fri Feb 14 02:01:11 2020 daemon.err glorytun[5778]: read: Operation timed out
Fri Feb 14 02:01:11 2020 daemon.info glorytun[5778]: STOPPED tun0
Fri Feb 14 02:01:12 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: connected
Fri Feb 14 02:01:20 2020 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.101.1 dev eth1
Fri Feb 14 02:01:20 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Fri Feb 14 02:01:21 2020 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Fri Feb 14 02:01:21 2020 user.notice Shadowsocks: Rules DOWN
Fri Feb 14 02:01:21 2020 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Fri Feb 14 02:01:22 2020 user.notice Shadowsocks: Rules UP
Fri Feb 14 02:01:22 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.102.1 dev eth2
Fri Feb 14 02:01:22 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: key exchange failed
Fri Feb 14 02:01:22 2020 daemon.info glorytun[5778]: STOPPED tun0
Fri Feb 14 02:01:23 2020 user.notice Shadowsocks: Reload omr-bypass rules
Fri Feb 14 02:01:23 2020 user.notice omr-bypass: Starting OMR-ByPass...
Fri Feb 14 02:01:23 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: connected
Fri Feb 14 02:01:25 2020 user.notice omr-bypass: Reload dnsmasq...
Fri Feb 14 02:01:26 2020 daemon.info dnsmasq[22413]: read /etc/hosts - 4 addresses
Fri Feb 14 02:01:26 2020 daemon.info dnsmasq[22413]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Fri Feb 14 02:01:26 2020 daemon.info dnsmasq-dhcp[22413]: read /etc/ethers - 0 addresses
Fri Feb 14 02:01:26 2020 user.notice omr-bypass: OMR-ByPass is running
Fri Feb 14 02:01:33 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: key exchange failed
Fri Feb 14 02:01:33 2020 daemon.info glorytun[5778]: STOPPED tun0
Fri Feb 14 02:01:34 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: connected
Fri Feb 14 02:01:44 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: key exchange failed
Fri Feb 14 02:01:44 2020 daemon.info glorytun[5778]: STOPPED tun0
Fri Feb 14 02:01:45 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: connected
Fri Feb 14 02:01:55 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: key exchange failed
Fri Feb 14 02:01:55 2020 daemon.info glorytun[5778]: STOPPED tun0
Fri Feb 14 02:01:56 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: connected
Fri Feb 14 02:02:06 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: key exchange failed
Fri Feb 14 02:02:06 2020 daemon.info glorytun[5778]: STOPPED tun0
Fri Feb 14 02:02:07 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: connected
Fri Feb 14 02:02:17 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: key exchange failed
Fri Feb 14 02:02:17 2020 daemon.info glorytun[5778]: STOPPED tun0
Fri Feb 14 02:02:18 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: connected
Fri Feb 14 02:02:28 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: key exchange failed
Fri Feb 14 02:02:28 2020 daemon.info glorytun[5778]: STOPPED tun0
Fri Feb 14 02:02:29 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: connected
Fri Feb 14 02:02:39 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: key exchange failed
Fri Feb 14 02:02:39 2020 daemon.info glorytun[5778]: STOPPED tun0
Fri Feb 14 02:02:40 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: connected
Fri Feb 14 02:02:50 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: key exchange failed
Fri Feb 14 02:02:50 2020 daemon.info glorytun[5778]: STOPPED tun0
Fri Feb 14 02:03:00 2020 daemon.err glorytun[5778]: 34.77.160.206.65001: connected
Fri Feb 14 02:03:06 2020 daemon.info glorytun[5778]: STARTED tun0
Fri Feb 14 02:03:15 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Fri Feb 14 02:03:15 2020 user.notice post-tracking-post-tracking: Tunnel up disable use of ShadowSocks for UDP
Fri Feb 14 02:03:15 2020 user.notice Shadowsocks: Rules DOWN
Fri Feb 14 02:03:15 2020 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Fri Feb 14 02:03:15 2020 user.notice Shadowsocks: Rules UP
Fri Feb 14 02:03:16 2020 user.notice Shadowsocks: Reload omr-bypass rules
Fri Feb 14 02:03:16 2020 user.notice omr-bypass: Starting OMR-ByPass...
Fri Feb 14 02:03:18 2020 user.notice omr-bypass: Reload dnsmasq...
Fri Feb 14 02:03:19 2020 daemon.info dnsmasq[22413]: read /etc/hosts - 4 addresses
Fri Feb 14 02:03:19 2020 daemon.info dnsmasq[22413]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Fri Feb 14 02:03:19 2020 daemon.info dnsmasq-dhcp[22413]: read /etc/ethers - 0 addresses
Fri Feb 14 02:03:19 2020 user.notice omr-bypass: OMR-ByPass is running
Fri Feb 14 02:03:50 2020 daemon.err /usr/bin/ss-local[15973]: getpeername: Socket not connected
Fri Feb 14 02:04:34 2020 daemon.err uhttpd[3535]: luci: accepted login on / for root from 192.168.100.228

**Log of VPS:** (gmt time)
Feb 14 00:58:29 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb 14 00:58:54 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
[... 102 lines cut ...]
Feb 14 01:06:52 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb 14 01:07:23 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb 14 01:07:44 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb 14 01:07:44 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb 14 01:07:49 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb 14 01:07:52 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb 14 01:07:55 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb 14 01:08:16 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb 14 01:08:16 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb 14 01:08:24 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb 14 01:08:27 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected

## Steps to Reproduce the Problem

1. Installer l'image pour Raspberry PI 3 / 3B + (64 bits)
2. Installateur VPS Debian 10 Buster x86_64
3. Configurez le routeur avec le wan1 statique et le wan2 statique
4. Ajoutez l'adresse IP VPS et, toutes les informations récupérées sur le VPS
5. Deux clés 4G sur USB PI 3B

## Specifications

- Version OpenMPTCProuter: version 0.53.3
- Version VPS OpenMPTCProuter: Version 0.1008
- Plateforme OpenMPTCProuter: RPI3B + 64 bits

Ysurac, 2020-02-14:
When the interface defined as master is down the tunnel go down. I think the interface also changed IP, so glorytun TCP doesn't known it and doesn't allow it before some time. You can try another VPN.

CoSysopK, 2020-02-14:
So, my WAN1 and WAN2 connections must be with a static IP.
4G keys are with dynamic IP. But the IP does not change if the 4G key has not been OFF/ON.

**Better to use dsvpn or openvpn so that this error does not happen again ?**

For information, there was an update on the google cloud VPS server. I did it.

CoSysopK, 2020-02-15:
When there are WANs with dynamic IP. Some options may need to be added to address this IP change concern. With **DynDNS** or **No-IP**. This will allow you to have a static IP and thus not suffer any untimely disconnection.

Ysurac, 2020-02-16:
DynDNS or No-IP doesn't give a static IP, but a domain or subdomain.
You can use another VPN.

CoSysopK, 2020-02-16:
I have my WAN1 which changes IP by itself and without rebooting. All VPN are causing me an error.
I tried: Glorytun tcp et udp, dsvpn, openvpn.

closed 2020-02-19 by CoSysopK