Ysurac/openmptcprouter #839: Ubuntu server 18.04 is not working any more as VPS platform

jfh75, 2020-02-17:
## Expected Behavior
OMR should be able to connect to VPS and upload keys with ubuntu server 18.04

## Actual Behavior
Both with upgrade and new install from formatted HDD, load of "half configured" libraries that lead to CANT GET TOKEN, UNABLE TO CONNECT
Ubuntu 18.04 should be dropped from documentation, it is not working any longer.
Debian buster 10.3 works fine

## Steps to Reproduce the Problem

  1.upgrade VPS with ubuntu 18.04 or new install
  2.run the update script
 
## Specifications

  - OpenMPTCProuter version: 53.3
  - OpenMPTCProuter VPS version: 1008
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-02-17:
It's fixed in snapshot script.

closed 2020-02-17 by Ysurac

jfh75, 2020-02-17:
Hello, thank you for you answer, what is snapshot script please ? 
Last try was last week on ubuntu server 18.04
Best regards
JFH

Ysurac, 2020-02-17:
Snapshot script is the script that is in dev: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots
But I don't make any support on this, and this can fail sometimes, it's in dev.

jfh75, 2020-02-18:
Thank you for you detailed answer :=) !
Now I have managed to install debian, I think I will stick to a supported platform. Keeping in mind there is always this unsupported solution in case....
JFH