Ysurac/openmptcprouter #840: Which server key

jpadie, 2020-02-17:
Using the latest commit and a pi3B+
Which server key generated by the VPS script do I put into the router wizard interface?  There seem to be a couple of possibilities.

And what is the other server key used for?

Ysurac, 2020-02-17:
There is only one OpenMPTCProuter server key.
Keys are also in ```/root/openmptcprouter_config.txt```

Other keys are for VPNs, they are retrieved via the API.

closed 2020-02-17 by Ysurac

jpadie, 2020-02-17:
there is a server key and an admin api server key.  

Could you explain the purpose of each please? or point to the documentation where the purpose is described?

`
Your OpenMPTCProuter ADMIN API Server key: 607061AA7C4C5BBB2A9EB12EC497D699E6512F6DBF5BF4***************F8009F

Your OpenMPTCProuter Server key: 824A383882552DD09ABD259D617899A642F9C76F3D72BF*******846C0BA2

Your OpenMPTCProuter Server username: openmptcprouter
`

Ysurac, 2020-02-17:
As you can see, there is only one named "OpenMPTCProuter Server key".
Admin API server key is used to make changes to the server API directly, you don't need this.