Ysurac/openmptcprouter #841: Aggregation issue  / VPS <-> Router does not connect [WRT3200ACM]

rubaz100, 2020-02-17:
## Expected Behavior

  Aggregation works fine.
Portforwarding without issues
OMR Bypass without issues.

## Actual Behavior
[image]
Log:
http://pastebin.com/4FgVdeUq

## Steps to Reproduce the Problem

  1. I use GloryTun UDP and I reset (format) my VPS.
  2. How can I enable MLVPN? Btw.
  3.

## Specifications

  - OpenMPTCProuter version: (v0.53.3)
  - OpenMPTCProuter VPS version: (v0.1008)
  - OpenMPTCProuter platform: (WRT3200)

Ysurac, 2020-02-18:
I see dsvpn in the log, how do you set the VPN ? via the wizard ?
MLVPN is not available in current release.

rubaz100, 2020-02-18:
Yes via the wizard. 

Advanced Settings and then changed the GloryTun TCP to GloryTun UDP

I tried dsvpn to but that did not work too.

rubaz100, 2020-02-18:
And iPhone  / iPad users can not use the internet. They are only connected with ipv6 (I disabled it already) I don't know why.

Ysurac, 2020-02-18:
I need your configuration in System->OpenMPTCProuter and "show all settings" tab. Remove keys before paste.

rubaz100, 2020-02-18:
here is the log

https://pastebin.com/G711ELnQ

Ysurac, 2020-02-19:
What is your VPS ? Ports are open on the VPS and it answer to ping ?

rubaz100, 2020-02-19:
Ports are open. Ping is ok.
 I installed Debian 10 fresh. 
Problem is not solved ;(

Ysurac, 2020-02-19:
What is the VPS provider ?
The 0 latency seems strange to me.

rubaz100, 2020-02-19:
Yeah that's strange too. It worked 10 days ago

The provider is fastpipe.io

Ysurac, 2020-02-19:
In your config I see this line: "shadowsocks-libev.sss0.server='8'" the "8" should be your server IP. It's an error in the paste or it's really that ?

rubaz100, 2020-02-19:
No I just erased the IP addresses. The 8 was only the end.

rubaz100, 2020-02-24:
So I think I have a problem with GloryTun
dsvpn works but it's too slow for me.

closed 2020-08-12 by Ysurac