Ysurac/openmptcprouter #843: OMR Bypass issues

darthclide, 2020-02-18:
## Expected Behavior
Netflix and my game Warframe should bypass the router completely, and use only my ethernet line (DSL) to connect

## Actual Behavior
Netflix somehow detects the bonded server even after I set it in the protocol, as well as adding the domain netflix.com

Warframe I got 50% working by forwarding the ports for the game + adding warframe.com to the domain. However, IRC does not connect (even though I set IRC in the protocol to use my DSL because I also want IRC on twitch.tv to not use the bonded line). And in the logs, it shows warframe detecting my VPS server and it is blacklisted. How the heck can they see what my VPS server is, if I am bypassing it completely with my DSL?

## Steps to Reproduce the Problem

  1. Add netflix and warframe in OMR Bypass through various methods

## Specifications

  - OpenMPTCProuter version: v0.53.3 r0+11546-a74095c68c
  - OpenMPTCProuter VPS version: I assume v0.53.3 r0+11546-a74095c68c since my NoRIP tech guy set it up both on my machine as well as the server
  - OpenMPTCProuter platform: x86_64 but it could be x86

I will update you with the correct information on those last 2 points if you need it (Waiting for my friend to respond), but I hope you can help me figure this out in the mean time.

Jon

Ysurac, 2020-02-21:
It's possible that Netflix use a new domain that is not bypassed. I don't have a netflix account for now to test that.
For twitch.tv IRC how do you connect to it ? via an IRC client ?
For warframe, it may use others domains than only warframe.com

For the VPS part, I think that VPS script is used.

darthclide, 2020-02-21:
> It's possible that Netflix use a new domain that is not bypassed. I don't have a netflix account for now to test that.

So you are saying that your Netflix protocol will do no good until you update it? And that the only thing that will work for the time being is finding that new domain for netflix?

> For twitch.tv IRC how do you connect to it ? via an IRC client ?

It is connected with mIRC (no problem because I can bind it to an adapter) and Twitch (a problem because I am using the bonded connection to watch Youtube videos while I stream). So far, the bypass seems to work just fine for Twitch IRC.

> For warframe, it may use others domains than only warframe.com

Correct, but since the logs of my friend show IRC statements, I can't fathom why forwarding the IRC protocol works for Twitch, but not for Warframe?

> For the VPS part, I think that VPS script is used.

I am unsure what you mean? Are you saying there is no way to hide my VPS IP even when I use OMR Bypass?

Ysurac, 2020-02-22:
> > It's possible that Netflix use a new domain that is not bypassed. I don't have a netflix account for now to test that.
> 
> So you are saying that your Netflix protocol will do no good until you update it? And that the only thing that will work for the time being is finding that new domain for netflix?

Yes

> It is connected with mIRC (no problem because I can bind it to an adapter) and Twitch (a problem because I am using the bonded connection to watch Youtube videos while I stream). So far, the bypass [...]

So where is the problem with Twitch ?

> > For warframe, it may use others domains than only warframe.com
> 
> Correct, but since the logs of my friend show IRC statements, I can't fathom why forwarding the IRC protocol works for Twitch, but not for Warframe?

I don't use warframe, so I need more info. What is the warframe IRC server ?

> > For the VPS part, I think that VPS script is used.
> 
> I am unsure what you mean? Are you saying there is no way to hide my VPS IP even when I use OMR Bypass?

No, I say that version you write for the VPS part is not the good one.

darthclide, 2020-02-22:
There is no issue with Twitch. I just don't understand why the OMR bypass works for Twitch IRC, but not Warframe IRC since I have set the IRC protocol to my DSL. I do not know the IRC server for Warframe, but he logs do show IRC connections being made.

Do you think you will ever get into the habit of updating the Netflix protocol regularly? I need to know so I can just give up and hook my parent's TV up to a connection. Even though I would prefer not to... If the bonder doesn't realize a netflix movie is running, it might ask for more bandwidth than it can actually receive.

Is this the correct version you are looking for?  0.1006-test1 4.19.80-mptcp

Ysurac, 2020-02-22:
Warframe IRC is maybe not real IRC. I can't verify without the server.

For Netflix, I can check when I will be able to paid subscription.
I don't understand what you mean with the bonder and netflix ?
What is the error you get on Netflix ?

The VPS version is not on a stable version.

darthclide, 2020-02-22:
Sadly Warframe support will not respond to my request to lift the blacklist on my VPS IP. And they also will not tell me what the protocol or server is for their chat. I guess I will assume that even though the logs clearly attempt connecting to IRC, that it is something fake similar to what mixer.com uses

I mean that I want my parents to be able to use this bonded connection (I will eventually be setting this up on a pi) and I don't want them having to deal with Netflix not working because the OMR bypass fails. I have a subscription, feel free to tell me what tools I need to get the info.

I messaged the guy who setup the server. Maybe he knows more about my setup and what I need? And that is maybe why he didn't go for a more recent version?

Ysurac, 2020-02-22:
You only need to add the domain used by warframe. I don't know what log you are talking about.

To get the info, you need a browser like Firefox or Chrome and go to developers tools then in network tab and check what domains are used when you have a warning from Netflix.

Ysurac, 2020-02-22:
I don't see on list that I have that netflix use new domains, IPs. You are using OpenMPTCProuter as DNS ?

rubaz100, 2020-02-22:
Netflix is super easy to bypass

Whitelist that here:

"
btstatic.com
customerevents.netflix.com
movies.netflix.com
nccp.netflix.com
nflxext.com
nflximg.com
nflximg.net
nflxvideo.net
secure.netflix.com

For a more update list, please see: https://openconnect.netflix.com/mobiledeliverydomains.txt

"
And add the ASN

darthclide, 2020-02-22:
@Ysurac  I assume so? Since I currently have set my Vbox adapter's gateway to 192.168.100.1 (which is the IP for the router running on the VM).

@rubaz100 What should I put into ASN? I assumed all those URLs go into the domain section?

Ysurac, 2020-02-22:
This domains are already bypassed (only btstatic is not, but I don't think it's needed).
Gateway is not the same as DNS. DNS must be set to 192.168.100.1 (and not 8.8.8.8 or some others DNS), some TVs set some static DNS.

rubaz100, 2020-02-22:
"Finally, a reminder that if you are peering with Netflix and do IRR filtering, our prefix set is RS-
NETFLIX and our as-set is AS-NFLX. Please be sure to accept advertisements that originate
from ASNs: 2906, 40027, and 55095."

[deploymentguide.pdf](https://github.com/Ysurac/openmptcprouter/files/4239336/deploymentguide.pdf)
(Source: Netflix)

darthclide, 2020-02-23:
> This domains are already bypassed (only btstatic is not, but I don't think it's needed).
> Gateway is not the same as DNS. DNS must be set to 192.168.100.1 (and not 8.8.8.8 or some others DNS), some TVs set some static DNS.

Yes the DNS is set to 192.168.100.1 I am currently testing this on the Windows 10 Netflix App. Perhaps the UWP app has different domains to bypass or something?

darthclide, 2020-02-23:
> "Finally, a reminder that if you are peering with Netflix and do IRR filtering, our prefix set is RS-
> NETFLIX and our as-set is AS-NFLX. Please be sure to accept advertisements that originate
> from [...]

No idea how to modify the "prefix set" and "as-set". But I will go ahead and add those ASN numbers in and see what happens.

darthclide, 2020-02-23:
And adding those ASN numbers did no good. But just to see if omr bypass is working correctly for another big name domain I tried adding the Youtube protocol and sending it to my AT&T connection. It did not work. The youtube video continued to use my bonded line to load the video. I am really confused on why some of the bypass is working (such as just logging into the Warframe game but not chat) but then others it completely fails even on something as simple as Youtube?

I am unsure if this is related or if I need to make a new thread: My bonded connection (DSL + AT&T + Verizon) seems to work okay, but if I open up about 5 Youtube tabs, I watch my task manager and almost all of the download is coming through my DSL. Is there any way to force specific domains to use more download on another interface?

Ysurac, 2020-02-23:
How do you check that it's using the bonded line ?

Ysurac, 2020-02-23:
Download use the interface with the smaller RTT first. So if DSL have the lowest RTT, no problem here.

darthclide, 2020-02-23:
> How do you check that it's using the bonded line ?

Because I change my default route to 192.168.100.1 and my browser reports my external IP as the VPS server IP.

darthclide, 2020-02-23:
> Download use the interface with the smaller RTT first. So if DSL have the lowest RTT, no problem here.

Okay, but my problem is that once my DSL is saturated, nothing else gets used. I want to see my DSL at 16 down, AT&T at 16 down, and my Verizon and 16 down. All loading the Youtube videos so I get 0 buffering + extra 30 seconds on all of them.

Ysurac, 2020-02-23:
> > How do you check that it's using the bonded line ?
> 
> Because I change my default route to 192.168.100.1 and my browser reports my external IP as the VPS server IP.

I speak about youtube

Youtube is not a good way to test bandwidth. There is ```omr-test-speed``` available on the router via SSH. ```omr-test-speed wan1``` will test speed via wan1, without an interface this test aggregated speed. You can stop test with ctrl+c after a few minutes.

closed 2020-08-12 by Ysurac