Ysurac/openmptcprouter #844: Port forwarding steps

Archana7700, 2020-02-18:
I have followed ur documentations https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding
i m not able to configure the port forwarding steps Please if you have any new documentations can you share it . Because as the it is new version (v0.53.3.r) the documentations which u have provided is old version (v0.5.1) i m unable to find that page in the new version please can u share the new documentations.

dabuide, 2020-02-20:
It's very simple :
`Network / Firewall / Port Forward / Add / Name (Services) / Protocol (TCP or UDP) / Source zone (VPN) /External port (port open) / Destination zone (LAN) / Internal IP (IP for machine open port) / Internal port (port open)`
And on :
`System / OpenMPTCProuter / Advanced Settings / check > "Redirect all ports from server to this router"`

Archana7700, 2020-02-24:
Hi Ysurac/openmptcprouter,
Thank you for your email,i have got the firewall setting but i have some
other issue could you please help me out.
 I m facing following issues:

1.Please can you help me out with openvpn, actually this is the first time
i have been working with openvpn .I have set the server by the  script
which u have given in tutorial while running i found the open vpn
certificate, but i dont know how to create the certificate at the client
side i have been using raspberry pi i have been working on this from a week
n i m unable to figure it out.

2.In port forwarding you have used (192.168.100.3 ip ) for what that ip
belongs to as in at the client side or the server side what port has to be
forwarded to.

3.In shadowsocks at the remote server page i have set the ip of my
server,as well as password,port(8388),but when i m checking in logread i m
finding an error as shoadowsocks rules down why is that so?

4.Can i create other interfaces and bridge it to the wan1 and wan2 ?if so
when i tried it i faced the warning as multipath is blocked at the
connection as well as wan is down but ip was pinging internet was proper tx
and rx was increasing.

5.whenever i m using the luci page for long time i m facing the issue as
luci page got corrupted and XHR request timeout and it works only after i
flash the sd card do i m changing any configuration which is making all
files corrupted i m not able to correct this out.

closed 2020-09-14 by github-actions[bot]

Candyffm, 2022-01-30:
The guide is still valid but I incomplete for people like me that have never used openwrt before. I set to forwarding as need for my plex server
[image]
but I missed the tiny blue button in the top right corner that says 'x amount unsaved changes' to apply all settings.

I am used to having settings applied when the "save" button is pressed. Wasted a couple of hours on that. Obviously that button is not specific to port forwarding but it might be worth a little notice in the debug section.