Ysurac/openmptcprouter #846: Low Speedtest results and laggy response

jpadie, 2020-02-19:
## Expected Behavior

1. the overall speed test results as reported by web-based speed test apps to be faster than the individual speeds of the links.

2. there not to be lagging on almost all page requests.  currently I type a web request or search and the response is that the internet is disconnected, then a second or two later the page is refreshed with the results.

3. running telephony over the circuit is very poor quality whereas it is crystal clear over either aggregated line

## Actual Behavior

as above.

## Steps to Reproduce the Problem

  1.  set up VPS on Ubuntu 18.04 at digital ocean using the install script.  Kernel verified.
  2. follow instructions for router installation.  Raspberry Pi 3B+
  3. wan interfaces on 192.168.8.1 and 192.168.1.1.  Both 4G routers, one connected to SFR and getting 16/16 Mbps and the other to Orange (13/15 Mpbs).  
  4. lan interface on 192.168.100.1/24 into smart switch.
  5. wifi distributed by unifi AP

## Specifications

  - OpenMPTCProuter version: 0.53.3 squashfs.
  - OpenMPTCProuter VPS version: 0.1008
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) RIP3B+

Is there a particular setting to make OMR play more nicely with 4G routers?  and for telephony?  
It's odd behaviour.  The router interface frequently saying that the WAN servers are down (understood given they are 4G) and the VPN is down (less understood), shadowsocks down etc.  I imagine these are all because for some reason the 4G routers are disconnecting themselves pending requests.  

Netflix is also blocked but I understand that to be a known issue that can be circumvented with OMR-Bypass. 

thanks in advance for pointers.

Ysurac, 2020-02-19:
What is the latency you get on status page for each WAN ?
If you have an interface that is more stable than the other, you can set it as master (disable the change of master in OpenMPTCProuter menu advanced settings tab and change parameter in Network->MPTCP).
How routers are connected to RPI ? if it's usb do you use a powered USB hub ?

jpadie, 2020-02-19:
Hi

sometimes the latency is more than 500ms, and sometimes, as mentioned, the core issue is that the web page report that the internet is disconnected.

both interfaces are rock-solid when used on their own (directly plugged into the switch).  it's only when they are part of the OMR grouping that there are perceived problems.  

each modem/router has a static IP and is plugged into a switch.  the rPI is plugged into the same switch.  the switch is powered.

Ysurac, 2020-02-19:
I speak about latency in status page, that is latency between WAN and VPS part. I hope it's not 500ms.

jpadie, 2020-02-19:
I will check and report back.  I've had to step away for 20 minutes. thanks.

jpadie, 2020-02-19:
no latency disclosed on the OMR status page nor of the OpenWRT status page.  I'm not sure what published data point you are asking for.  

[image]

note the red text in the bottom box comes on and off relatively 'randomly'.  i.e. i cannot discern any change in status in the VPS or on the modem admin pages that would help.  However the modems do not allow any configuration of the UMTS connections themselves. 

sometimes the status is like this too (verified that omr-admin.py is running).

[image]

as another example, this has taken several minutes to send as the internet reports as being unavailable.  despite of course the links seeming to be up from an examination of the modems.  

any help gratefully appreciated.

Ysurac, 2020-02-19:
It seems that the router have difficulties to contact the VPS. You should use the wan1 as master, the wan2 seems to have even more difficulties, there is a latency available for wan1 on screenshot, but never for wan2.

jpadie, 2020-02-19:
Thanks Yannick
i have changed the master to wan1 and will see what happens.  At the moment the same red warnings are given for the VPS.  
I am using a digital ocean droplet as a VPS.  Do you recommend an alternative?  or is this more likely to be a 4G issue that i cannot work around?

jpadie, 2020-02-19:
[image]

Ysurac, 2020-02-19:
4G is working with OpenMPTCProuter, but in your case there is problem to contact VPS.
Try with a Debian, there is issues in current script with Ubuntu (solved in develop, fixes will be available in next release).

jpadie, 2020-02-19:
thanks again.

I have pointed everything at a Debian build as suggested but the behaviour is identical.  

Is there any specific setting needed for 4G routers or anything else I might try?

jpadie, 2020-02-19:
[image]

jpadie, 2020-02-19:
Just in case helpful, this is the system log of the last few minutes. 

`
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25757]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25758]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25758]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25757]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25757]: send: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25759]: send: Broken pipe
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25756]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25758]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25757]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25758]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25756]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25757]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25758]: remote recv: Connection reset by peer
Wed Feb 19 16:18:54 2020 daemon.err /usr/bin/ss-redir[25757]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25757]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25758]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25757]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25757]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25757]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25757]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25758]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25757]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25758]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25757]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25758]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25757]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25758]: send: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-local[25755]: remote_recv_cb_recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25758]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25758]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25756]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-redir[25759]: remote recv: Connection reset by peer
Wed Feb 19 16:18:55 2020 daemon.err /usr/bin/ss-local[25755]: remote_recv_cb_recv: Connection reset by peer
[... cut ...]
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
vnstat.@vnstat[0].backup='0'
`

Ysurac, 2020-02-19:
You are using multiples VPS ? This work with one VPS ?

jpadie, 2020-02-19:
sorry - no.
i added another server just before I posted the config.  the problems over the last few days have all been with an ubuntu server in Digital Ocean and, today, a Debian server in the same environment.   Not both at the same time (until a few minutes ago as a trial).

Ysurac, 2020-02-19:
This doesn't help to debug

jpadie, 2020-02-19:
yes.  i appreciate that.  I posted that config before remembering to delete the second VPS.

here is the current config.  I believe that I have reverted everything to the previous behaviour.

`
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.start='10'
dhcp.lan.limit='200'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.vlan100=dhcp
dhcp.vlan100.interface='vlan100'
dhcp.vlan100.limit='150'
dhcp.vlan100.leasetime='12h'
dhcp.vlan100.start='10'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
[... 1273 lines cut ...]
upnpd.config.enable_upnp='1'
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
vnstat.@vnstat[0].backup='0'
`

jpadie, 2020-02-19:
and here are the logs since reverting the config

`
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.059944] [    280]     0   280      204       10    28672        0             0 askfirst
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.070911] [    336]     0   336      235       11    28672        0             0 urngd
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.081441] [   1339]     0  1339      302       48    36864        0             0 logd
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.091742] [   1380]     0  1380      579       95    32768        0             0 rpcd
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.101823] [   1671]     0  1671      245       11    24576        0             0 dropbear
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.112291] [   2440]     0  2440      451       65    28672        0             0 netifd
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.122572] [   2531]     0  2531      241       10    24576        0             0 rngd
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.132625] [   2581]     0  2581      354       32    28672        0             0 odhcpd
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.142757] [   2886]     0  2886      310       14    24576        0             0 crond
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.152796] [   3420]     0  3420     1248      186    40960        0             0 uhttpd
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.163440] [   3753]     0  3753      407       27    32768        0             0 dbus-daemon
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.174248] [   3832]     0  3832     3607      293    57344        0             0 ModemManager
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.184922] [   5848]     0  5848      469       20    28672        0             0 hostapd
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.195207] [   5970]     0  5970      469       20    32768        0             0 wpa_supplicant
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.206070] [   9797]     0  9797      310       16    28672        0             0 ntpd
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.216216] [   9907]     0  9907      333       37    28672        0             0 sh
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.226104] [   9914]     0  9914      905       63    32768        0             0 adb
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.236072] [   9946]     0  9946      978       63    36864        0             0 adb
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.246029] [  29442] 65536 29442     5940     4249    73728        0             0 unbound
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.256872] [  25755]     0 25755      585       53    32768        0             0 ss-local
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.267410] [  25756]     0 25756    19304    16649   184320        0             0 ss-redir
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.277867] [  25757]     0 25757    18816    15676   180224        0             0 ss-redir
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.288484] [  25758]     0 25758    19253    16073   184320        0             0 ss-redir
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.298953] [  25759]     0 25759    19467    15638   184320        0             0 ss-redir
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.309452] [  26026]     0 26026      331       47    36864        0             0 glorytun
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.319970] [  26725]     0 26725      653      138    32768        0             0 vnstatd
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.330390] [  28005]     0 28005      602       37    28672        0             0 nginx
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.341122] [  28071] 65534 28071    17521    14721   163840        0             0 nginx
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.351619] [  28072] 65534 28072    24500    21850   225280        0             0 nginx
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.361944] [  28073] 65534 28073    33717    28780   299008        0             0 nginx
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.372363] [  28074] 65534 28074    11044     9413   110592        0             0 nginx
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.382770] [  30703]     0 30703      362       70    28672        0             0 sh
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.392872] [  32163]     0 32163      309      152    28672        0             0 sh
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.403221] [   3629]     0  3629      347       54    32768        0             0 sh
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.413693] [   3630]     0  3630      339       43    32768        0             0 sh
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.423770] [   3631]     0  3631      340       44    28672        0             0 sh
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.433876] [   3632]     0  3632      313       18    28672        0             0 sh
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.444342] [   6883]   453  6883      689       40    32768        0             0 dnsmasq
Wed Feb 19 16:39:44 2020 kern.info kernel: [104378.454966] [   6894]     0  6894      358       63    32768        0             0 sh
[... cut ...]
Wed Feb 19 16:44:22 2020 daemon.err /usr/bin/ss-redir[25757]: remote recv: Connection reset by peer
Wed Feb 19 16:44:22 2020 daemon.err /usr/bin/ss-redir[25758]: remote recv: Connection reset by peer
Wed Feb 19 16:44:22 2020 daemon.err /usr/bin/ss-redir[25757]: remote recv: Connection reset by peer
Wed Feb 19 16:44:22 2020 daemon.err /usr/bin/ss-local[25755]: remote_recv_cb_recv: Connection reset by peer
Wed Feb 19 16:44:23 2020 daemon.err /usr/bin/ss-redir[25757]: remote recv: Connection reset by peer
`

i'm seeing the memory errors.  Are they anticipated?

Ysurac, 2020-02-19:
The shadowsocks-libev and glorytun host are still "127.0.0.1", possible that I don't restore correct IP after adding another VPS...
Memory errors are strange, do you use P2P or something that use many connections ?

jpadie, 2020-02-19:
The pi is your image with no added functionality.  I've found the key ingestion from the server admin script has never worked over the few days I've been trying it.  All the keys have had to be added manually in the advanced interface.  If that helps diagnosis 

No other services are run on the Pi. 

Inside the network there are the usual assortment of laptops and less usually there are Sonos, unifi devices, a Ring door bell, some sonoffs, a couple of echo dots and a couple of smart TVs.

Ysurac, 2020-02-19:
Can you try a ```systemctl restart omr-admin``` on the VPS and check if you have an answer on https://<yourvpsip>:65500/ ? Then can you check in ```/var/log/daemon.log``` on the VPS ?

jpadie, 2020-02-19:
I've had to leave the site today, and because the internet has been unusable I've removed the Pi+ from the network and made one of the 4G routers the gateway.   It's up solidly.  

I have followed your instructions above on the VPS. the admin script is available on that port.  daemon.log is available here: https://1drv.ms/u/s!Aob6wYJSSDO0kOo5IligJzpjEqySkw

Ysurac, 2020-02-19:
I found an error in the admin script, you can copy https://github.com/Ysurac/openmptcprouter-vps-admin/raw/develop/omr-admin.py to ```/usr/local/bin/``` on the VPS (and ```chmod u+x omr-admin.py``` to make it executable, ```systemctl restart omr-admin``` to start it) or run snapshot script https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

jpadie, 2020-02-19:
will give it a whirl tomorrow.  

thanks for the support today.

jpadie, 2020-02-25:
Please excuse the delay in progressing.  I've not been in the country for a while but should be back next week to carry out further testing.

closed 2020-09-14 by github-actions[bot]