Ysurac/openmptcprouter #848: Request: MPTCP over VPN without OPENVPN

lars18th, 2020-02-20:
Hi @Ysurac ,

I see that the new functionality of MPTCP over VPN (in Beta) is using _only_ OpenVPN as the transport layer: https://github.com/Ysurac/openmptcprouter-feeds/blob/aa545395a029c0b454f1a257fe793dbd186efc87/openmptcprouter/files/etc/init.d/mptcpovervpn

From my point of view, as the OMR has support for a lot of other VPN managers (using UDP or TCP), then I suggest to support _other_ protocols. I want for example to use DSVPN and/or Glorytun-UDP instead of OpenVPN.

Please, can consider this enhancement?
Thank you!

Ysurac, 2020-02-20:
I need to be able to bind the VPN to an interface, and it's also far easier if the server can receive multi client. For now only OpenVPN can do that on OpenMPTCProuter.

lars18th, 2020-02-21:
Hi @Ysurac ,

Perhaps this can make things more easy:

- Add to every interface a flag to indicate if OMR requires to use it, or not. If not, then you don't need to touch/change anything for this interface (in practice, it will be for OMR as not used or inexistent). In all these interfaces the MPTCP will be disabled. With this change, the user can configure _any_ connection (VPN, dialup, tunnel, etc.) inside the OMR without disturbing your work.
- Second, for MPTCP over VPN you apply an small change in the platform. Instead of using directly an OpenVPN interface, you use _only_ bridges. Then you configure the bridge to do the task, add you include too support for the user to add to the VPN bridge _any_ of the interfaces marked as non-OMR (the one of the previous paragraph).

Then for OpenVPN the difference with the current behaviour is the requirement to add the interface to the VPN brigde after configure the connection.

With this approach then ANY interface can be used, as the VPN interface is _decoupled_ from the used interface.

From my point of view, one change required in OMR for more flexibility is the option to configure interfaces not used by you. This will provide more flexibility to configure complex scenarios.

I hope you agree.
Regards.

Ysurac, 2020-02-21:
* You can set an interface to be used or not by OMR in Network->MPTCP
* I don't understand why this is related ? the MPTCP over VPN is here to make it easier to configure, but you can do it manually without issues.

lars18th, 2020-02-21:
Well, try to do this: Configure one VPN interface without using OpenVPN **and** not used by OMR. After that configure a bridge and try to use it with the OMR as the current "MPTCP over VPN". Are you sure this will work?

If you can do that without troubles, then please explain the steps... I don't know how to achieve it.

In any case, thank you for this impresive project! :heart:

Ysurac, 2020-02-21:
Why using a bridge here ?
If you set an UDP VPN to the VPS that is not bind to an interface, it will use only master interface but this should work with MPTCP over VPN. But I don't know why anybody would want that.

closed 2020-05-18 by Ysurac