Ysurac/openmptcprouter #849: Request: PROXY support for TCP VPN protocols

lars18th, 2020-02-20:
Hi @Ysurac ,

In my environment some of my Wan connections are using proxies (Socks4 and HTTP). So I suggest if you can add support for PROXIES using VPN protocols with TCP connections.

Please! :wink:

lars18th, 2020-09-09:
Hi @Ysurac ,

Please don't close this Request! What I'm requesting is that you will support for WAN connections, not only plain IP connections, or PPPoE, but VPN connections too. My suggestion is to add the ability to "mark" _any_ network connection as "WAN for OMR" (and not use others for this objective).

I hope you agree.
Regards.

Ysurac, 2020-09-09:
It's already possible to mark any connection as available by enabling Multipath on it.

lars18th, 2021-01-02:
Hi @Ysurac 

> It's already possible to mark any connection as available by enabling Multipath on it.

A multipath connection implies that's a WAN connection?

Ysurac, 2021-01-02:
This implies that it's a connection that should have direct Internet access.

lars18th, 2021-01-02:
> This implies that it's a connection that should have direct Internet access.

Why? I can think on almost two different cases in which this is not true:

1. When the "target" network doens't have "direct" Internet access, but access through a proxy.
2. When the "target" network is not Intenert, but a corporate network with multiple subnetworks with LAN private addresses.

My suggestion is simple: Please, not enforce that a connection labeled "WAN" (in the OMR) is a "direct connection to Internet". It only needs to be a connection with: 1) Multipath support; 2) And second, with the presence of a gateway server (that can or cannot have Internet access).

Regards.

Ysurac, 2021-01-02:
What you want exactly ?
WAN is a connection to internet, at least it's not a LAN connection. All WANs should have access to same WAN (internet or any WAN).
Then I don't enforce anything, if a connection have multipath enabled then some route needed for multipath to work are set.

lars18th, 2021-04-06:
> What you want exactly ?

I want ot use OMR to route to PRIVATE LANs _only_.

> WAN is a connection to internet, at least it's not a LAN connection. All WANs should have access to same WAN (internet or any WAN).
> Then I don't enforce anything, if a connection have multipath enabled then some route needed for multipath to work are set.

IMHO, the current implementation has the assumption that the WAN can only be achieved over the Multipath connection. Why this? Why not provide a method to leave the user to use any route over any link? For example, perhaps by default OMR thinks the best is to route all traffic over the multipath route to the server if it's available. But some user perhaps wants to route only some networks over this multipath link to the remote server and leave other networks (including the default route) to go over another link.

Please, consider this "complex use case".
Thank you.

lars18th, 2021-07-07:
Hi @Ysurac ,

I hope at some time you will enhance the OMR to support _PRIVATE REMOTE NETWORKS_. The current implementation doesn't support directly these two use cases:

1. When the "public" server has to be connected using a proxy. That's the case when the VPN protocol requires to run over a proxy or private connection (and for this case the strong encryption is useless and a wasting of power resources).
2. When the "public" server not provides a connection to the public Internet but to a group of private networks. Then the "WAN" connection is not the "multipath VPN connection" but a regular single WAN. In this case you only want to route specific LANs over the VPN connection.

So in a simple graph:
```
               Internet
                  |
                  |
Client ----> LAN Router 1 ----> OMR Router --/MP-VPN/--> Private LAN
                  |
                  |
             LAN Router 2
                  |
                  |
               Internet
```

In this case the CLIENT is using these routes:
- Default: Using LAN Router 1 (or 2) as default gateway.
- Private LAN: Using OMR Router as a gateway.

And with this constrains:
- The OMR Router uses the default gateway of the local LAN for the WAN connection.
- The VPN connections of the OMR Router use a PROXY (or tunnel).

I hope you will want to support this.
Regards.

lars18th, 2021-10-07:
Hi,

This request continues to be valid. Please, don't close it! The current configuration doesn't provide support to connect to private LANs "only". Please, add support to "Not Route to Internet over the Multi-path connection".

Thank you!

lars18th, 2022-01-10:
Hi,

This request is still valid. Please, consider it.

lars18th, 2022-10-11:
Request still valid.

lars18th, 2023-01-11:
Hi @Ysurac ,

I continue requesting this. Please, consider it!

cr08, 2023-01-27:
I'm also adding in my 2 cents to this. In my use case scenario using cellular WAN connections via a tethered phone, for some devices/carriers it may be necessary to use something like PDAnet to not get hard throttled to abysmal speeds even on speedy, uncapped LTE/5G connections. However how these usually work for clients that cannot run their own software is using a simple proxy server per their documentation linked below (It indicates this primaryly works over Wifi Direct. I have not had a chance to see if this also applies via USB which is my preferred connection method here).

http://pdanet.co/a/wifi/others.php

Ideally the proxy use would be isolated to each individual WAN connection and just OMPTCP's internal use to gain a functional, full speed WAN link and not interfere with the other functionality.

cr08, 2023-01-29:
Expanding on my comment above and this may stray off of the intended needs for the OP, but for anyone else aiming for unthrottled cellular connections, I found a working alternative at least for Android devices:

There's an application called EasyTether which basically does the same as PDAnet. However they provide installable network drivers for various operating systems to accomplish the same thing instead of using a proxy. They have also provided compiled packages for OpenWRT. The files are all relatively dated with the latest version of the OpenWRT drivers provided labelled for version 19.07.3+. However testing it seems to function just fine on the current version OMPTCP uses. This is with a Pi 2B test bench. It just shows up as another available network interface to use and allows full unthrottled bandwidth.

Figured I'd toss this out for anyone who may come across this.

lars18th, 2023-01-29:
I am glad to hear others asking for the same thing: please PROXY suppport for _all_ VPN protocols. 😢

Ysurac, 2023-01-29:
For now, the only possible solution I can see is to use MPTCP over VPN with OpenVPN as VPN with http proxy configured.
But this will be slow...

lars18th, 2023-01-29:
Hi @Ysurac ,

> For now, the only possible solution I can see is to use MPTCP over VPN with OpenVPN as VPN with http proxy configured.
But this will be slow...

Why? You can proxy _any_ TCP stream with a HTTP or Socks4 proxy. And _any_ UDP transport with a Socks4/5 proxy. So in fact _ALL_ VPN protocols used by OMR could operate using proxies.

lars18th, 2023-02-20:
Hi @Ysurac ,

I feel the problem with the PROXY support could be a miss interpretation of the technical aspects. Please, see  this environment:
```
CLIENT --/ TCP /--> OMR (mTCP support) --/ mTCP path 1 /--> PROXY-1 --> OMR-Server (mTCP support)
                                      |--/ mTCP path 2 /--> PROXY-2 --^                          
```

Then my question is: The PROXY-1 and PROXY-2 servers require mTCP support or not?
My feeling is that the mTCP support is not required for intermediate proxies. This is true or not?

And another question is why you want to execute a TCP VPN protocol over the top if you can directly transport the TCP _stream_ over a TCP connection. TCP-over-TCP is a bad idea, however STREAM-over-TCP is not a problem... that's is what SSH is doing with forwardings.

lars18th, 2023-05-22:
Hi @Ysurac ,

Please, I hope you want to consider this.

closed 2023-08-27 by github-actions[bot]

lars18th, 2023-08-30:
Hi @Ysurac ,

I continue requesting to support the PROXY functionality for the TCP VPN protocols.