Ysurac/openmptcprouter #850: Question: Can an OMR VPS also host another low-volume server like an MQTT Broker?

bnhf, 2020-02-20:
## Expected Behavior

Hopefully, that I can use my OMR VPS (1v core, 1GB RAM, 25GB SSD on Vultr.com) to host another server or two like an MQTT Broker, and/or a Ubiquiti UniFi Cloud Controller. Both would just be for my personal use, so would have modest traffic

## Actual Behavior

Don't know. I'm hoping somebody else has tried this.

## Steps to Reproduce the Problem

  1. Install VPS
  2. Install Mosquito
  3. Install Unifi Cloud Controller

## Specifications

OpenMPTCProuter version: v0.53.3 r0+11546-a74095c68c
OpenMPTCProuter VPS version: 0.1008
OpenMPTCProuter platform: RPi4

Ysurac, 2020-02-21:
Yes, this should work even if not recommended.

Dragonox77, 2020-02-24:
I think you should change the server.
This is my vps with only OMR installed on it.
http://prntscr.com/r716yc

bnhf, 2020-02-26:
Thanks for the replies. I can see now that putting the MQTT broker on the OpenMPTCProuter side of things (with appropriate port forwarding), rather than installing it on the VPS, is a better solution. Alternatively, if the traffic ever gets to be more than I want to have on my cellular links, I'll spin-up another virtual server on Vultr.

closed 2020-02-29 by bnhf