Ysurac/openmptcprouter #851: Can't aggregate DSL + 4G LTE

Telejunky, 2020-02-22:
## Expected Behavior

 Bonding 150MBit/s DSL + 60MBit/s 4G LTE = MORE THAN 150MBit/s

## Actual Behavior

**Only** the "multipath: master" is used for Connections!!
When changing master to 4G LTE than **only** 4G is used for Connections.

## Steps to Reproduce the Problem

  1. Set up openmptcprouter with Wizard
  2. make speed tests
  3. only 1 WAN (WAN1 DSL) is used

## Specifications

  - OpenMPTCProuter version: **v0.53.3** r0+11546-a74095c68c
  - OpenMPTCProuter VPS version: 0.1008
  - OpenMPTCProuter platform: x86_64

**All router settings:**
https://pastebin.com/ZSDYXkYS

```
root@OpenMPTCProuter:~# traceroute 1.1.1.1
traceroute to 1.1.1.1 (1.1.1.1), 30 hops max, 46 byte packets
 1  *  10.255.255.1 (10.255.255.1)  7.730 ms  *
 2  62.240.xxx.xxx (62.240.xxx.xxx)  8.355 ms  *  7.964 ms
 3  193.9.xxxxxxxxx.xxxrou1.xxxxx.at (193.9.xxxxxxxx)  8.162 ms  8.409 ms  8.362 ms
........
 7  vix.as13335.net (193.203.0.195)  20.229 ms  17.594 ms  11.578 ms
 8  one.one.one.one (1.1.1.1)  8.572 ms  8.184 ms  8.589 ms
```

**omr-test-speed**
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  3 10.0G    3  322M    0     0  14.5M      0  0:11:43  0:00:22  0:11:21 13.6M

**omr-test-speed wan1**
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  2 10.0G    2  243M    0     0  10.8M      0  0:15:47  0:00:22  0:15:25 17.3M

 **omr-test-speed wan2**
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 76.2M    0     0  5414k      0  0:32:16  0:00:14  0:32:02 7045k

[image]

Ysurac, 2020-02-22:
What is the VPS provider ? What is the bandwidth available ? What is CPU and memory of this VPS ?
Same for OpenMPTCProuter, what is the CPU and memory available ?
You can try another "Multipath TCP scheduler" in Network->MPTCP.
What is the result of ```uname -a``` on the VPS ?

Telejunky, 2020-02-22:
VPS: uname -a
Linux WehxxxxxxxT1 4.19.80-mptcp #1 SMP Wed Nov 20 12:28:05 UTC 2019 x86_64 GNU/Linux

VPS Provider is my company (VM on Proxmox 6.1-1)
Quad Core CPU with 4GB RAM
1Gigabit/s Down & 1Gigabit/s Up

OpenMPTCProuter also on VM on Proxmox 6.1-1 (at my home)
Quad Core CPU with 4GB RAM
WAN1 bridged to ETH0
WAN2 bridged to ETH0
ETH0 bridged to Server NIC .. the two gateways are in one LAN

I already tried a lot of different Multipath TCP scheduler.

I just made a speedtest on my VPS:
Testing download speed................................................................................
Download: 894.05 Mbit/s
Testing upload speed......................................................................................................
Upload: 4.16 Mbit/s

Apparently there is a problem with the Uploadspeed. I am in contact with the network team. Please let me wait for response before closing the issue.
Thanks! :)

Slivikkus, 2020-02-22:
Speedtest-cli showing always 4.16Mbits in upload, so ...

Ysurac, 2020-02-22:
True, it's a bug in speedtest-cli.
You can also try to set shadowsocks-libev method to none, instead of, I suppose AES.

triple-it, 2020-02-24:
Or give speedtest.c a spin?

Telejunky, 2020-02-26:
My upload speed is ok and also over 900Mbit/s.

I tried a lot of different TCP Multipath schedulers and other settings. Also I tried shadowsocks-libev method to none.
Nothing helps.
Only 1 WAN is used.

Don't know what I can do further

Ysurac, 2020-02-26:
You can try to disable TCP Fastopen in OpenMPTCProuter->Advanced settings tab.

Telejunky, 2020-02-27:
disabling Fastopen didn't help..  i only get 150Mbit/s

Telejunky, 2020-02-28:
does this look good?

**WAN1**
1: 192.168.178.1 
2: 62.xxx.xxx.x  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
3: 10.xx.xx.xx  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
4: 10.xx.xx.x  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
5: 193.x.xxx.xxx  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
6: 62.xxx.xxx.xxx TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize -TCPOptionMPTCPCapable 

Result:
< Ethernet (14 bytes) :: DestinationMAC = 62:xx:xx:xx:xx:xx , SourceMAC = 98:xx:xx:xx:xxf:xx , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 44 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 59 , Protocol = 0x6 , CheckSum = 0xba5 , SourceIP = 62.240.130.135 , DestinationIP = 192.168.178.7 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 36999 , SeqNumber = 3653563929 , AckNumber = 1333935902 , DataOffset = 6 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 29200 , CheckSum = 0x4da , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1380 , >

**WAN2**
1: 192.168.1.1 
2: 176.xxx.xx.xx  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
3: 193.x.xxx.xx  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
4: 10.xx.xx.x  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
5: 193.x.xx.xxx  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
6: 62.xx.xx.xxx TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize -TCPOptionMPTCPCapable 

Result:
< Ethernet (14 bytes) :: DestinationMAC = 46:xx:xx:xx:xx:xx , SourceMAC = 10:xx:xx:xx:xx:xx , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 44 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 59 , Protocol = 0x6 , CheckSum = 0xbbb4 , SourceIP = 62.xxx.xxx.xxx , DestinationIP = 192.168.1.248 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 8498 , SeqNumber = 1873352163 , AckNumber = 2088706373 , DataOffset = 6 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 29200 , CheckSum = 0x5f6d , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1380 , >

Ysurac, 2020-02-28:
The "-TCPOptionMPTCPCapable" doesn't look good. Yes you should have a "+"
It strange that this happen at last hop.

Telejunky, 2020-02-28:
I'm in contact with network team because I think that our Cisco ASA is not supporting MPTCP option 30 by default.

see: https://www.cisco.com/c/en/us/support/docs/ip/transmission-control-protocol-tcp/116519-technote-mptcp-00.html
ASA
TCP Operations

By default, the Cisco ASA firewall replaces unsupported TCP options, which include the MPTCP option 30, with the NOOP option (option 1). In order to permit the MPTCP option, use this configuration:

    Define the policy in order to allow TCP option 30 (used by MPTCP) through the device:

    tcp-map my-mptcp
       tcp-options range 30 30 allow

    Define the traffic selection:

    class-map my-tcpnorm
       match any

    Define a map from traffic to action:

    policy-map my-policy-map
       class my-tcpnorm
           set connection advanced-options my-mptcp

    Activate it on the box or per-interface:

    service-policy my-policy-map global

Ysurac, 2020-02-28:
True. Someone already had this problem, I should add it in the FAQ.

Telejunky, 2020-03-12:
Currently, my network team has no time to test the solution. So we can close the thread for now..

closed 2020-03-12 by Telejunky