Ysurac/openmptcprouter #852: PCIE BCM4350 Wireless Network Adapter

gitxmax, 2020-02-23:
## Expected Behavior
  In English please.

Pcie card recognized, when the drivers are installed
## Actual Behavior
Kernel driver not loading

## Steps to Reproduce the Problem
Tried installing possibly compatible drivers
  1. brcmsmac-firmware
  2. kmod-brcmsmac

## Specifications

  - OpenMPTCProuter version: (Last is not a version)
openmptcprouter v0.53.3 r0+11546-a74095c68c / LuCI Master git-19.354.24352-2096d67
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64)
x85_64

Hello,
Have the AMD Athlon 200GE powered pc with DELL-DW1820A (BCM4350 chip) Pcie wireless card installed.
Tried installing a number of possibly compatible bcm43xx drivers, but failing to get the card recognized.

`root@OpenMPTCProuter:~# lspci
00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Root Complex
00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 IOMMU
00:01.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge
00:01.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 PCIe GPP Bridge [6:0]
00:08.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge
00:08.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Internal PCIe GPP Bridge 0 to Bus A
00:08.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Internal PCIe GPP Bridge 0 to Bus B
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller (rev 61)
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge (rev 51)
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 0
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 1
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 2
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 3
00:18.4 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 4
00:18.5 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 5
00:18.6 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 6
00:18.7 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 7
01:00.0 USB controller: Advanced Micro Devices, Inc. [AMD] Device 43bc (rev 02)
01:00.1 SATA controller: Advanced Micro Devices, Inc. [AMD] Device 43b8 (rev 02)
01:00.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 43b3 (rev 02)
02:04.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] 300 Series Chipset PCIe Port (rev 02)
02:05.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] 300 Series Chipset PCIe Port (rev 02)
02:06.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] 300 Series Chipset PCIe Port (rev 02)
02:07.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] 300 Series Chipset PCIe Port (rev 02)
03:00.0 Network controller: Broadcom Inc. and subsidiaries BCM4350 802.11ac Wireless Network Adapter (rev 08)
05:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
07:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Raven Ridge [Radeon Vega Series / Radeon Vega Mobile Series] (rev cb)
07:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Raven/Raven2/Fenghuang HDMI/DP Audio Controller
07:00.2 Encryption controller: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 10h-1fh) Platform Security Processor
07:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Raven USB 3.1
07:00.4 USB controller: Advanced Micro Devices, Inc. [AMD] Raven USB 3.1
08:00.0 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 61)
`
`dmesg | grep "03:00.0"
[    0.248466] pci 0000:03:00.0: [14e4:43a3] type 00 class 0x028000
[    0.248541] pci 0000:03:00.0: reg 0x10: [mem 0xfc400000-0xfc407fff 64bit]
[    0.248563] pci 0000:03:00.0: reg 0x18: [mem 0xfc000000-0xfc3fffff 64bit]
[    0.248693] pci 0000:03:00.0: supports D1 D2
[    0.248694] pci 0000:03:00.0: PME# supported from D0 D1 D2 D3hot D3cold
`

Ysurac, 2020-02-25:
I have no idea
Maybe this will be resolved in next release that update kernel/drivers.
You can check if you found doc about that on OpenWrt.

gitxmax, 2020-02-26:
Openwrt replied, no support will ever happen. Thanks

closed 2020-02-26 by gitxmax