Ysurac/openmptcprouter #853: Should I replace my USB adapters? Any other suggestions to help bond my connections?

darthclide, 2020-02-24:
## Expected Behavior

AT&T, Verizon, and DSL bond together so my Twitch stream never falls below my weakest link (which is DSL at 4 Mbps upload)

## Actual Behavior

AT&T and Verizon constantly "give up" and my DSL obviously can't handle my 5200 kbps stream. Weird spikes in the graph also happen. Maybe these are related to my problems?
WAN1 = DSL
WAN2 = AT&T
WAN3 = Verizon

https://media.discordapp.net/attachments/645483729497620480/681347696686268470/unknown.png
https://media.discordapp.net/attachments/645483729497620480/681367495201914927/unknown.png

## Steps to Reproduce the Problem

Simply boot up the VM and start the stream. Issue comes and goes with no apparent cause.

## Specifications

  - OpenMPTCProuter version: v0.53.3 r0+11546-a74095c68c
  - OpenMPTCProuter VPS version: Version 0.1009 4.19.80-mptcp
  - OpenMPTCProuter platform: VPS server = Debian 10 x64 My computer = host machine: Windows 10 Pro x64, with router software on Virtual Box (operating system Linux 2.6 / 3.x / 4.x (64-bit)

These are the adapters I am using:
https://www.amazon.com/Waxden-Adapter-1200Mbps-Wireless-Network/dp/B07V55C2XF/ref=sr_1_3?keywords=waxden+adapter&qid=1582521581&s=electronics&sr=1-3

AT&T is running on a Velocity 2
Verizon is running on a Jetpack 8800L
DSL is coming in on a CAT 5 cable

I want to combine my T-mobile signal through a hotspot on the phone (yes dun=0 is set, and speedtests show great results), but whenever I turn it on, it somehow screws up the bonder as well as my default route. Like my external IP shows as the VPS server, but my youtube barely loads (this happens when I am on T-mobile without a VPN. It suggests dun=0 only works slightly? But this shouldn't matter if I am connecting to an encrypted VPS server. T-mobile shouldn't be able to see I am bonding the connection right?

darthclide, 2020-02-26:
Still hoping to hear from someone on which adapters I should get as well as if I should get a Raspberry 4 or a 3B like my friend reedog has.

Upon further testing, I have determined Verizon is the cause of my headaches. However, I can connect perfectly to the Jetpack 8800L with my phone. It is just these adapters that struggle. Can anyone here confirm which adapters have the best connectivity to this Jetpack?

bnhf, 2020-02-29:
Sounds like you're connecting to the Velocity and Jetpack using those WiFi adapters. It'd be best to tether them using USB. If you're hosting OMR in VirtualBox on a Windows computer, busy with other tasks, you should consider moving it to dedicated hardware. An RPi 3B+ or 4 would be fine, or a dedicated x86 box would be better. OMR requires some horsepower.

darthclide, 2020-03-01:
If you have any suggestions for a similarly priced x86 box I am all ears. However, I was going for a PI not just for its price, but also the size. It would need at least 4 USB ports, and an ethernet jack. May I ask if a PI would have problems understanding an ethernet to USB adapter? I intend on bringing the DSL in with a CAT cable, and sending out the bonded single to an ASUS router (it will be in Access Point mode so that the PI is creating the IP addresses for devices) on another CAT cable

bnhf, 2020-03-01:
Hard to beat the RPi for its combination of price and size.

There are only certain USB Ethernet adapters that will work on the RPi running OpenWRT or its variants like OpenMPTCProuter or ROOter. The one I've used, and had good success with is:

https://www.amazon.com/gp/product/B00LIW8TBG/

Other adapters based on the same chipset should work too, if this Linksys unit seems too pricey.

Your plan sounds good. Off-loading the AP duties to the Asus is a great strategy, and combined with tethering the hotspot devices to the RPi via USB, should give you better results.

darthclide, 2020-03-02:
> Your plan sounds good. Off-loading the AP duties to the Asus is a great strategy, and combined with tethering the hotspot devices to the RPi via USB, should give you better results.

Maybe I misunderstood my friend reedog, but I assumed my ASUS router wasn't going to be handing out IPs? The reason I can't just do a typical setup are the following things I hope you or others can help with.

1. I need the ability to use my expressvpn subscription to save room on my 5 device limit. I want to provide a VPN connection to all devices in the house, except for my PC. I know this is possible with Merlin, and a guy on NoRIP discord said this guide is where I should start: https://www.leowkahman.com/2016/06/19/conditional-multiple-openvpn-routing-hostname-ip/

But I have no idea if I can layer this into the Raspberry PI on top of openmptcprouter, or if I can find a way to put it on my ASUS router without double natting (which I believe I would need to do if I install Merlin).

2. After all is said and done, I will only have 2 physical connections on my PC. 1 Ethernet cable for my bonded line (AT&T, T-mobile, Verizon) and 1 Ethernet cable for my DSL (not bonded). I need the ability to route multiplayer game traffic over my DSL, but all other connections on my PC should use the bonded line. ForceBindIP will not work on my programs now because A. It has not been updated in years and B. Many programs spawn multiple processes, and this injects the DLL only into the first process and C. I can't use it on Steam games because you must start most of them through the Steam launcher which bypasses any shortcuts you create and I have no idea how to inject the DLL with steam launch properties.

#2 does not need to be complicated. Even a "gimmicky" solution is great since I don't need a robust setup with multiple options. I saw that Untangle's SD-WAN has the ability to detect packets based off an application name, but their service is like 300$ a year, and I would only be using their software for 1% of its abilities.

bnhf, 2020-03-02:
I believe you can do most, if not all, of what you want using OpenMPTCProuter. Reducing the number of "moving parts" will make things simpler and easier to manage.

- First, use your Asus router in AP mode. It won't be your DHCP server, and you won't be able to run an OpenVPN client on it -- it'll just be for WiFi, and act like a switch if you want use its Ethernet ports. Flash it to Merlin if you like, but the factory firmware will work fine too.

- Second, OMR uses a VPN to do it's magic, so all of your bonded traffic is using it.  But, if you want the anonymity of a public VPN server as well, then your could run an OpenVPN instance on your Raspberry Pi too. You should be able to download a .ovpn file from ExpressVPN, and then create an OpenVPN client configuration on OMR that uses it.

Finally, OMR-Bypass should be able to take care of any situations where you don't want certain traffic going via the bonded WAN. It provides numerous options for identifying, and re-routing traffic to a particular WAN, or any WAN (just not through your bonded tunnel).

https://i.postimg.cc/MKDzbYYR/screenshot-192.png

darthclide, 2020-03-03:
I think you misunderstood what I need. Or if you do, can you walk me through the steps to setup a VPN connection for a specific device (such as a phone or smart tv) and leave other devices untouched?

Also, your OMR bypass would only work if I have DSL bonded into the connection. I do not intend to do this because I will need every ounce of bits on my DSL line for gaming. And even if I could attach DSL to openmptcprouter without actually bonding it (aka, it just sits there doing nothing until I do an OMR bypass), the bigger problems is hunting down what things I need to bypass to access my multiplayer game. So unless you have a 100% surefire way to discover ALL connections being made by a game, and never running into crazy things like Warframe where I can play the game, but chat won't load, I have no choice but to leave DSL out of the mix. Which means I still need a way to route any multiplayer game over DSL, and everything else on my PC over the bonded connection

darthclide, 2020-03-04:
Hoping to hear back from anyone on this. I really want to move forward with buying the PI but I want to be sure this will do all I need with regards to giving ExpressVPN connections to specific devices + a way to know 100% of connections being made by multiplayer games for OMR-bypass. Or you can suggest something outside of openmptcprouter. I am willing to try anything to route my traffic depending on what the application is.

Can anyone recommend a good power source for the PI? I will need enough power to keep a ZTE Velocity 2, Verizon Jetpack 8800L, and a Motorola Moto E6 at 100% battery. This should be easy enough when it is setup at home (https://www.raspberrypi.org/products/type-c-power-supply/ I assume this is good enough?), but is there any battery pack out there that could sustain a 5500kbps Twitch stream for ~1-2 hours?

darthclide, 2020-03-06:
To be clear, I am not looking to go all out with a capture card + fancy camera setup. I intend on just using my phone to do live streams for a awhile just to get my feet wet. So literally the only thing the raspberry PI would be doing is sending a 4500-5500 kbps stream to Twitch.

darthclide, 2020-03-08:
Still waiting for someone to get back with me on my issues.

Ysurac, 2020-03-08:
For games, it's not possible. Even if sometimes you will not get a higher ping using OpenMPTCProuter (you can have a low ping to get the VPS, and you may have a more direct route from your VPS provider to games servers). 

For ExpressVPN, it's like if you use any modem/router. Maybe it can be installed on the router, but I don't know anything about this commercial service.

For the power source, if it's USB, you may need an additional powered USB hub, else I have no idea at all and it's not directly related to OpenMPTCProuter.

darthclide, 2020-03-08:
> For games, it's not possible. Even if sometimes you will not get a higher ping using OpenMPTCProuter (you can have a low ping to get the VPS, and you may have a more direct route from your VPS provider to games servers).

The issue isn't my ping though... I would be perfectly fine with a ping of 100ms in my games, if it meant I had 0 issues connecting to the servers. These games are blocking the IP because they think it is a VPN and I am trying to hack or something. Let's say I can magically find ALL domains/ports for a multiplayer game. Can you confirm that I can attach my DSL into OMR without bonding it (aka, not a single bit will be sent over it unless OMR-bypass has a rule for it?)

> For ExpressVPN, it's like if you use any modem/router. Maybe it can be installed on the router, but I don't know anything about this commercial service.

I can manually set it up with a password and openvpn config file. This is currently what I am doing on my ASUS router, but 1. It does not allow me to specify which devices connect to it (only possible with Merlin) and 2. I can't do this kind of traffic shaping once I turn my ASUS router into an Access Point after I setup OMR on the Raspberry PI. I need to know if your software is capable of routing specific local IP addresses (or MAC addresses) through an OpenVPN configuration.

If not, is it possible for me to somehow layer this on top of OMR? https://www.leowkahman.com/2016/06/19/conditional-multiple-openvpn-routing-hostname-ip/
 
> For the power source, if it's USB, you may need an additional powered USB hub, else I have no idea at all and it's not directly related to OpenMPTCProuter.
I had hoped that you had people come in here complaining about your software not working, and you determined that the issue is Raspberry PI not being as compatible as an x86 computer running it. Aka, I shouldn't even attempt setting this up on a RPI

Ysurac, 2020-03-08:
You can try to find a VPS that is not blocked. Yes you can attach a DSL without bonding, only disable MPTCP on the interface.

You can use omr-bypass to route any IP/MAC/network/... to any interface (a VPN create an interface).

RPI is a supported device, so it's working on it. I use a RPI4 for my own routing.

darthclide, 2020-03-08:
> You can use omr-bypass to route any IP/MAC/network/... to any interface (a VPN create an interface).

So can you give me a step by step tutorial on how I activate an ExpressVPN openvpn config file for a specific device inside my own network? And I would need this connection to use my bonded connection as well. (Aka, I don't want to create a VPN tunnel over AT&T and give it to the device. I want to create a VPN tunnel on top of my VPS server for a specific device)

darthclide, 2020-03-12:
My PI will arrive in 7 days. I hope that between now and then you can walk me through the steps required to do what I mentioned in the previous comment.

closed 2020-05-18 by Ysurac