Ysurac/openmptcprouter #854: pas d'adresse ip publique récupérée par shadowsocks

bosseb, 2020-02-24:
- OpenMPTCProuter version: 53.3
  - OpenMPTCProuter VPS version: 0.1010
  - OpenMPTCProuter platform: x86_64)

j'ai le message suivant : pas d'adresse ip publique récupérée par shadowsocks impossible de contacter le script d'administration du serveur (pas de réponse)

Ysurac, 2020-02-24:
Il va falloir plus d'infos.
Une capture d'écran de la page status.
Et la configuration complète sans les clefs, venant de Système->OpenMPTCProuter puis onglet "Voir tous les paramètres".

Ysurac, 2020-02-24:
Et sur le VPS, voir dans /var/log/daemon.log si il y a des erreurs.

bosseb, 2020-02-24:
[image]

bosseb, 2020-02-24:
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301,65011'
dscp.@classify[3].comment='OMR vpn'
dscp.@classify[4]=classify
dscp.@classify[4].direction='both'
dscp.@classify[4].proto='udp'
dscp.@classify[4].class='cs6'
dscp.@classify[4].dest_port='65001'
dscp.@classify[4].comment='OMR vpn'
dscp.@domains[0]=domains
[... 1515 lines cut ...]
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.channel='36'
wireless.radio1.hwmode='11a'
wireless.radio1.path='pci0000:00/0000:00:1c.5/0000:3a:00.0'
wireless.radio1.htmode='VHT80'
wireless.radio1.disabled='1'
wireless.default_radio1=wifi-iface
wireless.default_radio1.device='radio1'
wireless.default_radio1.network='lan'
wireless.default_radio1.mode='ap'
wireless.default_radio1.ssid='OpenWrt'
wireless.default_radio1.encryption='none'

bosseb, 2020-02-24:
je ne peux pas aller sur le VPS pour voir le log via filezilla quand il y a cette erreur en cours

Ysurac, 2020-02-24:
Ah ? Normalement tu devrais pouvoir en toute circonstance.
Dans la config je vois "shadowsocks-libev.sss0.key=''" tu as supprimé la clef ou elle est pas présente ?

bosseb, 2020-02-24:
je l'ai viré comme tu me l'as demandé

Ysurac, 2020-02-24:
ok
Il faudrait vraiment le ```/var/log/daemon.log``` sur le VPS du coup.
Et aussi un la table de routage donc soit ```ip r``` via SSH sur le routeur ou sur la page Status->Routes partie "Routes IPv4 actives".

bosseb, 2020-02-24:
[image]

Ysurac, 2020-02-24:
ok tout semble bon au niveau du routeur. Et il y a bien une route directe vers le VPS que tu devrais donc pouvoir toujours joindre.

bosseb, 2020-02-24:
merci mais impossible d'y acceder, je me connecte avec clé en ssh via filezilla sur le port 65222 et il refuse tout accès quand je suis avec cette erreur

Ysurac, 2020-02-24:
Et en SSH depuis le routeur ?

bosseb, 2020-02-24:
il refuse la connexion,

Ysurac, 2020-02-24:
Je peux plus aider la. Faut voir avec un KVM ou autre selon le fournisseur. Ou via une autre connexion si des fois ça fonctionne...

bosseb, 2020-02-24:
ok

bosseb, 2020-02-24:
merci

bosseb, 2020-02-24:
pour info j'étais auparavant chez scaleway , je viens de basculer sur firstheberg et le soucis est le même ...

Ysurac, 2020-02-24:
Tu peux me donner le log dans Etat->Journal système ?

bosseb, 2020-02-25:
Tue Feb 25 06:19:15 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:19:18 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:19:18 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:19:26 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:19:30 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:19:34 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:19:38 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:19:42 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:19:50 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:19:51 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:19:58 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:19:58 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:20:06 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:20:07 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:20:11 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:20:15 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:20:22 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:20:23 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:20:29 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:20:31 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:20:35 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:20:39 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:20:47 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:20:48 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:20:53 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:20:56 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:21:01 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:21:04 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:21:08 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:21:12 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:21:17 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:21:20 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:21:26 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:21:28 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:21:30 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:21:36 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:21:42 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:21:44 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:21:45 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:21:53 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:21:58 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:22:01 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:22:03 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:22:09 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:22:15 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:22:17 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:22:20 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:22:25 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:22:30 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:22:33 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:22:33 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:22:42 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:22:45 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:22:50 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:22:53 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:22:58 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:22:59 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:23:06 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:23:11 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:23:14 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:23:16 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:23:22 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:23:28 2020 user.notice OMR-VPS: Can't get token, try later
Tue Feb 25 06:23:31 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 06:23:37 2020 user.notice OMR-VPS: Can't get token, try later
[... cut ...]
Tue Feb 25 07:14:17 2020 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.5.1 dev wan1
Tue Feb 25 07:14:18 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Tue Feb 25 07:14:22 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 07:14:23 2020 daemon.err /usr/bin/ss-local[10099]: getpeername: Socket not connected
Tue Feb 25 07:14:26 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Tue Feb 25 07:14:26 2020 user.notice OMR-VPS: Can't get token, try later

bosseb, 2020-02-25:
j'ai pu trouver une solution, j'étais reparti de zero avec une config a blanc et cela plantait, j'ai donc réinjecté une vieille sauvegarde et mis à jour les données du VPS et cela a fonctionné, et ce qu'il y a moyen de comparer les fichiers de sauvegarde par rapport à la sauvegarde quand je suis reparti de zéro afin de voir les différences et de trouver l'origine du soucis ?

bosseb, 2020-03-06:
je reviens sur cette issue car le soucis se reproduit a chaque mise à jour d'OMR, j'arrive à m'en sortir à chaque fois mais j'aimerai juste savoir ce qui se produit pour que je sois obligé de mettre une vieille sauvegarde d'OMR pour que tout refonctionne, par avance merci :)

Ysurac, 2020-03-06:
Aucune idée, ce n'est pas un soucis que je rencontre.
Une revalidation du wizard peut corriger certains soucis.

closed 2020-09-14 by github-actions[bot]