Ysurac/openmptcprouter #855: Unable to connect VPS in ssh trought OMR

vinceducat, 2020-02-24:
## Expected Behavior

connect my vps in milkywan trought OMR ...

## Actual Behavior
doesnt work, work with direct connexion via wan 1 gateway not trought OMR

## Steps to Reproduce the Problem

  1. ssh -l user xxx.xxx.xxx.xxx -p 65222 
  2.
  3.
return connexion refused ....
## Specifications

  - OpenMPTCProuter version: 0.53.3
  - OpenMPTCProuter VPS version: 1010
  - OpenMPTCProuter platform: RPI3

vinceducat, 2020-02-24:
root@OpenMPTCProuter:~# ip r
default via 10.255.255.1 dev tun0
default via 192.168.0.254 dev wan1 metric 5
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2
45.xxx.xxx.xxx via 192.168.0.254 dev wan1 metric 5
127.0.0.0/8 dev lo proto static scope link metric 1
192.168.0.0/24 dev wan1 scope link metric 5

vinceducat, 2020-02-24:
works if i ssh vps on omr ...
root@OpenMPTCProuter:~# ssh -l xxx 45.xxx.xxx.xxx -p 65222
xxx@45.xxx.xxx.xxx's password:

vinceducat, 2020-02-24:
not trought a machine on my lan 

ssh -l xxx 45.xxx.xxx.xxx -p 65222
ssh: connect to host 45.xxx.xxx.xxx port 65222: Connection refused

vinceducat, 2020-02-25:
after changing from Mikywan VPS to Hetzner same problem .... so will try to put in place a new OMR fresh install....

vinceducat, 2020-02-25:
fresh install OMR > simple manual config >>>ok
restore config on OMR > problem 
so config in OMR is te problem but dont know why

so i reconfigure all my OMR 

ok for the moment

vinceducat, 2020-02-25:
@Ysurac sorry for the problem. do you want the backup with problem to try to solve why there is this pb ?

Ysurac, 2020-02-25:
Yes, I could try to see what setting was the problem.

vinceducat, 2020-02-25:
where can i send you the files ?

vinceducat, 2020-02-25:
voici les fichiers

vinceducat, 2020-02-26:
do you receive files?

Ysurac, 2020-02-26:
No. You can send them to contact@openmptcprouter.com

vinceducat, 2020-02-26:
done ;-)

vinceducat, 2020-02-29:
do u analyse the backup files and find the problem ?

i think this post could be go to solved state...

closed 2020-03-01 by vinceducat