Ysurac/openmptcprouter #860: Need avise | Mwan equivalent here

dur3x, 2020-02-26:
I'm currently using a mwan3 on another router in front of openmptcprouter. 
Why? It's because, in my 3 wan routers setup, for a part of my network I need to be able to not use the mptcp protocol and prefer to direct use wan1 or wan2 and never wan3 (wan3 is over the 4g and should be taken only for critical applications). So with mwan3 I'm able to say "ok for this source ip please use wan1; if wan1 is down then use wan2 and if wan2 is down we stop here and never user wan3"
Currently i'm checking to see if can do the same thing directly inside openmpctprouter by with the mechanism of the OMRBypaas it looks that I can only say "ok for this source ip please use wan1" (or wan2) but I cannot make a preference/failover wan1 > wan2.

Not sure my explanation is well clear but if your understand me, do you have any suggestion (to be able to make this preference/failover with wan1/wan2 in a bypass style) ?
Thanks in advance (and again thanks a lot for all your work on openmptcprouter, this project is just amazing ;-))

Ysurac, 2020-02-26:
When you set Default interface in OMR-ByPass this use the interface defined as master, when the interface is down this use another interface. You can't set preference, but it's a failover.

dur3x, 2020-02-26:
Thank you it's clear enough; I will reopen(or create) this issue if needed

closed 2020-02-26 by dur3x