Ysurac/openmptcprouter #861: XHR request timeout

Archana7700, 2020-02-27:
whenever i m configuring or doing changes in configurations not only while configuring but even if
i use it for long time it get corrupt again and  again all of sudden i m facing this issue every time. 
I have to flash it to use it again or else it gets hangs again. It takes longer time to open the page or to reload, i have tried rebooting it but it didn't workout.
Its gives an error message as:

"/usr/lib/lua/luci/dispatcher.lua:388: /etc/config/luci seems to be corrupt, unable to find section 'main'"

How to resolve this issue

Ysurac, 2020-02-27:
Via ssh on the router, check log using ```logread``` about rpcd (because I would like to know why it crash)
Then ```/etc/init.d/rpcd restart``` and this should work.

Archana7700, 2020-02-27:
Hi there,
There is no issue in rpcd checked logs, it is working fine without any error.
Logread after /etc/init.d/rpcd restart:
Thu Feb 27 07:24:48 2020 daemon.err /usr/bin/ss-local[11655]: getpeername: Socket not connected
Thu Feb 27 07:24:49 2020 kern.err kernel: [ 2981.191290] ieee80211 phy0: brcmf_run_escan: error (-110)
Thu Feb 27 07:24:49 2020 kern.err kernel: [ 2981.196713] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Thu Feb 27 07:24:49 2020 daemon.notice wpa_supplicant[4289]: wlan0: CTRL-EVENT-SCAN-FAILED ret=-110 retry=1
Thu Feb 27 07:24:53 2020 kern.err kernel: [ 2984.711295] ieee80211 phy0: brcmf_run_escan: error (-110)
Thu Feb 27 07:24:53 2020 kern.err kernel: [ 2984.716717] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Thu Feb 27 07:24:53 2020 daemon.notice wpa_supplicant[4289]: wlan0: CTRL-EVENT-SCAN-FAILED ret=-110 retry=1
Thu Feb 27 07:24:56 2020 daemon.err /usr/bin/ss-local[11655]: getpeername: Socket not connected
Thu Feb 27 07:24:56 2020 kern.err kernel: [ 2988.231293] ieee80211 phy0: brcmf_run_escan: error (-110)
Thu Feb 27 07:24:56 2020 kern.err kernel: [ 2988.236716] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Thu Feb 27 07:24:56 2020 daemon.notice wpa_supplicant[4289]: wlan0: CTRL-EVENT-SCAN-FAILED ret=-110 retry=1
Thu Feb 27 07:25:00 2020 kern.err kernel: [ 2991.751463] ieee80211 phy0: brcmf_run_escan: error (-110)
Thu Feb 27 07:25:00 2020 kern.err kernel: [ 2991.756887] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Thu Feb 27 07:25:00 2020 daemon.notice wpa_supplicant[4289]: wlan0: CTRL-EVENT-SCAN-FAILED ret=-110 retry=1
Thu Feb 27 07:25:02 2020 daemon.err /usr/bin/ss-local[11655]: getpeername: Socket not connected
Thu Feb 27 07:25:03 2020 kern.err kernel: [ 2995.271292] ieee80211 phy0: brcmf_run_escan: error (-110)
Thu Feb 27 07:25:03 2020 kern.err kernel: [ 2995.276714] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
Thu Feb 27 07:25:03 2020 daemon.notice wpa_supplicant[4289]: wlan0: CTRL-EVENT-SCAN-FAILED ret=-110 retry=1

root@OpenMPTCProuter:~# ps |grep rpcd
25163 root      2100 S    /sbin/rpcd -s /var/run/ubus.sock -t 90

need your further support!

Ysurac, 2020-02-27:
Reboot, if no success check file ```/etc/config/luci```.

Archana7700, 2020-02-27:
With reboot it is not working
Contents: /etc/config/luci
config core 'main'
	option lang 'auto'
	option resourcebase '/luci-static/resources'
	option mediaurlbase '/luci-static/openmptcprouter'

config extern 'flash_keep'
	option uci '/etc/config/'
	option dropbear '/etc/dropbear/'
	option openvpn '/etc/openvpn/'
	option passwd '/etc/passwd'
	option opkg '/etc/opkg.conf'
	option firewall '/etc/firewall.user'
	option uploads '/lib/uci/upload/'

config internal 'languages'
	option ca 'Català (Catalan)'
	option cs 'Čeština (Czech)'
	option de 'Deutsch (German)'
	option el 'Ελληνικά (Greek)'
	option en 'English'
	option es 'Español (Spanish)'
	option fr 'Français (French)'
	option he 'עִבְרִית (Hebrew)'
	option hu 'Magyar (Hungarian)'
	option it 'Italiano (Italian)'
	option ja '日本語 (Japanese)'
	option ko '한국어 (Korean)'
	option ms 'Bahasa Melayu (Malay)'
	option no 'Norsk (Norwegian)'
	option pl 'Polski (Polish)'
	option pt 'Português (Portuguese)'
	option pt_br 'Português do Brasil (Brazialian Portuguese)'
	option ro 'Română (Romanian)'
	option ru 'Русский (Russian)'
	option sk 'Slovenčina (Slovak)'
	option sv 'Svenska (Swedish)'
	option tr 'Türkçe (Turkish)'
	option uk 'Українська (Ukrainian)'
	option vi 'Tiếng Việt (Vietnamese)'
	option zh_cn '中文 (Chinese)'
	option zh_tw '臺灣華語 (Taiwanese)'

config internal 'sauth'
	option sessionpath '/tmp/luci-sessions'
	option sessiontime '3600'

config internal 'ccache'
	option enable '1'

config internal 'themes'
	option OpenMPTCProuter '/luci-static/openmptcprouter'

config internal 'apply'
	option rollback '30'
	option holdoff '4'
	option timeout '5'
	option display '1.5'

config internal 'diag'
	option ping 'openmptcprouter.com'
	option dns 'openmptcprouter.com'
	option route 'openmptcprouter.com'
	option iperf3 'ping-ams1.online.net'
	option getip 'ip.openmptcprouter.com'

Olli69, 2020-03-16:
Hi, 
I have had XHR request timeout with MacOS Chrome, but not that much with Safari. Especially with backup and restore. There could be something browser dependant. What is the preferred / most tested browser for configuration?

Ysurac, 2020-03-16:
Any, but you have to clear browser cache.

rajava2010, 2020-04-06:
XHR Timeout happens in Chrome on OS X , but Safari works like a charm, you really need to fix the Chrome issues as majority of users uses Chrome.

Ysurac, 2020-04-07:
Which issue ? I only see browser cache issue about that.

closed 2020-09-14 by github-actions[bot]