Ysurac/openmptcprouter #862: soucis wifi  vps version 0.1011

bosseb, 2020-02-28:
- OpenMPTCProuter version: 53.3
  - OpenMPTCProuter VPS version:0.1011
  - OpenMPTCProuter platform:x86_64

Depuis le passage sur 0.1011 au niveau VPS ma connection est instable sur le wifi principalement

ci dessous le journal system

Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty13, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty41, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty41
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty41, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyS1, subsystem=tty, sysfspath=/sys/devices/platform/serial8250/tty/ttyS1
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyS1, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty31, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty31
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty31, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty21, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty21
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty21, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty5, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty5
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty5, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty11, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty11
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty11, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty58, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty58
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty58, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty3, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty3
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty3, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty48, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty48
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty48, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyS8, subsystem=tty, sysfspath=/sys/devices/platform/serial8250/tty/ttyS8
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyS8, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty38, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty38
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty38, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=sit0, subsystem=net, sysfspath=/sys/devices/virtual/net/sit0
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=sit0, subsystem=net
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ifb0, subsystem=net, sysfspath=/sys/devices/virtual/net/ifb0
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ifb0, subsystem=net
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ifb1, subsystem=net, sysfspath=/sys/devices/virtual/net/ifb1
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ifb1, subsystem=net
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI0, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI0
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI0, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI1, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI1
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI1, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI2, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI2
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI2, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI3, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI3
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI3, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI4, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI4
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI4, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI5, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI5
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI5, subsystem=tty
Fri Feb 28 22:09:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
[... 494 lines cut ...]
Fri Feb 28 22:12:10 2020 kern.warn kernel: [  180.284452]  genl_rcv_msg+0x43/0x90
Fri Feb 28 22:12:10 2020 kern.warn kernel: [  180.286290]  ? genl_family_rcv_msg+0x350/0x350
Fri Feb 28 22:12:10 2020 kern.warn kernel: [  180.288230]  netlink_rcv_skb+0x3f/0x110
Fri Feb 28 22:12:10 2020 kern.warn kernel: [  180.290047]  genl_rcv+0x23/0x40
Fri Feb 28 22:12:10 2020 kern.warn kernel: [  180.291877]  netlink_unicast+0x16b/0x1e0
Fri Feb 28 22:12:10 2020 kern.warn kernel: [  180.293707]  netlink_sendmsg+0x1c0/0x3c0
[... cut ...]
Fri Feb 28 22:12:10 2020 kern.warn kernel: [  180.321254] RDX: 0000000000000000 RSI: 00007ffebc5329d8 RDI: 0000000000000007
Fri Feb 28 22:12:10 2020 kern.warn kernel: [  180.323147] RBP: 000000000000002e R08: 0000000000000000 R09: 0000000000000000
Fri Feb 28 22:12:10 2020 kern.warn kernel: [  180.325036] R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000000000
Fri Feb 28 22:12:10 2020 kern.warn kernel: [  180.326931] R13: 00007ffebc538f6c R14: 0000000000000001 R15: 0000000000dd2730
Fri Feb 28 22:12:10 2020 kern.warn kernel: [  180.328827] ---[ end trace b4338c2ef9b5fd2e ]---

Ysurac, 2020-02-28:
Vu l'erreur ce n'est pas lié à la mise à jour du VPS. En tout cas pas directement.
Quel est le VPN utilisé (même si ça ne semble pas lié non plus) ?

bosseb, 2020-02-28:
glorytun tcp

bosseb, 2020-02-29:
une idée ?

Ysurac, 2020-02-29:
essaye la v0.54rc5 https://github.com/Ysurac/openmptcprouter/issues/864

bosseb, 2020-02-29:
ça va déjà mieux après quelques minutes, c sympa le niveau de réception dans le wan 4g :)

merci bien

Ysurac, 2020-02-29:
tant mieux que ça fonctionne, c'est quoi comme dongle/router ?

bosseb, 2020-02-29:
HUAWEI B715s-23c

Ysurac, 2020-02-29:
ok merci, content de savoir que ça fonctionne sur ce modèle aussi :)

bosseb, 2020-02-29:
par contre je viens de me rendre compte que mon débit est en chute libre ... je suis passé de 180 mega 30 mega .... que ce soit en wifi ou en filaire

Ysurac, 2020-02-29:
oui il y a un soucis. Je vais corriger ça.

Ysurac, 2020-02-29:
Voila, ça sera normalement mieux dans la v0.54rc6 qui est en cours de compilation.

bosseb, 2020-02-29:
super merci beaucoup je regarde ça
bonne soirée

bosseb, 2020-03-01:
Je viens de mettre la 0.54rc7 mon débit est revenu

merci

closed 2020-09-14 by github-actions[bot]