Ysurac/openmptcprouter #863: Router Install Question

rustequal, 2020-02-28:
Hello Yannick,

I have a question about installing the router. The [documentation](https://github.com/Ysurac/openmptcprouter/wiki/Router-install) contains the phrase:

**You don't need ethernet adapters, only a switch is enough. Modems doesn't need to be directly connected to RPI ethernet port.**

What is that mean ? I tested various configurations with a single physical eth0 interface. I mean the configurations in which LAN and WAN interfaces use a single physical interface eth0. And in all cases, the iptables rules of the INPUT, FORWARD, etc. chains mixed up and interfered with each other. And on the "Network overview" tab, there is a warning that the interface is duplicated.

bnhf, 2020-02-29:
I'm a new user of OMR myself, but have spent quite a bit of time experimenting with both the RPi and x86 versions. The RPi version comes with two macvlan interfaces enabled on first boot, the x86 version does not.

The phrase you're quoting from the documentation assumes you're using macvlan, though I never considering using a single Ethernet port for LAN and WAN traffic -- I've always had a minimum of two ports, one for LAN and the other for all of my WAN interfaces.

On the x86 version you'd need to create those virtual adapters first under Network - Macvlan, and then you should find them available for assignment using the OMR wizard. Might work to have them all on a single physical adapter, but I haven't tried it.

In my case, I settled on a Qotom x86 box for my production WAN bonding router, and I'm using VLANs rather than macvlan.  I already had a network of gateways on the WAN side of my primary router, all configured for VLAN use. OMR absolutely rocks, I've been getting great, near linear aggregation, with all the robustness you'd expect form this type of setup.

Have fun!

Ysurac, 2020-02-29:
True, on the RPI MacVLAN is used, so this create some "virtual" interfaces and no problems with iptables or anything else.

rustequal, 2020-02-29:
Thanks for the hint ;)

closed 2020-02-29 by rustequal