Ysurac/openmptcprouter #864: OpenMPTCProuter v0.54 Release Candidate

Ysurac, 2020-02-29:
In this new release:
* Signal power for some mobile router/dongle
* Fix disconnect when master down
* Many fixes and small changes

This release need server part v0.1011

You can download images here: https://www.openmptcprouter.com/release/develop/4.19/

Latest version: v0.54rc5

vinceducat, 2020-02-29:
hello i m in OMR 0.53.3 stable and VPS 1010 must have  to go in 0.54rc5 and 1011 ?

vinceducat, 2020-02-29:
not a godd idea i try update OMR and VPS

now in state view i have on OMR "le vpn ne r2pond pas" :-(

Ysurac, 2020-02-29:
What is the VPN used ?

vinceducat, 2020-02-29:
I dont change any vpn so i think is the default vpn glorytun... where can i find this info...

Ysurac, 2020-02-29:
In System->OpenMPTCProuter and advanced setting checkbox.

vinceducat, 2020-03-01:
go to 0.54 rc7 and 1012 all seem ok now

vinceducat, 2020-03-01:
VPN is OPen VPN...

Ysurac, 2020-03-01:
No. OpenVPN is the slowest available VPN.

vinceducat, 2020-03-01:
So glorytun tcp?

Ysurac, 2020-03-01:
Yes or glorytun UDP. You can make tests. The VPN is used only for anything that is not TCP.

vinceducat, 2020-03-01:
ok thanks

vinceducat, 2020-03-01:
ok i change vpn to glorytun tcp seems to be ok to...

Slivikkus, 2020-03-06:
Pour l'instant cette version fonctionne parfaitement avec moi, je remettrais un message pour confirmer que ça tient bien dans le temps :)

Slivikkus, 2020-03-07:
[image]
C'est le seul problème j'ai eu pour l'instant, sois côté VPS ou VPN down

Ysurac, 2020-03-07:
You have some high load on both router and VPS. What is the usage ? some p2p or really slow arch ?

Slivikkus, 2020-03-08:
I use several vps for redundancy, however the connection looks more stable than on the old version

Ysurac, 2020-03-10:
v0.54 is released.

closed 2020-03-10 by Ysurac