Ysurac/openmptcprouter #865: VPN not respond

jgreenhalgh51, 2020-03-01:
hi
I have a vps (1vcpu and 512mo of RAM) at ionos
But my router said alwaus that my vpn not respond.

I have tested all version 0.53.3 and 0.54rc7
but there are not good config who work.

vinceducat, 2020-03-01:
do you update vps to 1012 ?

jgreenhalgh51, 2020-03-01:
yes

jgreenhalgh51, 2020-03-01:
the vpn is not stable when i charge 
example with nperf

vinceducat, 2020-03-01:
i let Ysurac make advice because i dont know what is the best VPN to use... ;-)

jgreenhalgh51, 2020-03-01:
ok

Ysurac, 2020-03-01:
For RC problems, use RC topic: https://github.com/Ysurac/openmptcprouter/issues/864
(and it's a know problem, SQM should fix that, else I'm making some tests)

jgreenhalgh51, 2020-03-01:
i have the same problem with 0.53.3
if you want i can sent you a mail for log of VPS and Router

vinceducat, 2020-03-01:
i think ysurac need to known witch you use...

jgreenhalgh51, 2020-03-01:
I have the same problem that issue 866.

Ysurac, 2020-03-01:
So you always have the issue ? You said it's when you charge...
And what is the VPN used ?
SQM is used ?
What is the CPU/memory usage on the VPS ?

jgreenhalgh51, 2020-03-01:
I have tested glorytun tcp and udp -> nok 
Sqm is not use.
On vps cpu 10% memory 5%

Ysurac, 2020-03-01:
So always down or only on charge ?

jgreenhalgh51, 2020-03-01:
After a charge i reboot vps and routeur

jgreenhalgh51, 2020-03-01:
Same without charge the vpn is instable

Ysurac, 2020-03-01:
You should check in Status->System log.

jgreenhalgh51, 2020-03-01:
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=429, dest=[ff05::c]:1900): sendto: Address not available
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=488, dest=[ff05::c]:1900): sendto: Address not available
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=429, dest=[ff05::c]:1900): sendto: Address not available
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=492, dest=[ff05::c]:1900): sendto: Address not available
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto failed to send 13 packets
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=484, dest=[ff0e::c]:1900): sendto: Address not available
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=484, dest=[ff0e::c]:1900): sendto: Address not available
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=500, dest=[ff0e::c]:1900): sendto: Address not available
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=496, dest=[ff0e::c]:1900): sendto: Address not available
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=484, dest=[ff0e::c]:1900): sendto: Address not available
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=482, dest=[ff0e::c]:1900): sendto: Address not available
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=429, dest=[ff0e::c]:1900): sendto: Address not available
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=468, dest=[ff0e::c]:1900): sendto: Address not available
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=429, dest=[ff0e::c]:1900): sendto: Address not available
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=488, dest=[ff0e::c]:1900): sendto: Address not available
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=429, dest=[ff0e::c]:1900): sendto: Address not available
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=492, dest=[ff0e::c]:1900): sendto: Address not available
Sun Mar  1 21:22:23 2020 daemon.err miniupnpd[17626]: try_sendto failed to send 12 packets
Sun Mar  1 21:22:23 2020 daemon.warn miniupnpd[17626]: HTTP Connection from 192.168.100.253 closed unexpectedly
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
Sun Mar  1 21:22:53 2020 daemon.err miniupnpd[17626]: sendto(udp_notify=8, [fd2e:484:a4f6::1]): Address not available
[... cut ...]
Sun Mar  1 21:25:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=500, dest=[ff0e::c]:1900): sendto: Address not available
Sun Mar  1 21:25:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=496, dest=[ff0e::c]:1900): sendto: Address not available
Sun Mar  1 21:25:23 2020 daemon.err miniupnpd[17626]: try_sendto(sock=8, len=484, dest=[ff0e::c]:1900): sendto: Address not available
Sun Mar  1 21:25:23 2020 daemon.err miniupnpd[17626]: try_sendto failed to send 5 packets
Sun Mar  1 21:25:28 2020 user.notice post-tracking-post-tracking: Change master interface from wan1 (82 ms) to wan2 (51 ms)

jgreenhalgh51, 2020-03-01:
on vps daemon.log

r  1 20:54:31 localhost shorewall[113088]:    WARNING: The LOAD_HELPERS_ONLY configuration option is no longer supported /etc/shorewall/shorewall.conf (line 191)
Mar  1 20:54:31 localhost shorewall[113088]:    WARNING: The MAPOLDACTIONS configuration option is no longer supported /etc/shorewall/shorewall.conf (line 199)
Mar  1 20:54:31 localhost shorewall[113088]:    WARNING: The MODULE_SUFFIX configuration option is no longer supported /etc/shorewall/shorewall.conf (line 205)
Mar  1 20:54:31 localhost shorewall[113088]: Loading Modules...
Mar  1 20:54:32 localhost shorewall[113088]: Compiling /etc/shorewall/zones...
Mar  1 20:54:32 localhost shorewall[113088]: Compiling /etc/shorewall/interfaces...
Mar  1 20:54:32 localhost shorewall[113088]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 18)
Mar  1 20:54:32 localhost shorewall[113088]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 19)
Mar  1 20:54:32 localhost shorewall[113088]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 20)
Mar  1 20:54:32 localhost shorewall[113088]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 21)
Mar  1 20:54:32 localhost shorewall[113088]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 22)
Mar  1 20:54:33 localhost shorewall[113088]: Determining Hosts in Zones...
Mar  1 20:54:33 localhost shorewall[113088]: Locating Action Files...
Mar  1 20:54:33 localhost shorewall[113088]: Compiling /etc/shorewall/policy...
Mar  1 20:54:34 localhost shorewall[113088]: Adding Anti-smurf Rules
Mar  1 20:54:34 localhost shorewall[113088]: Adding rules for DHCP
Mar  1 20:54:34 localhost shorewall[113088]: Compiling TCP Flags filtering...
Mar  1 20:54:34 localhost shorewall[113088]: Compiling Kernel Route Filtering...
Mar  1 20:54:34 localhost shorewall[113088]: Compiling Martian Logging...
Mar  1 20:54:34 localhost shorewall[113088]: Compiling Accept Source Routing...
Mar  1 20:54:34 localhost shorewall[113088]: Compiling /etc/shorewall/tcinterfaces...
Mar  1 20:54:35 localhost shorewall[113088]: Compiling /etc/shorewall/snat...
Mar  1 20:54:35 localhost shorewall[113088]: Compiling MAC Filtration -- Phase 1...
Mar  1 20:54:35 localhost shorewall[113088]: Compiling /etc/shorewall/rules...
Mar  1 20:54:43 localhost shorewall[113088]: Compiling /etc/shorewall/conntrack...
Mar  1 20:54:43 localhost shorewall[113088]: Compiling MAC Filtration -- Phase 2...
Mar  1 20:54:43 localhost shorewall[113088]: Applying Policies...
Mar  1 20:54:43 localhost shorewall[113088]: Generating Rule Matrix...
Mar  1 20:54:44 localhost shorewall[113088]: Creating iptables-restore input...
Mar  1 20:54:45 localhost shorewall[113088]: Compiling /etc/shorewall/stoppedrules...
Mar  1 20:54:45 localhost shorewall[113088]: Shorewall configuration compiled to /var/lib/shorewall/.reload
Mar  1 20:54:47 localhost shorewall[113088]: Reloading Shorewall....
Mar  1 20:54:47 localhost shorewall[113088]: Initializing...
Mar  1 20:54:48 localhost shorewall[113088]: Setting up Route Filtering...
Mar  1 20:54:48 localhost shorewall[113088]: Setting up Martian Logging...
Mar  1 20:54:48 localhost shorewall[113088]: Setting up Accept Source Routing...
Mar  1 20:54:48 localhost shorewall[113088]: Setting up Traffic Control...
Mar  1 20:54:48 localhost shorewall[113088]: Preparing iptables-restore input...
Mar  1 20:54:48 localhost shorewall[113088]: Running /sbin/iptables-restore --wait 60...
Mar  1 20:54:48 localhost shorewall[113088]: IPv4 Forwarding Enabled
Mar  1 20:54:48 localhost shorewall[113088]: done.
Mar  1 20:54:48 localhost systemd[1]: Reloading Shorewall IPv4 firewall.
Mar  1 20:54:48 localhost shorewall[113503]: Compiling using Shorewall 5.2.3.2...
Mar  1 20:55:00 localhost shorewall[113503]: Processing /etc/shorewall/params ...
Mar  1 20:55:00 localhost shorewall[113503]: Processing /etc/shorewall/shorewall.conf...
Mar  1 20:55:00 localhost shorewall[113503]:    WARNING: The CHAIN_SCRIPTS configuration option is no longer supported /etc/shorewall/shorewall.conf (line 147)
Mar  1 20:55:00 localhost shorewall[113503]:    WARNING: The INLINE_MATCHES configuration option is no longer supported /etc/shorewall/shorewall.conf (line 183)
Mar  1 20:55:00 localhost shorewall[113503]:    WARNING: The LOAD_HELPERS_ONLY configuration option is no longer supported /etc/shorewall/shorewall.conf (line 191)
Mar  1 20:55:00 localhost shorewall[113503]:    WARNING: The MAPOLDACTIONS configuration option is no longer supported /etc/shorewall/shorewall.conf (line 199)
Mar  1 20:55:00 localhost shorewall[113503]:    WARNING: The MODULE_SUFFIX configuration option is no longer supported /etc/shorewall/shorewall.conf (line 205)
Mar  1 20:55:00 localhost shorewall[113503]: Loading Modules...
Mar  1 20:55:02 localhost shorewall[113503]: Compiling /etc/shorewall/zones...
Mar  1 20:55:02 localhost shorewall[113503]: Compiling /etc/shorewall/interfaces...
Mar  1 20:55:02 localhost shorewall[113503]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 18)
[... cut ...]
Mar  1 20:55:56 localhost shorewall[114282]: Setting up Accept Source Routing...
Mar  1 20:55:56 localhost shorewall[114282]: Setting up Traffic Control...
Mar  1 20:55:56 localhost shorewall[114282]: Preparing iptables-restore input...
Mar  1 20:55:56 localhost shorewall[114282]: Running /sbin/iptables-restore --wait 60...
Mar  1 20:55:56 localhost shorewall[114282]: IPv4 Forwarding Enabled
Mar  1 20:55:56 localhost shorewall[114282]: done.
Mar  1 20:55:56 localhost systemd[1]: Reloaded Shorewall IPv4 firewall.

jgreenhalgh51, 2020-03-01:
[image]

Ysurac, 2020-03-01:
You use upnp ? The log is full of errors.

jgreenhalgh51, 2020-03-01:
yes for xbox an ps4

jgreenhalgh51, 2020-03-01:
i haven't change config just active upnp

jgreenhalgh51, 2020-03-01:
i have desactivated upnp but not change the problem

jgreenhalgh51, 2020-03-01:
[image]

jgreenhalgh51, 2020-03-01:
i have desactivated upnp

un Mar  1 21:31:29 2020 daemon.err miniupnpd[17626]: Failed to broadcast good-bye notifications
Sun Mar  1 21:32:26 2020 user.notice MPTCP: Flush main table
Sun Mar  1 21:32:27 2020 user.notice MPTCP: Add route 192.168.250.0/24 via 192.168.100.250 dev eth0
Sun Mar  1 21:32:27 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.100.253 dev wan1
Sun Mar  1 21:32:27 2020 user.notice post-tracking-post-tracking: Set server ionos (212.227.164.208) route via 192.168.100.253 metric 3
Sun Mar  1 21:32:28 2020 user.notice post-tracking-post-tracking: Set server ionos (212.227.164.208) route via 192.168.100.254 metric 4
Sun Mar  1 21:32:28 2020 daemon.notice netifd: Network device 'tun0' link is down
Sun Mar  1 21:32:28 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Sun Mar  1 21:32:28 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Sun Mar  1 21:32:28 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Sun Mar  1 21:32:28 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Sun Mar  1 21:32:28 2020 daemon.err omr-tracker[15797]: no device
Sun Mar  1 21:32:28 2020 daemon.info glorytun-udp: starting glorytun-udp vpn instance vpn
Sun Mar  1 21:32:28 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.254.1 dev tun0
Sun Mar  1 21:32:28 2020 daemon.notice netifd: Network device 'tun0' link is up
Sun Mar  1 21:32:28 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Sun Mar  1 21:32:28 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Sun Mar  1 21:32:28 2020 daemon.notice netifd: Interface 'omrvpn' is now up
Sun Mar  1 21:32:28 2020 daemon.err glorytun-udp[13201]: running on device tun0 as pid 13201
Sun Mar  1 21:32:28 2020 user.notice SQM: Stopping SQM on tun0
Sun Mar  1 21:32:28 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev tun0 ingress
Sun Mar  1 21:32:28 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Sun Mar  1 21:32:28 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev tun0 root
Sun Mar  1 21:32:28 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Sun Mar  1 21:32:28 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev ifb4tun0 root
Sun Mar  1 21:32:28 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Sun Mar  1 21:32:28 2020 user.notice omr-tracker: Launching...
Sun Mar  1 21:32:29 2020 user.notice ModemManager: hotplug: remove network interface ifb4tun0: event processed
Sun Mar  1 21:32:29 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=ifb4tun0, subsystem=net
Sun Mar  1 21:32:29 2020 user.notice omr-tracker: Launched
Sun Mar  1 21:32:29 2020 user.notice SQM: Starting SQM script: simple.qos on tun0, in: 0 Kbps, out: 0 Kbps
Sun Mar  1 21:32:29 2020 user.notice SQM: Using generic sqm_start_default function.
Sun Mar  1 21:32:29 2020 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
Sun Mar  1 21:32:29 2020 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Sun Mar  1 21:32:29 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Mar  1 21:32:29 2020 daemon.warn [9161]: <warn>  Invalid sysfs path read for net/TMP_IFB_4_SQM
Sun Mar  1 21:32:29 2020 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Sun Mar  1 21:32:29 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Sun Mar  1 21:32:30 2020 daemon.info vnstatd[896]: Info: Interface "ifb4tun0" disabled.
Sun Mar  1 21:32:30 2020 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
Sun Mar  1 21:32:30 2020 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Sun Mar  1 21:32:30 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Mar  1 21:32:30 2020 daemon.warn [9161]: <warn>  Invalid sysfs path read for net/TMP_IFB_4_SQM
Sun Mar  1 21:32:30 2020 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Sun Mar  1 21:32:30 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Sun Mar  1 21:32:31 2020 user.notice ModemManager: hotplug: add network interface ifb4tun0: event processed
Sun Mar  1 21:32:31 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ifb4tun0, subsystem=net
Sun Mar  1 21:32:31 2020 user.notice SQM: simple.qos was started on tun0 successfully
Sun Mar  1 21:32:31 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Mar  1 21:32:31 2020 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Sun Mar  1 21:32:31 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sun Mar  1 21:32:35 2020 daemon.info vnstatd[896]: Info: Interface "ifb4tun0" enabled.
Sun Mar  1 21:32:36 2020 user.notice omr-bypass: Restart dnsmasq...
Sun Mar  1 21:32:37 2020 daemon.info dnsmasq[30453]: exiting on receipt of SIGTERM
[... cut ...]
Sun Mar  1 21:32:37 2020 daemon.info dnsmasq[16015]: using nameserver 192.168.100.254#53 for domain free.fr
Sun Mar  1 21:32:37 2020 daemon.info dnsmasq[16015]: using nameserver 192.168.100.253#53 for domain orange.fr
Sun Mar  1 21:32:37 2020 daemon.info dnsmasq[16015]: using nameserver 127.0.0.1#5353
Sun Mar  1 21:32:37 2020 daemon.info dnsmasq[16015]: using local addresses only for domain lan
Sun Mar  1 21:32:37 2020 daemon.info dnsmasq[16015]: read /etc/hosts - 4 addresses
Sun Mar  1 21:32:37 2020 daemon.info dnsmasq[16015]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses

jgreenhalgh51, 2020-03-01:
Config 
hcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].domain='datahome.fr'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/orange.fr/192.168.100.253' '/free.fr/192.168.100.254'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.cot/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/www.jeuxvideo.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/www.inmac-wstore.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/www.3ds-paradise.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/www.edf.fr/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/particulier.edf.fr/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/jvc.gg/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/image.jeuxvideo.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/www.googletagmanager.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/Freebox.fr/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/Orange.fr/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/www.boulanger.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/smtp.orange.fr/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/imap.orange.fr/omr_dst_bypass_wan1,omr6_dst_bypass_wan1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.ignore='1'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.gw_wan1=tag
dhcp.gw_wan1.dhcp_option='3,192.168.100.253' '6,81.253.149.14,80.10.246.7' '15,orange.fr' '125,00:00:0d:e9:24:04:06:EC:BE:DD:D8:DC:2F:05:0f:4c:4b:31:39:32:31:35:44:50:39:39:32:33:36:36:06:09:4c:69:76:65:62:6f:78:20:34'
dhcp.gw_wan2=tag
dhcp.gw_wan2.dhcp_option='3,192.168.100.254' '6,192.168.100.254' '15,free.fr'
dhcp.@host[0]=host
dhcp.@host[0].name='LiveboxTV'
dhcp.@host[0].ip='192.168.100.160'
dhcp.@host[0].mac='EC:BE:DD:D8:DC:2F'
dhcp.@host[0].tag='gw_wan1'
dhcp.@host[1]=host
dhcp.@host[1].name='Freeboxmini'
dhcp.@host[1].dns='1'
dhcp.@host[1].mac='E4:9E:12:2E:63:8A'
dhcp.@host[1].ip='192.168.100.141'
dhcp.@host[1].tag='gw_wan2'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@domains[0]=domains
dscp.@domains[0].name='googlevideo.com'
dscp.@domains[0].class='cs4'
dscp.@domains[0].comment='Youtube'
dscp.@domains[1]=domains
[... cut ...]
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth0' 'tun0' 'tun1' 'wan1' 'wan2' 'wan3'

jgreenhalgh51, 2020-03-01:
Mar  1 21:33:03 localhost systemd[1]: Reloading Shorewall IPv4 firewall.
Mar  1 21:33:03 localhost shorewall[123548]: Compiling using Shorewall 5.2.3.2...
Mar  1 21:33:24 localhost shorewall[123548]: Processing /etc/shorewall/params ...
Mar  1 21:33:24 localhost shorewall[123548]: Processing /etc/shorewall/shorewall.conf...
Mar  1 21:33:24 localhost shorewall[123548]:    WARNING: The CHAIN_SCRIPTS configuration option is no longer supported /etc/shorewall/shorewall.conf (line 147)
Mar  1 21:33:24 localhost shorewall[123548]:    WARNING: The INLINE_MATCHES configuration option is no longer supported /etc/shorewall/shorewall.conf (line 183)
Mar  1 21:33:24 localhost shorewall[123548]:    WARNING: The LOAD_HELPERS_ONLY configuration option is no longer supported /etc/shorewall/shorewall.conf (line 191)
Mar  1 21:33:24 localhost shorewall[123548]:    WARNING: The MAPOLDACTIONS configuration option is no longer supported /etc/shorewall/shorewall.conf (line 199)
Mar  1 21:33:24 localhost shorewall[123548]:    WARNING: The MODULE_SUFFIX configuration option is no longer supported /etc/shorewall/shorewall.conf (line 205)
Mar  1 21:33:24 localhost shorewall[123548]: Loading Modules...
Mar  1 21:33:27 localhost shorewall[123548]: Compiling /etc/shorewall/zones...
Mar  1 21:33:27 localhost shorewall[123548]: Compiling /etc/shorewall/interfaces...
Mar  1 21:33:27 localhost shorewall[123548]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 18)
Mar  1 21:33:27 localhost shorewall[123548]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 19)
Mar  1 21:33:27 localhost shorewall[123548]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 20)
Mar  1 21:33:27 localhost shorewall[123548]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 21)
Mar  1 21:33:27 localhost shorewall[123548]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 22)
Mar  1 21:33:27 localhost shorewall[123548]: Determining Hosts in Zones...
Mar  1 21:33:27 localhost shorewall[123548]: Locating Action Files...
Mar  1 21:33:28 localhost shorewall[123548]: Compiling /etc/shorewall/policy...
Mar  1 21:33:29 localhost shorewall[123548]: Adding Anti-smurf Rules
Mar  1 21:33:29 localhost shorewall[123548]: Adding rules for DHCP
Mar  1 21:33:29 localhost shorewall[123548]: Compiling TCP Flags filtering...
Mar  1 21:33:29 localhost shorewall[123548]: Compiling Kernel Route Filtering...
Mar  1 21:33:29 localhost shorewall[123548]: Compiling Martian Logging...
Mar  1 21:33:29 localhost shorewall[123548]: Compiling Accept Source Routing...
Mar  1 21:33:29 localhost shorewall[123548]: Compiling /etc/shorewall/tcinterfaces...
Mar  1 21:33:32 localhost shorewall[123548]: Compiling /etc/shorewall/snat...
Mar  1 21:33:32 localhost shorewall[123548]: Compiling MAC Filtration -- Phase 1...
Mar  1 21:33:32 localhost shorewall[123548]: Compiling /etc/shorewall/rules...
Mar  1 21:34:19 localhost shorewall[123548]: Compiling /etc/shorewall/conntrack...
Mar  1 21:34:19 localhost shorewall[123548]: Compiling MAC Filtration -- Phase 2...
Mar  1 21:34:19 localhost shorewall[123548]: Applying Policies...
Mar  1 21:34:19 localhost shorewall[123548]: Generating Rule Matrix...
Mar  1 21:34:23 localhost shorewall[123548]: Creating iptables-restore input...
Mar  1 21:34:23 localhost shorewall[123548]: Compiling /etc/shorewall/stoppedrules...
Mar  1 21:34:24 localhost shorewall[123548]: Shorewall configuration compiled to /var/lib/shorewall/.reload
Mar  1 21:34:27 localhost shorewall[123548]: Reloading Shorewall....
Mar  1 21:34:27 localhost shorewall[123548]: Initializing...
Mar  1 21:34:27 localhost shorewall[123548]: Setting up Route Filtering...
Mar  1 21:34:28 localhost shorewall[123548]: Setting up Martian Logging...
Mar  1 21:34:28 localhost shorewall[123548]: Setting up Accept Source Routing...
Mar  1 21:34:28 localhost shorewall[123548]: Setting up Traffic Control...
Mar  1 21:34:28 localhost shorewall[123548]: Preparing iptables-restore input...
Mar  1 21:34:28 localhost shorewall[123548]: Running /sbin/iptables-restore --wait 60...
Mar  1 21:34:28 localhost shorewall[123548]: IPv4 Forwarding Enabled
Mar  1 21:34:28 localhost shorewall[123548]: done.

Ysurac, 2020-03-01:
You are in china ? You use obfuscation.

jgreenhalgh51, 2020-03-01:
no it's a test but not change anything !

jgreenhalgh51, 2020-03-01:
Mar  1 21:49:45 localhost shorewall[127648]: done.
Mar  1 21:49:47 localhost systemd[1]: Reloading Shorewall IPv4 firewall.
Mar  1 21:49:48 localhost shorewall[128510]: Compiling using Shorewall 5.2.3.2...
Mar  1 21:49:52 localhost ss-server[126794]: remote recv: Connection reset by peer
Mar  1 21:50:34 localhost ss-server[126794]: getpeername: Transport endpoint is not connected
Mar  1 21:50:35 localhost shorewall[128510]: Processing /etc/shorewall/params ...
Mar  1 21:50:35 localhost shorewall[128510]: Processing /etc/shorewall/shorewall.conf...
Mar  1 21:50:35 localhost shorewall[128510]:    WARNING: The CHAIN_SCRIPTS configuration option is no longer supported /etc/shorewall/shorewall.conf (line 147)
Mar  1 21:50:35 localhost shorewall[128510]:    WARNING: The INLINE_MATCHES configuration option is no longer supported /etc/shorewall/shorewall.conf (line 183)
Mar  1 21:50:35 localhost shorewall[128510]:    WARNING: The LOAD_HELPERS_ONLY configuration option is no longer supported /etc/shorewall/shorewall.conf (line 191)
Mar  1 21:50:35 localhost shorewall[128510]:    WARNING: The MAPOLDACTIONS configuration option is no longer supported /etc/shorewall/shorewall.conf (line 199)
Mar  1 21:50:35 localhost shorewall[128510]:    WARNING: The MODULE_SUFFIX configuration option is no longer supported /etc/shorewall/shorewall.conf (line 205)
Mar  1 21:50:36 localhost shorewall[128510]: Loading Modules...
Mar  1 21:50:45 localhost shorewall[128510]: Compiling /etc/shorewall/zones...
Mar  1 21:50:45 localhost shorewall[128510]: Compiling /etc/shorewall/interfaces...
Mar  1 21:50:45 localhost shorewall[128510]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 18)
Mar  1 21:50:46 localhost shorewall[128510]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 19)
Mar  1 21:50:46 localhost shorewall[128510]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 20)
Mar  1 21:50:47 localhost shorewall[128510]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 21)
Mar  1 21:50:47 localhost shorewall[128510]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 22)
root@localhost:/var/log#

Ysurac, 2020-03-01:
No need to put same logs multiples times. I already saw that shorewall is reloading multiples times.

jgreenhalgh51, 2020-03-01:
ok sorry
you want acces to vps ? or routeur ?

Ysurac, 2020-03-01:
No.
Your configuration is strange, wan3 seems to be not totally configured.
I will do some test tomorrow, for time it's to sleep for me.

jgreenhalgh51, 2020-03-01:
wan 3 is an openwrt who is connected to my phone when i share my 4G

vinceducat, 2020-03-01:
Connect in usb?

vinceducat, 2020-03-01:
Can you copy the view of wizard and the viewof interface

jgreenhalgh51, 2020-03-03:
[image]

Ysurac, 2020-03-03:
What is the result of ```ps aux | grep ss``` on the server ? 
What is also the result of ```grep ss | /var/log/dameon.log``` always on the server ?

jgreenhalgh51, 2020-03-03:
Linux localhost 4.19.104-mptcp #1 SMP Mon Feb 17 11:08:15 UTC 2020 x86_64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
< OpenMPTCProuter VPS 0.1012 >
Last login: Sun Mar  1 21:25:06 2020 from 82.64.203.98
root@localhost:~# ps aux | grep ss
message+    676  0.0  0.7   8700  3368 ?        Ss   17:03   0:00 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only
root        815  0.0  1.3  15852  6372 ?        Ss   17:03   0:00 /usr/sbin/sshd -D
root       1155  0.0  1.6  16656  7604 ?        Ss   17:03   0:00 sshd: root@pts/0
root       1427  0.0  0.5   7648  2468 ?        Ss   17:03   0:00 /usr/bin/ss-manager -c /etc/shadowsocks-libev/manager.json
root       1433  0.0  0.8  11488  4212 ?        Ss   17:03   0:00 ss-server --manager-address 127.0.0.1:8839 -f /root/.shadowsocks/.shadowsocks_65101.pid -c /root/.shadowsocks/.shadowsocks_65101.conf --acl /etc/shadowsocks-libev/local.acl -t 600 -u --fast-open --reuse-port --no-delay --mptcp -6 -s 0.0.0.0
root       2352  0.0  1.6  16656  7992 ?        Ss   17:04   0:00 sshd: root@pts/1
root       2365  0.0  0.1   8604   820 pts/0    S+   17:04   0:00 grep ss
root@localhost:~#

jgreenhalgh51, 2020-03-03:
root@localhost:/var/log# grep ss | daemon.log
-bash: daemon.log: command not found

Envoyé de mon iPhone

> Le 3 mars 2020 à 16:44, Ycarus (Yannick Chabanois) <notifications@github.com> a écrit :
> 
> grep ss | /var/log/dameon.log

jgreenhalgh51, 2020-03-03:
root@localhost:/var/log# ls
alternatives.log       shorewall6-init.log
apt                    shorewall-init.log
auth.log               syslog
btmp                   syslog.1
chrony                 user.log
cloud-init.log         vmware-network.1.log
cloud-init-output.log  vmware-network.2.log
daemon.log             vmware-network.3.log
debug                  vmware-network.4.log
dpkg.log               vmware-network.5.log
faillog                vmware-network.log
kern.log               vmware-vmsvc.1.log
lastlog                vmware-vmsvc.2.log
messages               vmware-vmsvc.3.log
openvpn                vmware-vmsvc.log
private                wtmp
root@localhost:/var/log# grep ss | daemon.log
-bash: daemon.log: command not found

Ysurac, 2020-03-03:
```grep ss daemon.log``` sorry

jgreenhalgh51, 2020-03-03:
[log.txt](https://github.com/Ysurac/openmptcprouter/files/4282770/log.txt)
Voila

Ysurac, 2020-03-03:
You can try to disable TCP Fast Open in System->OpenMPTCProuter and advanced settings tab.

jgreenhalgh51, 2020-03-03:
i am in udp and it's same problem

jgreenhalgh51, 2020-03-03:
[image]

jgreenhalgh51, 2020-03-03:
[image]

jgreenhalgh51, 2020-03-03:
maybe the vps is not powerful enough

jgreenhalgh51, 2020-03-03:
with all errors the cpu of VPS go to 100% after i must reboot the vps

Ysurac, 2020-03-03:
TCP Fastopen is also used by shadowsocks.
It's possible that the VPS is really bad...

jgreenhalgh51, 2020-03-03:
ok i test with an other vps

jgreenhalgh51, 2020-03-03:
Tcp fastopen disable 
But i have same problem

jgreenhalgh51, 2020-03-03:
my vps

1 vCore Intel® Xeon® E5 (CPU)
512 Mo de RAM
10 Go de SSD

jgreenhalgh51, 2020-03-05:
last error on kvm console on my vps 

[image]

jgreenhalgh51, 2020-03-05:
i must reboot vps

jgreenhalgh51, 2020-03-05:
i have change vps but i have the same problem

jgreenhalgh51, 2020-03-05:
[image]

jgreenhalgh51, 2020-03-10:
Hi 
with the new version , the tunnel is down but the tunnel work !
I have the aggregation of my adsl and 4g connection but in wizard interface the vpn is down

Ysurac, 2020-03-10:
VPN is used only for anything that is not UDP, so TCP aggregation will still work.
So when the VPN is show as down, the VPN is really down.
Check that VPS was rebooted after restart, if it's the case try to valid the wizard again.

jgreenhalgh51, 2020-03-12:
after reboot vpn up - after down in 1 min

jgreenhalgh51, 2020-03-12:
Thu Mar 12 03:51:58 2020 daemon.info /usr/bin/ss-redir[20553]: tcp port reuse enabled
Thu Mar 12 03:51:58 2020 daemon.info /usr/bin/ss-redir[20553]: UDP relay enabled
Thu Mar 12 03:51:58 2020 daemon.info /usr/bin/ss-redir[20553]: udp port reuse enabled
Thu Mar 12 03:51:58 2020 daemon.info /usr/bin/ss-redir[20553]: running from root user
Thu Mar 12 03:51:58 2020 user.notice Shadowsocks: Rules DOWN
Thu Mar 12 03:51:58 2020 daemon.warn dnsmasq[19621]: failed to create listening socket for 10.255.254.2: Address not available
Thu Mar 12 03:51:58 2020 daemon.warn dnsmasq[19621]: failed to create listening socket for 10.255.254.2: Address not available
Thu Mar 12 03:51:58 2020 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Thu Mar 12 03:51:58 2020 user.notice MPTCP: Add route 192.168.250.0/24 via 192.168.100.250 dev eth0
Thu Mar 12 03:51:59 2020 user.notice Shadowsocks: Rules UP
Thu Mar 12 03:51:59 2020 daemon.err omr-tracker[12893]: RTNETLINK answers: Permission denied
Thu Mar 12 03:51:59 2020 user.notice Shadowsocks: Reload omr-bypass rules
Thu Mar 12 03:51:59 2020 user.notice omr-bypass: Starting OMR-ByPass...
Thu Mar 12 03:52:01 2020 user.notice omr-bypass: Reload dnsmasq...
Thu Mar 12 03:52:01 2020 daemon.info dnsmasq[19621]: read /etc/hosts - 4 addresses
Thu Mar 12 03:52:01 2020 daemon.info dnsmasq[19621]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Thu Mar 12 03:52:01 2020 user.notice omr-bypass: OMR-ByPass is running
Thu Mar 12 03:52:01 2020 user.info : using tcp fast open
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-redir[22185]: enable TCP no-delay
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-redir[22185]: resolving hostname to IPv6 address first
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-redir[22185]: initializing ciphers... chacha20-ietf-poly1305
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-local[22183]: using tcp fast open
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-local[22183]: enable TCP no-delay
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-local[22183]: initializing ciphers... chacha20-ietf-poly1305
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-redir[22185]: listening at 0.0.0.0:1100
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-redir[22185]: tcp port reuse enabled
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-redir[22185]: UDP relay enabled
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-redir[22185]: udp port reuse enabled
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-local[22183]: listening at 127.0.0.1:1111
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-local[22183]: tcp port reuse enabled
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-local[22183]: udprelay enabled
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-local[22183]: udp port reuse enabled
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-local[22183]: running from root user
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-redir[22185]: running from root user
Thu Mar 12 03:52:01 2020 user.info : using tcp fast open
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-redir[22184]: enable TCP no-delay
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-redir[22184]: resolving hostname to IPv6 address first
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-redir[22184]: initializing ciphers... chacha20-ietf-poly1305
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-redir[22184]: listening at 0.0.0.0:1100
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-redir[22184]: tcp port reuse enabled
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-redir[22184]: UDP relay enabled
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-redir[22184]: udp port reuse enabled
Thu Mar 12 03:52:01 2020 daemon.info /usr/bin/ss-redir[22184]: running from root user
Thu Mar 12 03:52:01 2020 daemon.info procd: - init complete -
Thu Mar 12 03:52:01 2020 daemon.info vnstatd[22426]: No interfaces found in database, adding available interfaces...
Thu Mar 12 03:52:01 2020 daemon.info vnstatd[22426]: Interface "ifb4wan4" added with 1000 Mbit bandwidth limit.
Thu Mar 12 03:52:01 2020 daemon.info vnstatd[22426]: Interface "ifb4tun0" added with 1000 Mbit bandwidth limit.
Thu Mar 12 03:52:01 2020 daemon.info vnstatd[22426]: Interface "eth1" added with 10000 Mbit bandwidth limit.
Thu Mar 12 03:52:01 2020 daemon.info vnstatd[22426]: Interface "wan2" added with 10000 Mbit bandwidth limit.
Thu Mar 12 03:52:01 2020 daemon.info vnstatd[22426]: Interface "wan1" added with 10000 Mbit bandwidth limit.
Thu Mar 12 03:52:01 2020 daemon.info vnstatd[22426]: Interface "eth0" added with 10000 Mbit bandwidth limit.
Thu Mar 12 03:52:01 2020 daemon.info vnstatd[22426]: Interface "ifb1" added with 1000 Mbit bandwidth limit.
Thu Mar 12 03:52:01 2020 daemon.info vnstatd[22426]: Interface "teql0" added with 1000 Mbit bandwidth limit.
Thu Mar 12 03:52:01 2020 daemon.info vnstatd[22426]: Interface "tun0" added with 10 Mbit bandwidth limit.
Thu Mar 12 03:52:01 2020 daemon.info vnstatd[22426]: Interface "wan4" added with 10000 Mbit bandwidth limit.
Thu Mar 12 03:52:01 2020 daemon.info vnstatd[22426]: Interface "ifb0" added with 1000 Mbit bandwidth limit.
Thu Mar 12 03:52:01 2020 daemon.info vnstatd[22426]: -> 11 new interfaces found.
Thu Mar 12 03:52:01 2020 daemon.info vnstatd[22426]: Limits can be modified using the configuration file. See "man vnstat.conf".
Thu Mar 12 03:52:01 2020 daemon.info vnstatd[22426]: Unwanted interfaces can be removed from monitoring with "vnstat --remove".
[... cut ...]
Thu Mar 12 03:52:02 2020 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 212.27.48.10)
Thu Mar 12 03:52:46 2020 user.notice post-tracking-post-tracking: wan4 switched off because gateway down and ping from 192.168.3.2 error (4.2.2.1,8.8.8.8,8.8.4.4)
Thu Mar 12 03:52:46 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.3.1 dev wan4
Thu Mar 12 03:53:12 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Thu Mar 12 03:53:16 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done

[15 more replies]

closed 2020-03-13 by Ysurac