Ysurac/openmptcprouter #866: J'ai de nouveau le problème comme le #834

Slivikkus, 2020-03-01:
## Actual Behavior
Bonjour, j'ai exactement le même problème que sur le #834 

J'ai tout réinstaller (VPS et routeur), ça à fonctionner pendant quelques heures après ça me refait la même chose, ça commence à bien impacté mes serveurs (qui tournent un peu dans le vide pour l'instant, j'en ai même du éteindre certains serveurs puisque la connexion fonctionne très mal)

[image]
(et encore là j'ai reboot le VPS ça me fait ça)

## Specifications

  - OpenMPTCProuter version: (0.53.3 (ESXI)
  - OpenMPTCProuter VPS version: (Inconnue)
  - OpenMPTCProuter platform: (x86_64)

Slivikkus, 2020-03-01:
Des fois ça peux être comme ça aussi (ça change presque rien en sois)

Ysurac, 2020-03-01:
Déjà il faudrait savoir la version du VPS, si ce n'est la v0.1012 il faut mettre à jour.
Tu peux aussi tester la dernière RC de la v0.54 d'OMR https://github.com/Ysurac/openmptcprouter/issues/864

jgreenhalgh51, 2020-03-01:
i haven't change config  just active upnp

Slivikkus, 2020-03-01:
J'ai la 0.1011 côté vps je met à jour tout à l'heure

Slivikkus, 2020-03-01:
Et où je peux trouver pour ESXI pour la version 0.54 ? :)

Slivikkus, 2020-03-01:
VPS à jour rien ne change pour l'instant

Slivikkus, 2020-03-02:
J'ai l'impression que je perd la connexion au VPS toutes les 30 secondes ...

closed 2020-03-03 by Slivikkus

Archana7700, 2020-03-04:
how to check the VPS version? i have flashed the new firmware and OMR version is now v0.54rc9 

but issue is the same as shown in pic "cant get the ip address from the shadowsocks cant contach the server "how to resolve this

Ysurac, 2020-03-04:
VPS release is available in ```/etc/motd```