Ysurac/openmptcprouter #870: ERR_CONNECTION_CLOSED issues

darthclide, 2020-03-03:
## Expected Behavior

URL will open upon first visit.

## Actual Behavior

I have to repeatedly hit refresh until the page randomly decides to load. The issue comes and goes seemingly at random as well.

## Steps to Reproduce the Problem

Visit any URL (including popular locations such as google.com)

## Specifications

OpenMPTCProuter version: v0.53.3 r0+11546-a74095c68c (running on Virtual Box on my Windows 10 machine)
OpenMPTCProuter VPS version: Version 0.1009 4.19.80-mptcp
OpenMPTCProuter platform: VPS server = Debian 10 x64 My computer = host machine: Windows 10 Pro x64, with router software on Virtual Box (operating system Linux 2.6 / 3.x / 4.x (64-bit)

Extra info: I only started to encounter this issue after I updated to 0.1009 4.19.80-mptcp on the server. A couple days before I did that update, I added T-mobile into the mix, which uses an APN with an IPv6 address (I have no choice. I can't connect to the APN with IPv4). So perhaps it is a combination of this update and this carrier that is screwing with things? In my VM my T-mobile is getting reset approximately every hour for no apparent reason. I will attach some screenshots in a response to this post when these messages come through as I stream tonight.

Ysurac, 2020-03-03:
IPv6 only ? In this case this will make problem. Mixing IPv4 only and IPv6 only is not supported yet.
For IPv4 only, the server need to be contacted over IPv4, and over IPv6 for IPv6 only.

darthclide, 2020-03-03:
[image]

So what do you suggest I do? I don't know of any way to tell a Jetpack 8800L or a Velocity 2 to connect over IPv6. Should I call T-mobile and see if they can get IPv4 to work at my location?

darthclide, 2020-03-03:
What is confusing on top of all this, is that sometimes Google reports back an IPv4 address as my external one (when I just connect to T-mobile by itself). So it is almost as if I technically CAN get an IPv4 address, but it is out of my hands or something?

darthclide, 2020-03-04:
A small update, I don't know if it matters or not, but while I can't change my APN to IPv4 only, I can change it to IPv4/IPv6. Also, my roaming APN is set to IPv4/IPv6. I don't know much about how these protocols work, but can I assume it simply just chose IPv6 even though I am giving it the option to choose IPv4?

Ysurac, 2020-03-04:
You have to set it to IPv4/IPv6, and check if you get an IPv4. Else I think it should be possible to get one with 464xlat but I don't know much about that.

darthclide, 2020-03-04:
I do get an IPv4 50% of the time, but A. I don't know how to check if I am on IPv6 once I have bonded the traffic (I am suspicious that it is constantly changing back and forth between IPv4 and IPv6) and B. I am still getting those errors I showed in the previous post, even when I know that I was in IPv4 right before starting up the VM.

[image]

Am I supposed to call up T-mobile and ask about 464xlat? Or is this some piece of software I have to run on my phone?

Further, you mentioned "not supported yet". Did you say this to inspire confidence in you and the team's ability to upgrade this software? Or is this feature A. So niche that nobody wants it and/or B. It is very complicated to implement?

Ysurac, 2020-03-04:
What use the ip 192.168.42.242 ?

darthclide, 2020-03-04:
[image]
[image]

NDIS gives it an IP, but then your software gives it a different IP in the same subnet. I assumed this doesn't hurt anything? Especially since everything works as it should, except for webpages sometimes getting stuck on ERR_CONNECTION_CLOSED

Ysurac, 2020-03-04:
No idea, but it seems that eth1 is seen sometimes down. This can give results like that. You can try in Network->MPTCP to increase "Multipath TCP SYN retries"

darthclide, 2020-03-04:
What is the maximum you recommend? I came across a post where you said 5, but I have already tried that to no avail.

Are you now questioning whether this is IPv6 that is the cause? I find it interesting that your software is talking about IPv6 for my Virtual Box adapter when it uses an IPv4 address:
[image]

Further, it is interesting that this issue comes and goes. Do you have a tool inside your router that allows me to check what my external IP is for a specific interface? Because if I could do this at the same time I get ERR_CONNECTION CLOSED, I could confirm whether it is a shifting IP address that is causing this issue.

darthclide, 2020-03-06:
More info: I switched my Virtual Box adapter type from Intel PRO/1000 MT Desktop (82540EM)" to "PCnet - FAST III (Am79C973) on all adapters and while I haven't seen any resets happen or that weird IPv6 NETDEV message, I have received "err_empty_response" a few times when I visit a url. Perhaps the reset is still happening, but due to this adapter type change, the log isn't picking up on the reset? Also things seem a little bit more sluggish for URL page loads.

darthclide, 2020-03-08:
Hoping for any additional thoughts before I take the plunge and buy a Raspberry PI. I really want to solve this as quickly as possible as I am currently already spending 120$ a month on DSL, plus all these carriers.

Ysurac, 2020-03-08:
Using USB on a RPI is not always a good idea. This can use too much power for the RPI. At least a RPI4 doesn't share USB bandwidth with network bandwidth like RPI2/3.
For problems on a VM under windows, many problems can happen. This can be related to some config from the host.

darthclide, 2020-03-08:
> Using USB on a RPI is not always a good idea. This can use too much power for the RPI.

I can get around this by using a powered USB hub I think?
> At least a RPI4 doesn't share USB bandwidth with network bandwidth like RPI2/3.

Are you saying the RPI4 is superior for this reason? Or that the 2/3 are better?
> For problems on a VM under windows, many problems can happen. This can be related to some config from the host.

So in your estimation, what are the odds my adapter is being reset because of something dumb in Virtual Box, versus a Raspberry PI working without these problems? If you give me >50% chance it is my VM acting dumb, then please tell me so I can go through with this purchase.

Ysurac, 2020-03-08:
Yes, you can use a powered USB hub.

RPI4 is better.

No idea. I can't test all possibilities and I didn't tested the hardware you use for connections.

darthclide, 2020-03-08:
Well I know you said it is T-mobile's IPV6 that was the issue. But seeing as I get an external IPv4 address, I don't think it is IPV6 exclusive. I gave up bonding Verizon because of issues connecting over WiFi. So that means I am currently bonding the following:

AT&T over 2.4 Ghz WiFi from a ZTE Velocity 2
DSL over ethernet coming directly from the modem to my computer
T-mobile plugged into my computer with a USB cable with a Motorola Moto E6

In addition, these ERR_CONNECTION_CLOSED url messages do not sync up with those errors showing up in the OMR log for my T-mobile interface. It just happens at random times.

With this in mind, would you say it is worth a shot to try a RPI 4?

darthclide, 2020-03-12:
Well I took the plunge and bought the RPI 4 plus a USB hub. You wouldn't believe how hard it is to find a HUB that allows you to inject power into the hub separately from where the data is going out over a USB type A cable. I will report back if this ERR_CONNECTION_CLOSED issue magically disappears when I move this off the VirtualBox and onto physical hardware.

darthclide, 2020-09-09:
While I have had other struggles, switching to a Raspberry Pi 4 solved the majority of my issues. But here is a big recommendation: Don't use USB tethering. It is such a pain dealing with USB problems as well as the Pi not having good WiFi chips. Get some nice 4G modems (that have the best coverage for the carrier's sim card you are using) with Ethernet ports. Plug these into a router. Then plug that router into the Pi's Ethernet port. Do some configuration that both pulls in internet from all the WANs on that port but also provides a LAN out of that port. (Ysurac can help)

closed 2020-09-09 by darthclide