Ysurac/openmptcprouter #871: OMR VPN ne répond pas

florianlog, 2020-03-04:
Bonjour, ce message fais surement doublon et je m'en excuse mais après essayer toutes les manips dans les autres messages je n'arrive pas à avoir une connexion stable sur mon VPN

Ma config : 
- VPS chez OVH
- OMR sur WRT3200ACM
- Wan 1 sur freebox et Wan 2 sur Routeur 4G (FREE)

[image]

Merci d'avance si vous avez une solution, j'ai installé un autre OMR précédemment et j'ai pas ce type de soucis pourtant je suis en même config.

florianlog, 2020-03-04:
Mes journaux :

Wed Mar  4 10:24:02 2020 user.notice omr-bypass: OMR-ByPass is running
Wed Mar  4 10:24:04 2020 user.notice omr-tracker: Can't create route to 4.2.2.1 via 192.168.43.1 dev eth0.2 src 192.168.43.104. waiting...
Wed Mar  4 10:24:04 2020 user.notice OMR-VPS: Can't get token, try later
Wed Mar  4 10:24:04 2020 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Wed Mar  4 10:24:06 2020 user.notice post-tracking-post-tracking: Set server vps (51.91.100.155) route via 192.168.1.1 metric 4
Wed Mar  4 10:24:06 2020 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Wed Mar  4 10:24:06 2020 user.notice Shadowsocks: Rules DOWN
Wed Mar  4 10:24:07 2020 daemon.err omr-tracker[12306]: Command failed: Not found
Wed Mar  4 10:24:07 2020 user.notice Shadowsocks: Rules UP
Wed Mar  4 10:24:07 2020 daemon.err omr-tracker[12306]: RTNETLINK answers: Permission denied
Wed Mar  4 10:24:07 2020 user.notice Shadowsocks: Reload omr-bypass rules
Wed Mar  4 10:24:07 2020 user.notice omr-bypass: Starting OMR-ByPass...
Wed Mar  4 10:24:07 2020 daemon.err omr-tracker[12306]: 0+5 records in
Wed Mar  4 10:24:07 2020 daemon.err omr-tracker[12306]: 0+5 records out
Wed Mar  4 10:24:07 2020 daemon.err omr-tracker[12306]: 0+5 records in
Wed Mar  4 10:24:07 2020 daemon.err omr-tracker[12306]: 0+5 records out
Wed Mar  4 10:24:07 2020 user.notice OMR-VPS: Can't get token, try later
Wed Mar  4 10:24:08 2020 user.notice omr-bypass: Reload dnsmasq...
Wed Mar  4 10:24:09 2020 daemon.info dnsmasq[13955]: read /etc/hosts - 4 addresses
Wed Mar  4 10:24:09 2020 daemon.info dnsmasq[13955]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Wed Mar  4 10:24:09 2020 daemon.info dnsmasq-dhcp[13955]: read /etc/ethers - 0 addresses
Wed Mar  4 10:24:09 2020 user.notice omr-bypass: OMR-ByPass is running
Wed Mar  4 10:24:09 2020 user.info : using tcp fast open
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-redir[15520]: enable TCP no-delay
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-tunnel[15522]: initializing ciphers... chacha20-ietf-poly1305
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-redir[15520]: resolving hostname to IPv6 address first
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-redir[15520]: initializing ciphers... chacha20-ietf-poly1305
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-tunnel[15522]: listening at 0.0.0.0:5353
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-redir[15520]: listening at 0.0.0.0:1100
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-redir[15520]: tcp port reuse enabled
Wed Mar  4 10:24:09 2020 daemon.err /usr/bin/ss-tunnel[15522]: bind: Address in use
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-redir[15520]: UDP relay enabled
Wed Mar  4 10:24:09 2020 daemon.err /usr/bin/ss-tunnel[15522]: bind() error
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-redir[15520]: udp port reuse enabled
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-redir[15520]: running from root user
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-local[15519]: using tcp fast open
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-local[15519]: initializing ciphers... chacha20-ietf-poly1305
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-local[15519]: listening at 127.0.0.1:1111
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-local[15519]: tcp port reuse enabled
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-local[15519]: udprelay enabled
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-local[15519]: udp port reuse enabled
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-local[15519]: running from root user
Wed Mar  4 10:24:09 2020 user.info : using tcp fast open
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-redir[15521]: enable TCP no-delay
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-redir[15521]: resolving hostname to IPv6 address first
Wed Mar  4 10:24:09 2020 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact 1.1.1.1, 1.0.0.1, 212.27.48.10)
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-redir[15521]: initializing ciphers... chacha20-ietf-poly1305
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-redir[15521]: listening at 0.0.0.0:1100
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-redir[15521]: tcp port reuse enabled
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-redir[15521]: UDP relay enabled
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-redir[15521]: udp port reuse enabled
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-redir[15521]: running from root user
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-tunnel[15623]: initializing ciphers... chacha20-ietf-poly1305
Wed Mar  4 10:24:09 2020 daemon.info /usr/bin/ss-tunnel[15623]: listening at 0.0.0.0:5353
Wed Mar  4 10:24:09 2020 daemon.err /usr/bin/ss-tunnel[15623]: bind: Address in use
Wed Mar  4 10:24:09 2020 daemon.err /usr/bin/ss-tunnel[15623]: bind() error
Wed Mar  4 10:24:09 2020 user.notice Shadowsocks: Rules DOWN
Wed Mar  4 10:24:09 2020 daemon.info procd: - init complete -
Wed Mar  4 10:24:09 2020 daemon.info vnstatd[15696]: No interfaces found in database, adding available interfaces...
Wed Mar  4 10:24:09 2020 daemon.info vnstatd[15696]: Interface "tun0" added with 10 Mbit bandwidth limit.
Wed Mar  4 10:24:09 2020 daemon.info vnstatd[15696]: Interface "eth0.1" added with 1000 Mbit bandwidth limit.
Wed Mar  4 10:24:09 2020 daemon.info vnstatd[15696]: Interface "eth1.2" added with 1000 Mbit bandwidth limit.
[... cut ...]
Wed Mar  4 11:10:37 2020 daemon.err odhcp6c[5634]: Failed to send DHCPV6 message to ff02::1:2 (Address not available)
Wed Mar  4 11:12:29 2020 daemon.err odhcp6c[5634]: Failed to send DHCPV6 message to ff02::1:2 (Address not available)
Wed Mar  4 11:14:37 2020 daemon.err odhcp6c[5634]: Failed to send DHCPV6 message to ff02::1:2 (Address not available)
Wed Mar  4 11:16:40 2020 daemon.err odhcp6c[5634]: Failed to send DHCPV6 message to ff02::1:2 (Address not available)
Wed Mar  4 11:18:38 2020 daemon.err odhcp6c[5634]: Failed to send DHCPV6 message to ff02::1:2 (Address not available)

Ysurac, 2020-03-04:
Quelle est la version du script installée sur le serveur ? (visible dans /etc/motd)
Voir aussi dans /var/log/daemon.log (sous Debian) sur le serveur ce qui peut concerner le VPN.

florianlog, 2020-03-04:
le script est en 0.1008

Mar  4 11:20:30 vps791338 systemd[1]: Stopped Glorytun TCP on tun0.
Mar  4 11:20:30 vps791338 systemd[1]: Started Glorytun TCP on tun0.
Mar  4 11:20:30 vps791338 systemd-udevd[6866]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Mar  4 11:20:30 vps791338 systemd[1]: systemd-networkd-wait-online.service: Succeeded.
Mar  4 11:20:30 vps791338 systemd[1]: Stopped Wait for Network to be Configured.
Mar  4 11:20:30 vps791338 systemd[1]: Stopping Wait for Network to be Configured...
Mar  4 11:20:30 vps791338 systemd[1]: Stopping Network Service...
Mar  4 11:20:30 vps791338 systemd[1]: systemd-networkd.service: Succeeded.
Mar  4 11:20:30 vps791338 systemd[1]: Stopped Network Service.
Mar  4 11:20:30 vps791338 systemd[1]: Starting Network Service...
Mar  4 11:20:30 vps791338 systemd-networkd[6871]: dsvpn0: Gained IPv6LL
Mar  4 11:20:30 vps791338 systemd-networkd[6871]: omr-6in4: Gained IPv6LL
Mar  4 11:20:30 vps791338 systemd-networkd[6871]: tun1: Gained IPv6LL
Mar  4 11:20:30 vps791338 systemd-networkd[6871]: gt-udp-tun0: Gained IPv6LL
Mar  4 11:20:30 vps791338 systemd-networkd[6871]: tun0: Gained IPv6LL
Mar  4 11:20:30 vps791338 systemd-networkd[6871]: mlvpn0: Gained IPv6LL
Mar  4 11:20:30 vps791338 systemd-networkd[6871]: ens3: Gained IPv6LL
Mar  4 11:20:30 vps791338 systemd-networkd[6871]: Enumeration completed
Mar  4 11:20:30 vps791338 systemd[1]: Started Network Service.
Mar  4 11:20:30 vps791338 systemd[1]: Starting Wait for Network to be Configured...
Mar  4 11:20:30 vps791338 systemd-networkd[6871]: gt-tun0: Gained carrier
Mar  4 11:20:30 vps791338 systemd-networkd[6871]: gt-tun0: Gained IPv6LL
Mar  4 11:20:42 vps791338 systemd-networkd[6871]: gt-udp-tun0: Configured
Mar  4 11:20:42 vps791338 systemd-networkd[6871]: mlvpn0: Configured
Mar  4 11:20:42 vps791338 systemd-networkd[6871]: gt-tun0: Configured
Mar  4 11:20:42 vps791338 systemd-networkd-wait-online[6873]: ignoring: lo
Mar  4 11:20:42 vps791338 systemd[1]: Started Wait for Network to be Configured.
Mar  4 11:21:11 vps791338 ss-server[445]: getpeername: Transport endpoint is not connected
Mar  4 11:22:02 vps791338 ss-server[445]: getpeername: Transport endpoint is not connected
Mar  4 11:22:53 vps791338 ss-server[445]: getpeername: Transport endpoint is not connected
Mar  4 11:23:44 vps791338 ss-server[445]: getpeername: Transport endpoint is not connected
Mar  4 11:24:35 vps791338 ss-server[445]: getpeername: Transport endpoint is not connected
Mar  4 11:25:00 vps791338 glorytun-tcp-run[6862]: INITIALIZED gt-tun0
Mar  4 11:25:00 vps791338 systemd-networkd[6871]: gt-tun0: Lost carrier
Mar  4 11:25:00 vps791338 systemd[1]: Stopping Glorytun TCP on tun0...
Mar  4 11:25:00 vps791338 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Mar  4 11:25:00 vps791338 systemd[1]: Stopped Glorytun TCP on tun0.
Mar  4 11:25:00 vps791338 systemd[1]: Started Glorytun TCP on tun0.
Mar  4 11:25:00 vps791338 systemd-udevd[7327]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Mar  4 11:25:00 vps791338 systemd[1]: systemd-networkd-wait-online.service: Succeeded.
Mar  4 11:25:00 vps791338 systemd[1]: Stopped Wait for Network to be Configured.
Mar  4 11:25:00 vps791338 systemd[1]: Stopping Wait for Network to be Configured...
Mar  4 11:25:00 vps791338 systemd[1]: Stopping Network Service...
Mar  4 11:25:00 vps791338 systemd[1]: systemd-networkd.service: Succeeded.
Mar  4 11:25:00 vps791338 systemd[1]: Stopped Network Service.
Mar  4 11:25:00 vps791338 systemd[1]: Starting Network Service...
Mar  4 11:25:00 vps791338 systemd-networkd[7335]: dsvpn0: Gained IPv6LL
Mar  4 11:25:00 vps791338 systemd-networkd[7335]: omr-6in4: Gained IPv6LL
Mar  4 11:25:00 vps791338 systemd-networkd[7335]: tun1: Gained IPv6LL
Mar  4 11:25:00 vps791338 systemd-networkd[7335]: gt-udp-tun0: Gained IPv6LL
Mar  4 11:25:00 vps791338 systemd-networkd[7335]: tun0: Gained IPv6LL
Mar  4 11:25:00 vps791338 systemd-networkd[7335]: mlvpn0: Gained IPv6LL
Mar  4 11:25:00 vps791338 systemd-networkd[7335]: ens3: Gained IPv6LL
Mar  4 11:23:44 vps791338 ss-server[445]: getpeername: Transport endpoint is not connected
Mar  4 11:24:35 vps791338 ss-server[445]: getpeername: Transport endpoint is not connected
Mar  4 11:25:00 vps791338 glorytun-tcp-run[6862]: INITIALIZED gt-tun0
Mar  4 11:25:00 vps791338 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Mar  4 11:25:00 vps791338 systemd[1]: Stopped Glorytun TCP on tun0.
Mar  4 11:25:00 vps791338 systemd[1]: Started Glorytun TCP on tun0.
Mar  4 11:25:00 vps791338 systemd-udevd[7327]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Mar  4 11:25:00 vps791338 systemd[1]: systemd-networkd-wait-online.service: Succeeded.
Mar  4 11:25:00 vps791338 systemd[1]: Stopped Wait for Network to be Configured.
Mar  4 11:25:00 vps791338 systemd[1]: Stopping Wait for Network to be Configured...
Mar  4 11:25:00 vps791338 systemd[1]: Stopping Network Service...
Mar  4 11:25:00 vps791338 systemd[1]: systemd-networkd.service: Succeeded.
Mar  4 11:25:00 vps791338 systemd[1]: Stopped Network Service.
Mar  4 11:25:00 vps791338 systemd[1]: Starting Network Service...
Mar  4 11:25:00 vps791338 systemd-networkd[7335]: dsvpn0: Gained IPv6LL
[... 19 lines cut ...]
Mar  4 11:25:00 vps791338 systemd[1]: Stopping Glorytun TCP on tun0...
Mar  4 11:25:00 vps791338 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
[... cut ...]
Mar  4 11:25:00 vps791338 systemd-networkd[7335]: ens3: Gained IPv6LL
Mar  4 11:25:00 vps791338 systemd-networkd[7335]: Enumeration completed
Mar  4 11:25:00 vps791338 systemd-networkd[7335]: Enumeration completed
Mar  4 11:25:00 vps791338 systemd[1]: Started Network Service.
Mar  4 11:25:00 vps791338 systemd[1]: Starting Wait for Network to be Configured...
Mar  4 11:25:00 vps791338 systemd-networkd[7335]: gt-tun0: Gained carrier
Mar  4 11:25:00 vps791338 systemd-networkd[7335]: gt-tun0: Gained IPv6LL

florianlog, 2020-03-04:
Je viens de mettre à jour le script, pour info le VPN se connecte et se déconnecte en permanence j'arrive à me connecter à mon usine mais le débit est tellement lent que je ne peut rien modifier. j'ai deux autres usines à équiper avec OMR j'espère pourvoir résoudre ce soucis.

Ysurac, 2020-03-04:
Normalement le log sur le serveur doit être cependant différent. Après la mise à jour, le serveur a été redémarré ?

florianlog, 2020-03-04:
Le serveur et le routeur ont été redémarré apres la MAJ

florianlog, 2020-03-04:
le log du serveur VPS :

Mar  4 11:57:08 vps791338 shorewall[2739]: Compiling Martian Logging...
Mar  4 11:57:08 vps791338 shorewall[2739]: Compiling Accept Source Routing...
Mar  4 11:57:08 vps791338 shorewall[2739]: Compiling /etc/shorewall/tcinterfaces...
Mar  4 11:57:08 vps791338 shorewall[2739]: Compiling /etc/shorewall/snat...
Mar  4 11:57:08 vps791338 shorewall[2739]: Compiling MAC Filtration -- Phase 1...
Mar  4 11:57:08 vps791338 shorewall[2739]: Compiling /etc/shorewall/rules...
Mar  4 11:57:08 vps791338 shorewall[2739]: Compiling /etc/shorewall/conntrack...
Mar  4 11:57:08 vps791338 shorewall[2739]: Compiling MAC Filtration -- Phase 2...
Mar  4 11:57:08 vps791338 shorewall[2739]: Applying Policies...
Mar  4 11:57:08 vps791338 shorewall[2739]: Generating Rule Matrix...
Mar  4 11:57:08 vps791338 shorewall[2739]: Creating iptables-restore input...
Mar  4 11:57:08 vps791338 shorewall[2739]: Compiling /etc/shorewall/stoppedrules...
Mar  4 11:57:08 vps791338 shorewall[2739]: Shorewall configuration compiled to /var/lib/shorewall/.reload
Mar  4 11:57:08 vps791338 shorewall[2739]: Reloading Shorewall....
Mar  4 11:57:08 vps791338 shorewall[2739]: Initializing...
Mar  4 11:57:08 vps791338 shorewall[2739]: Setting up Route Filtering...
Mar  4 11:57:08 vps791338 shorewall[2739]: Setting up Martian Logging...
Mar  4 11:57:08 vps791338 shorewall[2739]: Setting up Accept Source Routing...
Mar  4 11:57:08 vps791338 shorewall[2739]: Setting up Traffic Control...
Mar  4 11:57:08 vps791338 shorewall[2739]: Preparing iptables-restore input...
Mar  4 11:57:08 vps791338 shorewall[2739]: Running /sbin/iptables-restore --wait 60...
Mar  4 11:57:08 vps791338 shorewall[2739]: IPv4 Forwarding Enabled
Mar  4 11:57:08 vps791338 shorewall[2739]: done.
Mar  4 11:57:08 vps791338 systemd[1]: Reloaded Shorewall IPv4 firewall.
Mar  4 11:57:09 vps791338 systemd[1]: Reloading Shorewall IPv4 firewall.
Mar  4 11:57:09 vps791338 shorewall[3048]: Compiling using Shorewall 5.2.3.2...
Mar  4 11:57:09 vps791338 shorewall[3048]: Processing /etc/shorewall/params ...
Mar  4 11:57:09 vps791338 shorewall[3048]: Processing /etc/shorewall/shorewall.conf...
Mar  4 11:57:09 vps791338 shorewall[3048]:    WARNING: The CHAIN_SCRIPTS configuration option is no longer supported /etc/shorewall/shorewall.conf (line 147)
Mar  4 11:57:09 vps791338 shorewall[3048]:    WARNING: The INLINE_MATCHES configuration option is no longer supported /etc/shorewall/shorewall.conf (line 183)
Mar  4 11:57:09 vps791338 shorewall[3048]:    WARNING: The LOAD_HELPERS_ONLY configuration option is no longer supported /etc/shorewall/shorewall.conf (line 191)
Mar  4 11:57:09 vps791338 shorewall[3048]:    WARNING: The MAPOLDACTIONS configuration option is no longer supported /etc/shorewall/shorewall.conf (line 199)
Mar  4 11:57:09 vps791338 shorewall[3048]:    WARNING: The MODULE_SUFFIX configuration option is no longer supported /etc/shorewall/shorewall.conf (line 205)
Mar  4 11:57:09 vps791338 shorewall[3048]: Loading Modules...
Mar  4 11:57:09 vps791338 shorewall[3048]: Compiling /etc/shorewall/zones...
Mar  4 11:57:09 vps791338 shorewall[3048]: Compiling /etc/shorewall/interfaces...
Mar  4 11:57:09 vps791338 shorewall[3048]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 18)
Mar  4 11:57:09 vps791338 shorewall[3048]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 19)
Mar  4 11:57:09 vps791338 shorewall[3048]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 20)
Mar  4 11:57:09 vps791338 shorewall[3048]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 21)
Mar  4 11:57:09 vps791338 shorewall[3048]:    WARNING: The 'routefilter' option is ignored when used with a wildcard physical name /etc/shorewall/interfaces (line 22)
Mar  4 11:57:09 vps791338 shorewall[3048]: Determining Hosts in Zones...
Mar  4 11:57:09 vps791338 shorewall[3048]: Locating Action Files...
Mar  4 11:57:09 vps791338 shorewall[3048]: Compiling /etc/shorewall/policy...
Mar  4 11:57:09 vps791338 shorewall[3048]: Adding Anti-smurf Rules
Mar  4 11:57:09 vps791338 shorewall[3048]: Adding rules for DHCP
Mar  4 11:57:09 vps791338 shorewall[3048]: Compiling TCP Flags filtering...
Mar  4 11:57:09 vps791338 shorewall[3048]: Compiling Kernel Route Filtering...
Mar  4 11:57:09 vps791338 shorewall[3048]: Compiling Martian Logging...
Mar  4 11:57:09 vps791338 shorewall[3048]: Compiling Accept Source Routing...
Mar  4 11:57:09 vps791338 shorewall[3048]: Compiling /etc/shorewall/tcinterfaces...
Mar  4 11:57:09 vps791338 shorewall[3048]: Compiling /etc/shorewall/snat...
Mar  4 11:57:09 vps791338 shorewall[3048]: Compiling MAC Filtration -- Phase 1...
Mar  4 11:57:09 vps791338 shorewall[3048]: Compiling /etc/shorewall/rules...
Mar  4 11:57:09 vps791338 shorewall[3048]: Compiling /etc/shorewall/conntrack...
Mar  4 11:57:09 vps791338 shorewall[3048]: Compiling MAC Filtration -- Phase 2...
Mar  4 11:57:09 vps791338 shorewall[3048]: Compiling /etc/shorewall/stoppedrules...
Mar  4 11:57:09 vps791338 shorewall[3048]: Shorewall configuration compiled to /var/lib/shorewall/.reload
Mar  4 11:57:09 vps791338 shorewall[3048]: Reloading Shorewall....
Mar  4 11:57:09 vps791338 shorewall[3048]: Initializing...
[... cut ...]
Mar  4 13:48:41 vps791338 ss-server[445]: remote recv: Connection reset by peer
Mar  4 13:48:54 vps791338 ss-server[445]: getpeername: Transport endpoint is not connected
Mar  4 13:49:48 vps791338 ss-server[445]: getpeername: Transport endpoint is not connected
Mar  4 13:50:39 vps791338 ss-server[445]: getpeername: Transport endpoint is not connected
Mar  4 13:50:39 vps791338 ss-server[445]: remote recv: Connection reset by peer
Mar  4 13:51:32 vps791338 ss-server[445]: getpeername: Transport endpoint is not connected

florianlog, 2020-03-04:
J'ai vu que le mise à jour RC 0.54rc9 existait mettre mon routeur à cette version pourrais peut être corriger mon problème ?

Ysurac, 2020-03-04:
Tu peux toujours essayer.
Mais vu le log ce n'est pas que le VPN qui fonctionne mal.

florianlog, 2020-03-04:
ah pas bon

florianlog, 2020-03-04:
Ce serait un problème de stabilité de mes box ou un problème de configuration refaire de zéro ?

Je viens de faire la MAJ, en effet idem lenteurs et déco.

Ysurac, 2020-03-04:
Peut être un peu des deux. Mais déjà je commencerais par repartir de zéro.

florianlog, 2020-03-04:
Bon après moultes recherches, l'appel de débit de caméras placées sur le réseau me font tombé le VPN dès que je m'y connecte, sans cela tout fonctionne bien

florianlog, 2020-03-05:
Hello, I summarize my installation:

- WAN 1: Freebox - 15mb / s
- WAN 2: 4g router - 6mb / s (3G)
Connected on a WRT3200ACM with OMR 0.54 + VPS OVH 0.1012
All VPS and OMR ports are redirected to a PFSENSE.

The connected equipment is industrial PLCs and 2 IP cameras.

When I connect to my automatons everything works slowly but works VPN understood but when I want to view the cameras the VPN cuts immediately causing the speed to drop.

vinceducat, 2020-03-06:
I also in charge of automation in a factory, i hope this are very well secure :-)

perhaps not a good idea to use OMR in production even if it is a good system...

closed 2020-09-14 by github-actions[bot]