Ysurac/openmptcprouter #873: Can't get publie IP address from ShadowSocks

Scubad1v3r, 2020-03-07:
## Actual Behavior

Le VPS affiche "Can't get publie IP address from ShadowSocks"
## Steps to Reproduce the Problem

Problème survenu le 5/02 en version 0.53
MàJ VPS et OMR
Vérifications des clés
Le message est toujours présent

## Specifications

  - OpenMPTCProuter version: 0.54rc12
  - OpenMPTCProuter VPS version: 0.1012 4.19.80-mptcp
  - OpenMPTCProuter platform: x86_64

[image]

Ysurac, 2020-03-07:
I need system log, in State->System log and log from the VPS, in ```/var/log/daemon.log```

Scubad1v3r, 2020-03-07:
System log
`Sat Mar  7 15:00:49 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:00:54 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:00:59 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:01:04 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:01:06 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:01:09 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:01:17 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:01:22 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:01:26 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:01:28 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:01:33 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:01:38 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:01:43 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:01:48 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:01:53 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:01:53 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:01:58 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:02:03 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:02:08 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:02:14 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:02:19 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:02:24 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:02:29 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:02:29 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:02:34 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:02:39 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:02:44 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:02:49 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:02:50 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:02:54 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:03:00 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:03:05 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:03:10 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:03:12 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:03:15 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:03:20 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:03:25 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:03:30 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:03:33 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:03:35 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:03:40 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:03:46 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:03:51 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:03:56 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:04:01 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:04:05 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:04:06 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:04:11 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:04:16 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:04:21 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:04:26 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:04:30 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:04:32 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:04:37 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:04:42 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:04:47 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:04:52 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
[... cut ...]
Sat Mar  7 15:49:13 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:49:17 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:49:18 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:49:23 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:49:28 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected
Sat Mar  7 15:49:33 2020 daemon.err /usr/bin/ss-local[20319]: getpeername: Socket not connected`

Scubad1v3r, 2020-03-07:
et voici le contenu du demain.log
[daemon.txt](https://github.com/Ysurac/openmptcprouter/files/4302091/daemon.txt)

Ysurac, 2020-03-07:
I don't think the "no space left on device" in daemon.log is really something good.
Same for the "cannot allocate memory".
What is this VPS from ?

Scubad1v3r, 2020-03-07:
VPS Milkywan, je comprends pas le "pas d'espace restant sur le périphérique" 
`df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            484M     0  484M   0% /dev
tmpfs           100M   11M   89M  11% /run
/dev/vda1        30G  3.7G   25G  14% /
tmpfs           496M     0  496M   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           496M     0  496M   0% /sys/fs/cgroup
tmpfs           100M     0  100M   0% /run/user/1004`

Ysurac, 2020-03-07:
Oui mais visiblement y'a eu un soucis le 6 mars  entre 18h15 et 19h03 selon le log.
Sinon je comprend pas pourquoi tu as des logs liés à mlvpn, tu as changé de VPN ?
Il me faudrait la config pour voir où sont les soucis (dans Système->OpenMPTCProuter puis onglet "voir tous les paramètres" en enlevant les clefs)

Scubad1v3r, 2020-03-07:
le 6 j'ai eu le problème qui est apparru.
pour essayer de le resoudre j'ai mis a jour le vps vers 0.1012 mais ca n'a rien changé.
ensuite le 7 j'ai mis a jour l'OMR 0.54rc12

Ysurac, 2020-03-07:
Le VPS a été redémarré après la mise à jour ?

Scubad1v3r, 2020-03-07:
oui

Ysurac, 2020-03-07:
Il me faudrait le résultat de ```ps aux | grep ss``` sur le VPS.
Ainsi que la configuration (sans la clef) dans ```/etc/shadowsocks-libev/manager.json```, toujours sur le VPS.

Scubad1v3r, 2020-03-07:
ps aux | grep ss
message+   201  0.0  0.1   8980  1308 ?        Ss   Mar06   0:01 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only
root       427  0.0  0.2  15768  2436 ?        Ss   Mar06   0:00 /usr/sbin/sshd -D
root      9583  0.2  0.7  16692  7800 ?        Ss   22:22   0:00 sshd: olivier [priv]
olivier   9592  0.0  0.4  16692  4732 ?        S    22:22   0:00 sshd: olivier@pts/1
olivier   9631  0.0  0.0   6144   824 pts/1    S+   22:22   0:00 grep ss
root     29576  0.0  0.7  16692  7920 ?        Ss   16:51   0:00 sshd: olivier [priv]
olivier  29665  0.0  0.5  17288  5840 ?        S    16:52   0:00 sshd: olivier@pts/0
root     30781  0.0  0.0   7564   308 ?        Ss   Mar06   0:00 /usr/bin/ss-manager -c /etc/shadowsocks-libev/manager.json

Scubad1v3r, 2020-03-07:
cat /etc/shadowsocks-libev/manager.json
{
    "mptcp": true,
    "ipv6_first": true,
    "no_delay": true,
    "ebpf": false,
    "server": [
        "[::0]",
        "0.0.0.0"
    ],
    "method": "chacha20-ietf-poly1305",
    "fast_open": true,
    "timeout": 600,
    "port_key": {
        "65101": " ",
    },
    "local_port": 1081,
    "verbose": 0,
    "acl": "/etc/shadowsocks-libev/local.acl",
    "mode": "tcp_and_udp",
    "reuse_port": true,
    "prefer_ipv6": false

Ysurac, 2020-03-07:
Manque un "ss-server" au niveau du ps aux.
Le manager.json est complet ? il manque pas le "}" à la fin ?
Que donne un ```systemctl restart shadowsocks-libev-manager@manager``` sur le VPS ?

Scubad1v3r, 2020-03-07:
xxxx@xxxx:~$ ps aux | grep ss
message+   201  0.0  0.1   8980  1308 ?        Ss   Mar06   0:01 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only
root       427  0.0  0.2  15768  2436 ?        Ss   Mar06   0:00 /usr/sbin/sshd -D
olivier  13515  0.0  0.0   6144   876 pts/0    S+   22:38   0:00 grep ss
root     29576  0.0  0.7  16692  7920 ?        Ss   16:51   0:00 sshd: olivier [priv]
olivier  29665  0.0  0.5  17288  5840 ?        S    16:52   0:00 sshd: olivier@pts/0
root     30781  0.0  0.0   7564   308 ?        Ss   Mar06   0:00 /usr/bin/ss-manager -c /etc/shadowsocks-libev/manager.json
olivier@olivierg:~$ cat /etc/shadowsocks-libev/manager.json
{
    "mptcp": true,
    "ipv6_first": true,
    "no_delay": true,
    "ebpf": false,
    "server": [
        "[::0]",
        "0.0.0.0"
    ],
    "method": "chacha20-ietf-poly1305",
    "fast_open": true,
    "timeout": 600,
    "port_key": {
        "65101": "            ",
    },
    "local_port": 1081,
    "verbose": 0,
    "acl": "/etc/shadowsocks-libev/local.acl",
    "mode": "tcp_and_udp",
    "reuse_port": true,
    "prefer_ipv6": false
xxxx@xxxx:~$

Scubad1v3r, 2020-03-07:
olivier@olivierg:~$ ps aux | grep ss
message+   201  0.0  0.1   8980  1308 ?        Ss   Mar06   0:01 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only
root       427  0.0  0.2  15768  2436 ?        Ss   Mar06   0:00 /usr/sbin/sshd -D
olivier  13515  0.0  0.0   6144   876 pts/0    S+   22:38   0:00 grep ss
root     29576  0.0  0.7  16692  7920 ?        Ss   16:51   0:00 sshd: olivier [priv]
olivier  29665  0.0  0.5  17288  5840 ?        S    16:52   0:00 sshd: olivier@pts/0
root     30781  0.0  0.0   7564   308 ?        Ss   Mar06   0:00 /usr/bin/ss-manager -c /etc/shadowsocks-libev/manager.json
olivier@olivierg:~$ cat /etc/shadowsocks-libev/manager.json
{
    "mptcp": true,
    "ipv6_first": true,
    "no_delay": true,
    "ebpf": false,
    "server": [
        "[::0]",
        "0.0.0.0"
    ],
    "method": "chacha20-ietf-poly1305",
    "fast_open": true,
    "timeout": 600,
    "port_key": {
        "65101": "  ",
    },
    "local_port": 1081,
    "verbose": 0,
    "acl": "/etc/shadowsocks-libev/local.acl",
    "mode": "tcp_and_udp",
    "reuse_port": true,
    "prefer_ipv6": false
olivier@olivierg:~$

Scubad1v3r, 2020-03-07:
je l'ai refait mais c'est pareil

Scubad1v3r, 2020-03-07:
systemctl restart shadowsocks-libev-manager@manager
Failed to restart shadowsocks-libev-manager@manager.service: The name org.freedesktop.PolicyKit1 was not provided by any .service files
See system logs and 'systemctl status shadowsocks-libev-manager@manager.service' for details.

Scubad1v3r, 2020-03-07:
il faut rajouter le  "}" à la fin du fichier manager.json?

Ysurac, 2020-03-07:
ah il est vraiment pas présent ? donc oui il doit être présent.
Sinon ça semble bon.

Scubad1v3r, 2020-03-07:
Si en fait il y est juste devant l'invite...
J'avais pas fait attention

Ysurac, 2020-03-08:
La commande ```systemctl restart shadowsocks-libev-manager@manager``` a été lancée en root ?
sinon ```sudo systemctl restart shadowsocks-libev-manager@manager```

Scubad1v3r, 2020-03-08:
non je viens de le faire.
Ca semblerait fonctionner de nouveau
Merci beaucoup Ycarus
Tu peux m'expliquer ce qui s'est passé?

Scubad1v3r, 2020-03-09:
Ce matin j'avais de nouveau le message : "Can't get publie IP address from ShadowSocks"

Obligé de refaire
>  sudo systemctl restart shadowsocks-libev-manager@manager
 pour qu'il disparaisse

Je comprends pas pourquoi ca se déconnecte tout seul

Ysurac, 2020-03-09:
Il faudrait voir dans les logs du VPS, dans ```/var/log/daemon.log``` et eventuellement via ```dmesg``` pourquoi shadowsocks plante ou est tué.

Scubad1v3r, 2020-03-09:
[daemon.txt](https://github.com/Ysurac/openmptcprouter/files/4305922/daemon.txt)
[dmesg.txt](https://github.com/Ysurac/openmptcprouter/files/4305933/dmesg.txt)
Voila les deux logs

closed 2020-03-12 by Scubad1v3r