Ysurac/openmptcprouter #874: Squashfs RPI3 and RPI4

MrTheSkream, 2020-03-08:
Hello,

I have a problem with raspberry pi 4 squashfs.

With Raspberry pi 3 squashfs the system is on RW and with RPI 4 squashfs the system is in RO.

It's normal ?

Thanks.

Ysurac, 2020-03-08:
No, but this happen often. It's why ext4 is recommended.

MrTheSkream, 2020-03-08:
Yes i know but ext4 is is sensitive to power failure.
And my system instantly shuts down the raspberry when I press the button..

Ysurac, 2020-03-08:
True, but in fact squashfs doesn't really work better on a RPI.

MrTheSkream, 2020-03-08:
Okey thanks.
It works perfectly on raspberry pi 3 with squashfs so I will continue to use with the old version of raspberry, as long as it works.

Ysurac, 2020-03-08:
maybe you can test the release candidate, this should be the latest and I hope to release a new final version tomorrow.

closed 2020-05-18 by Ysurac