Ysurac/openmptcprouter #875: VPN IPV6 tunnel DOWN

vinceducat, 2020-03-08:
## Expected Behavior

VPN IPV6 tunnel UP

## Actual Behavior
VPN IPV6 tunnel DOWN

## Steps to Reproduce the Problem

  1.install omr v054rc15
  2.tune on ipv6
  3.

## Specifications

  - OpenMPTCProuter version: 0.54rc15
  - OpenMPTCProuter VPS version: 1012
  - OpenMPTCProuter platform: RPI3

my VPN IPV4 is ok
glorytun used

vinceducat, 2020-03-08:
6in4-omr6in4@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 1280 qdisc noqueue state UNKNOWN group default qlen 1000
    link/sit 10.255.255.2 peer 10.255.255.1
    inet6 fe80::a00:2/128 scope link
       valid_lft forever preferred_lft forever
    inet6 fe80::aff:ff02/64 scope link
       valid_lft forever preferred_lft forever

on OMR

vinceducat, 2020-03-08:
: gt-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc prio state UNKNOWN group default qlen 500
    link/none
    inet 10.255.255.1/30 brd 10.255.255.3 scope global gt-tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::994a:95db:1ed0:9350/64 scope link stable-privacy
       valid_lft forever preferred_lft forever

on vps

vinceducat, 2020-03-08:
eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc prio state UP group default qlen 1000
    link/ether ce:0c:4f:44:7c:da brd ff:ff:ff:ff:ff:ff
    inet 45.xxx.xxx.xxx/32 brd 45.13.104.183 scope global eth0:1
       valid_lft forever preferred_lft forever
    inet6 2a0b:x:x:1::3d/64 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::cc0c:4fff:fe44:7cda/64 scope link
       valid_lft forever preferred_lft forever

on vps

Ysurac, 2020-03-08:
```systemctl -a | grep omr``` donne quoi sur le VPS ?

vinceducat, 2020-03-08:
systemctl -a | grep omr
  glorytun-tcp@omr.service                                                    loaded    inactive dead      Glorytun TCP on omr                  
  glorytun-udp@omr.service                                                    loaded    inactive dead      Glorytun UDP on omr                  
  omr-admin.service                                                           loaded    active   running   OMR-Admin                            
  omr.service                                                                 loaded    active   running   OMR                                  
● omr6in4@user1.service                                                       loaded    failed   failed    OMR6IN4 on user1                     
  system-omr6in4.slice                                                        loaded    active   active    system-omr6in4.slice

vinceducat, 2020-03-08:
oups semble dead, peux etre le corona :-)

vinceducat, 2020-03-08:
cat /var/log/daemon.log | grep omr
Mar  8 15:24:46 vincentm systemd-networkd[185]: omr-6in4-user0: Gained IPv6LL
Mar  8 15:24:46 vincentm systemd-networkd[185]: omr-6in4-user0: Gained carrier
Mar  8 15:24:46 vincentm systemd-networkd[185]: omr-6in4-user0: Lost carrier
Mar  8 15:24:46 vincentm systemd-networkd[185]: omr-6in4-user0: Gained IPv6LL
Mar  8 15:24:46 vincentm systemd-networkd[185]: omr-6in4-user0: Gained carrier
Mar  8 15:24:46 vincentm systemd[1]: omr6in4@user0.service: Succeeded.
Mar  8 16:20:33 vincentm systemd-networkd[185]: omr-6in4-user0: Lost carrier
Mar  8 16:20:33 vincentm systemd-networkd[185]: omr-6in4-user0: Gained IPv6LL
Mar  8 16:20:33 vincentm systemd-networkd[185]: omr-6in4-user0: Gained carrier
Mar  8 16:20:33 vincentm systemd-networkd[185]: omr-6in4-user0: Lost carrier
Mar  8 16:20:33 vincentm systemd-networkd[185]: omr-6in4-user0: Gained IPv6LL
Mar  8 16:20:33 vincentm systemd-networkd[185]: omr-6in4-user0: Gained carrier
Mar  8 16:20:33 vincentm systemd[1]: omr6in4@user0.service: Succeeded.
Mar  8 16:29:13 vincentm systemd[1]: Removed slice system-omr6in4.slice.
Mar  8 16:31:23 vincentm omr-6in4-run[442]: usage: omr-6in4-run FILE
Mar  8 16:31:23 vincentm systemd[1]: omr6in4@user1.service: Main process exited, code=exited, status=1/FAILURE
Mar  8 16:31:23 vincentm systemd[1]: omr6in4@user1.service: Failed with result 'exit-code'.
Mar  8 17:07:13 vincentm systemd[1]: Removed slice system-omr6in4.slice.
Mar  8 17:09:26 vincentm omr-6in4-run[401]: usage: omr-6in4-run FILE
Mar  8 17:09:26 vincentm systemd[1]: omr6in4@user1.service: Main process exited, code=exited, status=1/FAILURE
Mar  8 17:09:26 vincentm systemd[1]: omr6in4@user1.service: Failed with result 'exit-code'.

vinceducat, 2020-03-08:
cat /var/log/daemon.log | grep glory
Mar  8 00:02:59 vincentm glorytun-tcp-run[786]: 78.232.176.9.55007: connected
Mar  8 00:04:55 vincentm glorytun-tcp-run[786]: 78.232.176.9.55839: connected
Mar  8 00:05:08 vincentm glorytun-tcp-run[786]: 78.232.176.9.56023: connected
Mar  8 00:45:28 vincentm glorytun-tcp-run[786]: 78.232.176.9.24635: connected
Mar  8 00:45:39 vincentm glorytun-tcp-run[786]: 78.232.176.9.24703: connected
Mar  8 08:10:36 vincentm glorytun-tcp-run[786]: read: Connection timed out
Mar  8 08:14:36 vincentm glorytun-tcp-run[786]: 78.232.176.9.39629: connected
Mar  8 12:46:18 vincentm glorytun-tcp-run[786]: read: Connection timed out
Mar  8 12:49:02 vincentm glorytun-tcp-run[786]: 78.232.176.9.16179: connected
Mar  8 15:05:44 vincentm glorytun-tcp-run[786]: 78.232.176.9.60187: connected
Mar  8 15:05:57 vincentm glorytun-tcp-run[786]: 78.232.176.9.60379: connected
Mar  8 15:22:05 vincentm glorytun-tcp-run[786]: 78.232.176.9.10797: connected
Mar  8 15:22:38 vincentm glorytun-tcp-run[786]: 78.232.176.9.11111: connected
Mar  8 16:20:14 vincentm glorytun-tcp-run[786]: 78.232.176.9.27981: connected
Mar  8 16:20:40 vincentm glorytun-tcp-run[786]: 78.232.176.9.28491: connected
Mar  8 16:20:53 vincentm glorytun-tcp-run[786]: 78.232.176.9.28695: connected
Mar  8 16:27:38 vincentm glorytun-tcp-run[786]: 78.232.176.9.37089: connected
Mar  8 16:27:49 vincentm glorytun-tcp-run[786]: 78.232.176.9.37299: connected
Mar  8 16:31:23 vincentm glorytun-udp-run[459]: running on device gt-udp-tun0 as pid 459
Mar  8 16:31:24 vincentm glorytun-tcp-run[443]: 78.232.176.9.39061: connected
Mar  8 16:31:24 vincentm glorytun-tcp-run[443]: tun write: Input/output error
Mar  8 16:31:24 vincentm glorytun-tcp-run[443]: INITIALIZED gt-tun0
Mar  8 16:31:24 vincentm glorytun-tcp-run[443]: STARTED gt-tun0
Mar  8 16:31:24 vincentm glorytun-tcp-run[443]: STOPPED gt-tun0
Mar  8 16:31:24 vincentm systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Mar  8 16:31:25 vincentm systemd[1]: glorytun-tcp@tun0.service: Service RestartSec=100ms expired, scheduling restart.
Mar  8 16:31:25 vincentm systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 1.
Mar  8 16:31:26 vincentm glorytun-tcp-run[807]: 78.232.176.9.39067: connected
Mar  8 16:31:26 vincentm glorytun-tcp-run[807]: tun write: Input/output error
Mar  8 16:31:26 vincentm glorytun-tcp-run[807]: INITIALIZED gt-tun0
Mar  8 16:31:26 vincentm glorytun-tcp-run[807]: STARTED gt-tun0
Mar  8 16:31:26 vincentm glorytun-tcp-run[807]: STOPPED gt-tun0
Mar  8 16:31:26 vincentm systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Mar  8 16:31:26 vincentm systemd[1]: glorytun-tcp@tun0.service: Service RestartSec=100ms expired, scheduling restart.
Mar  8 16:31:26 vincentm systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 2.
Mar  8 16:31:27 vincentm glorytun-tcp-run[853]: 78.232.176.9.39069: connected
Mar  8 16:33:32 vincentm glorytun-tcp-run[853]: 78.232.176.9.15177: connected
Mar  8 16:42:34 vincentm glorytun-tcp-run[853]: 78.232.176.9.18559: connected
Mar  8 16:42:50 vincentm glorytun-tcp-run[853]: 78.232.176.9.18647: connected
Mar  8 16:43:26 vincentm glorytun-tcp-run[853]: 78.232.176.9.18923: connected
Mar  8 16:43:42 vincentm glorytun-tcp-run[853]: 78.232.176.9.19057: connected
Mar  8 16:45:35 vincentm glorytun-tcp-run[853]: 78.232.176.9.19781: connected
Mar  8 16:45:48 vincentm glorytun-tcp-run[853]: 78.232.176.9.19969: connected
Mar  8 16:49:12 vincentm glorytun-tcp-run[853]: 78.232.176.9.21763: connected
Mar  8 16:49:48 vincentm glorytun-tcp-run[853]: 78.232.176.9.22051: connected
Mar  8 16:53:05 vincentm glorytun-tcp-run[853]: 78.232.176.9.28385: connected
Mar  8 17:01:36 vincentm glorytun-tcp-run[853]: 78.232.176.9.30647: connected
Mar  8 17:01:47 vincentm glorytun-tcp-run[853]: 78.232.176.9.30751: connected
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: INITIALIZED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
[... 9 lines cut ...]
Mar  8 17:09:27 vincentm glorytun-tcp-run[402]: 78.232.176.9.33447: shutdown
Mar  8 17:09:27 vincentm glorytun-tcp-run[402]: tun write: Input/output error
Mar  8 17:09:27 vincentm glorytun-tcp-run[402]: tun write: Input/output error
[... cut ...]
Mar  8 18:49:47 vincentm glorytun-tcp-run[794]: 78.232.176.9.10251: connected
Mar  8 19:03:37 vincentm glorytun-tcp-run[794]: 78.232.176.9.14019: connected
Mar  8 19:03:47 vincentm glorytun-tcp-run[794]: 78.232.176.9.14091: connected
Mar  8 19:05:12 vincentm glorytun-tcp-run[794]: 78.232.176.9.14453: connected
Mar  8 19:05:25 vincentm glorytun-tcp-run[794]: 78.232.176.9.14545: connected
Mar  8 19:08:12 vincentm glorytun-tcp-run[794]: 78.232.176.9.61675: connected
Mar  8 19:17:55 vincentm glorytun-tcp-run[794]: 78.232.176.9.42841: connected

vinceducat, 2020-03-08:
cat /var/log/daemon.log | grep glory
Mar  8 00:02:59 vincentm glorytun-tcp-run[786]: 78xx.xx.xx.55007: connected
Mar  8 00:04:55 vincentm glorytun-tcp-run[786]: 78xx.xx.xx.55839: connected
Mar  8 00:05:08 vincentm glorytun-tcp-run[786]: 78xx.xx.xx.56023: connected
Mar  8 00:45:28 vincentm glorytun-tcp-run[786]: 78xx.xx.xx.24635: connected
Mar  8 00:45:39 vincentm glorytun-tcp-run[786]: 78xx.xx.xx.24703: connected
Mar  8 08:10:36 vincentm glorytun-tcp-run[786]: read: Connection timed out
Mar  8 08:14:36 vincentm glorytun-tcp-run[786]: 78xx.xx.xx.39629: connected
Mar  8 12:46:18 vincentm glorytun-tcp-run[786]: read: Connection timed out
Mar  8 12:49:02 vincentm glorytun-tcp-run[786]: 78xx.xx.xx.16179: connected
Mar  8 15:05:44 vincentm glorytun-tcp-run[786]: 78xx.xx.xx.60187: connected
Mar  8 15:05:57 vincentm glorytun-tcp-run[786]: 78xx.xx.xx.60379: connected
Mar  8 15:22:05 vincentm glorytun-tcp-run[786]: 78xx.xx.xx.10797: connected
Mar  8 15:22:38 vincentm glorytun-tcp-run[786]: 78xx.xx.xx.11111: connected
Mar  8 16:20:14 vincentm glorytun-tcp-run[786]: 78xx.xx.xx.27981: connected
Mar  8 16:20:40 vincentm glorytun-tcp-run[786]: 78xx.xx.xx.28491: connected
Mar  8 16:20:53 vincentm glorytun-tcp-run[786]: 78xx.xx.xx.28695: connected
Mar  8 16:27:38 vincentm glorytun-tcp-run[786]: 78xx.xx.xx.37089: connected
Mar  8 16:27:49 vincentm glorytun-tcp-run[786]: 78xx.xx.xx.37299: connected
Mar  8 16:31:23 vincentm glorytun-udp-run[459]: running on device gt-udp-tun0 as pid 459
Mar  8 16:31:24 vincentm glorytun-tcp-run[443]: 78xx.xx.xx.39061: connected
Mar  8 16:31:24 vincentm glorytun-tcp-run[443]: tun write: Input/output error
Mar  8 16:31:24 vincentm glorytun-tcp-run[443]: INITIALIZED gt-tun0
Mar  8 16:31:24 vincentm glorytun-tcp-run[443]: STARTED gt-tun0
Mar  8 16:31:24 vincentm glorytun-tcp-run[443]: STOPPED gt-tun0
Mar  8 16:31:24 vincentm systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Mar  8 16:31:25 vincentm systemd[1]: glorytun-tcp@tun0.service: Service RestartSec=100ms expired, scheduling restart.
Mar  8 16:31:25 vincentm systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 1.
Mar  8 16:31:26 vincentm glorytun-tcp-run[807]: 78xx.xx.xx.39067: connected
Mar  8 16:31:26 vincentm glorytun-tcp-run[807]: tun write: Input/output error
Mar  8 16:31:26 vincentm glorytun-tcp-run[807]: INITIALIZED gt-tun0
Mar  8 16:31:26 vincentm glorytun-tcp-run[807]: STARTED gt-tun0
Mar  8 16:31:26 vincentm glorytun-tcp-run[807]: STOPPED gt-tun0
Mar  8 16:31:26 vincentm systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Mar  8 16:31:26 vincentm systemd[1]: glorytun-tcp@tun0.service: Service RestartSec=100ms expired, scheduling restart.
Mar  8 16:31:26 vincentm systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 2.
Mar  8 16:31:27 vincentm glorytun-tcp-run[853]: 78xx.xx.xx.39069: connected
Mar  8 16:33:32 vincentm glorytun-tcp-run[853]: 78xx.xx.xx.15177: connected
Mar  8 16:42:34 vincentm glorytun-tcp-run[853]: 78xx.xx.xx.18559: connected
Mar  8 16:42:50 vincentm glorytun-tcp-run[853]: 78xx.xx.xx.18647: connected
Mar  8 16:43:26 vincentm glorytun-tcp-run[853]: 78xx.xx.xx.18923: connected
Mar  8 16:43:42 vincentm glorytun-tcp-run[853]: 78xx.xx.xx.19057: connected
Mar  8 16:45:35 vincentm glorytun-tcp-run[853]: 78xx.xx.xx.19781: connected
Mar  8 16:45:48 vincentm glorytun-tcp-run[853]: 78xx.xx.xx.19969: connected
Mar  8 16:49:12 vincentm glorytun-tcp-run[853]: 78xx.xx.xx.21763: connected
Mar  8 16:49:48 vincentm glorytun-tcp-run[853]: 78xx.xx.xx.22051: connected
Mar  8 16:53:05 vincentm glorytun-tcp-run[853]: 78xx.xx.xx.28385: connected
Mar  8 17:01:36 vincentm glorytun-tcp-run[853]: 78xx.xx.xx.30647: connected
Mar  8 17:01:47 vincentm glorytun-tcp-run[853]: 78xx.xx.xx.30751: connected
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: INITIALIZED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STARTED gt-tun0
Mar  8 17:07:13 vincentm glorytun-tcp-run[853]: STOPPED gt-tun0
[... 9 lines cut ...]
Mar  8 17:09:27 vincentm glorytun-tcp-run[402]: 78xx.xx.xx.33447: shutdown
Mar  8 17:09:27 vincentm glorytun-tcp-run[402]: tun write: Input/output error
Mar  8 17:09:27 vincentm glorytun-tcp-run[402]: tun write: Input/output error
Mar  8 17:09:27 vincentm glorytun-tcp-run[402]: tun write: Input/output error
[... cut ...]
Mar  8 18:49:47 vincentm glorytun-tcp-run[794]: 78xx.xx.xx.10251: connected
Mar  8 19:03:37 vincentm glorytun-tcp-run[794]: 78xx.xx.xx.14019: connected
Mar  8 19:03:47 vincentm glorytun-tcp-run[794]: 78xx.xx.xx.14091: connected
Mar  8 19:05:12 vincentm glorytun-tcp-run[794]: 78xx.xx.xx.14453: connected
Mar  8 19:05:25 vincentm glorytun-tcp-run[794]: 78xx.xx.xx.14545: connected
Mar  8 19:08:12 vincentm glorytun-tcp-run[794]: 78xx.xx.xx.61675: connected
Mar  8 19:17:55 vincentm glorytun-tcp-run[794]: 78xx.xx.xx.42841: connected

Ysurac, 2020-03-08:
Le service devrait être omr6in4@user0. Il y a une erreur dans le script, j'ai corrigé dans le snapshot et je sortirais la nouvelle version demain.

vinceducat, 2020-03-08:
ok thx

i have also 

Dead loop on virtual device 6in4-omr6in4, fix it urgently!
[   65.909281] Dead loop on virtual device 6in4-omr6in4, fix it urgently!

in kernel OMR log

vinceducat, 2020-03-08:
thanks a lot for this great work!!!

vinceducat, 2020-03-09:
install last 0.54 on OMR but always the same probleme.

even i f y resart OMR and VPS :-(
systemctl -a | grep omr
  glorytun-tcp@omr.service                                                    loaded    inactive dead      Glorytun TCP on omr                  
  glorytun-udp@omr.service                                                    loaded    inactive dead      Glorytun UDP on omr                  
  omr-admin.service                                                           loaded    active   running   OMR-Admin                            
  omr.service                                                                 loaded    active   running   OMR                                  
● omr6in4@user1.service                                                       loaded    failed   failed    OMR6IN4 on user1                     
  system-omr6in4.slice                                                        loaded    active   active    system-omr6in4.slice

vinceducat, 2020-03-09:
toujours user1 does i update VPS?

Ysurac, 2020-03-09:
v0.54 is not yet released.
snapshot VPS script is needed.

vinceducat, 2020-03-09:
oups so why i find the 0.54 in the https server from today :-( 
?

need i reinstall 0.54rc15 but where is it ?

vinceducat, 2020-03-09:
reinstall 0.54rc15 and now 

systemctl -a | grep omr
  sys-devices-virtual-net-omr\x2d6in4\x2duser0.device                         loaded    active   plugged   /sys/devices/virtual/net/omr-6in4-user0
  sys-subsystem-net-devices-omr\x2d6in4\x2duser0.device                       loaded    active   plugged   /sys/subsystem/net/devices/omr-6in4-user0
  glorytun-tcp@omr.service                                                    loaded    inactive dead      Glorytun TCP on omr                  
  glorytun-udp@omr.service                                                    loaded    inactive dead      Glorytun UDP on omr                  
  omr-admin.service                                                           loaded    active   running   OMR-Admin                            
  omr.service                                                                 loaded    active   running   OMR                                  
● omr6in4@user1.service                                                       loaded    failed   failed    OMR6IN4 on user1                     
  system-omr6in4.slice                                                        loaded    active   active    system-omr6in4.slice     

strange no?

vinceducat, 2020-03-09:
[image]

Ysurac, 2020-03-09:
v0.54 is compiled, but it's in testing before release.
VPS need to be on snapshot release to work with IPv6 https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

vinceducat, 2020-03-09:
so i have to install trought wget -O - http://www.openmptcprouter.com/server-test/debian9-x86_64.sh | sh the last VPS version ?

seems to be 1006-test in the file is it right ?

Ysurac, 2020-03-09:
Yes, I don't increase number in snapshot release.

vinceducat, 2020-03-09:
can i install snapshot and after when new release come revert back to the non snapshot version ?

vinceducat, 2020-03-09:
v1006-test1 and 0.54 seems to be ok

is it good ???

 systemctl -a | grep omr
  sys-devices-virtual-net-omr\x2d6in4\x2duser0.device                         loaded    active   plugged   /sys/devices/virtual/net/omr-6in4-user0
  sys-subsystem-net-devices-omr\x2d6in4\x2duser0.device                       loaded    active   plugged   /sys/subsystem/net/devices/omr-6in4-user0
  glorytun-tcp@omr.service                                                    loaded    inactive dead      Glorytun TCP on omr                  
  glorytun-udp@omr.service                                                    loaded    inactive dead      Glorytun UDP on omr                  
  omr-admin.service                                                           loaded    active   running   OMR-Admin                            
  omr.service                                                                 loaded    active   running   OMR                                  
  omr6in4@user0.service                                                       loaded    active   exited    OMR6IN4 on user0                     
● omr6in4@user1.service                                                       loaded    failed   failed    OMR6IN4 on user1                     
  system-omr6in4.slice                                                        loaded    active   active    system-omr6in4.slice

vinceducat, 2020-03-09:
i have a French FAI box, freebox v5 and i seems that ipv6 traffic goes directely trought the box...
ipv6 on web site flow test is the ipv6 of my FAI not the ipv6 of the vps...

vinceducat, 2020-03-10:
OMR 0.54 VPS 1013 all is ok :-)

thanks...

closed 2020-03-10 by Ysurac