Ysurac/openmptcprouter #876: ShadowSocks cannot get the IP from the server

Archana7700, 2020-03-09:
What configurations needs to done for the shadowsocks at the client side?

Even i have the checked latest issues where same issue was there "Can not get IP address from published ShadowSocks # 873" but i didn't get the  daemon "log from the VPS, in /var/log/daemon.log" at the server side 

> system log is:
Mon Mar  9 05:28:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Mar  9 05:28:08 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=TMP_IFB_4_SQM, subsystem=net, sysfspath=/sys/devices/virtual/net/TMP_IFB_4_SQM
Mon Mar  9 05:28:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Mon Mar  9 05:28:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Mar  9 05:28:08 2020 daemon.warn [3776]: <warn>  Invalid sysfs path read for net/TMP_IFB_4_SQM
Mon Mar  9 05:28:09 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.16.1 dev eth0
Mon Mar  9 05:28:09 2020 user.notice post-tracking-post-tracking: Set server vps (14.98.145.59) route via 192.168.16.1 metric 4
Mon Mar  9 05:28:09 2020 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Mar  9 05:28:09 2020 daemon.err omr-tracker[9345]: Command failed: Not found
Mon Mar  9 05:28:09 2020 user.err vnstat_backup: cannot restore, backup file does not exist (yet)
Mon Mar  9 05:28:09 2020 user.notice adb-enablemodem: INFO: waiting for device
Mon Mar  9 05:28:09 2020 daemon.notice procd: /etc/rc.d/S99omr-6in4: net.ipv6.conf.all.disable_ipv6 = 1
Mon Mar  9 05:28:09 2020 user.notice omr-6in4: Disable IPv6
Mon Mar  9 05:28:09 2020 user.notice omr-bypass: Starting OMR-ByPass...
Mon Mar  9 05:28:09 2020 user.info : plugin "/usr/bin/v2ray-plugin" enabled
Mon Mar  9 05:28:09 2020 user.info : plugin "/usr/bin/v2ray-plugin" enabled
Mon Mar  9 05:28:09 2020 user.info : using tcp fast open
Mon Mar  9 05:28:09 2020 user.info : using tcp fast open
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11856]: enable TCP no-delay
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11854]: enable TCP no-delay
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11854]: resolving hostname to IPv6 address first
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11854]: initializing ciphers... chacha20-ietf-poly1305
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11856]: initializing ciphers... chacha20-ietf-poly1305
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11854]: listening at 0.0.0.0:1100
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11856]: listening at 0.0.0.0:1101
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11856]: tcp port reuse enabled
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11854]: tcp port reuse enabled
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11856]: UDP relay enabled
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11854]: UDP relay enabled
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11856]: udp port reuse enabled
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11854]: udp port reuse enabled
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11856]: running from root user
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11854]: running from root user
Mon Mar  9 05:28:09 2020 user.info : plugin "/usr/bin/v2ray-plugin" enabled
Mon Mar  9 05:28:09 2020 user.info : plugin "/usr/bin/v2ray-plugin" enabled
Mon Mar  9 05:28:09 2020 user.info : using tcp fast open
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-local[11853]: using tcp fast open
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11857]: enable TCP no-delay
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-local[11853]: initializing ciphers... chacha20-ietf-poly1305
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11857]: initializing ciphers... chacha20-ietf-poly1305
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11857]: listening at 0.0.0.0:1101
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11857]: tcp port reuse enabled
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-local[11853]: listening at 0.0.0.0:1111
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11857]: UDP relay enabled
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-local[11853]: tcp port reuse enabled
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-local[11853]: udprelay enabled
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11857]: udp port reuse enabled
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-local[11853]: udp port reuse enabled
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11857]: running from root user
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-local[11853]: running from root user
Mon Mar  9 05:28:09 2020 user.info : plugin "/usr/bin/v2ray-plugin" enabled
Mon Mar  9 05:28:09 2020 user.info : using tcp fast open
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11855]: enable TCP no-delay
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11855]: resolving hostname to IPv6 address first
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11855]: initializing ciphers... chacha20-ietf-poly1305
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11855]: listening at 0.0.0.0:1100
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11855]: tcp port reuse enabled
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11855]: UDP relay enabled
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11855]: udp port reuse enabled
Mon Mar  9 05:28:09 2020 daemon.info /usr/bin/ss-redir[11855]: running from root user
Mon Mar  9 05:28:10 2020 user.notice omr-bypass: Restart dnsmasq...
Mon Mar  9 05:28:10 2020 daemon.info dnsmasq[4612]: exiting on receipt of SIGTERM
Mon Mar  9 05:28:10 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Mar  9 05:28:10 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Mar  9 05:28:10 2020 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
[... 697 lines cut ...]
Mon Mar  9 06:03:13 2020 user.notice OMR-VPS: Can't get token, try later
Mon Mar  9 06:03:15 2020 user.notice OMR-VPS: Can't get token, try later
Mon Mar  9 06:03:23 2020 user.notice OMR-VPS: Can't get token, try later
Mon Mar  9 06:03:26 2020 user.notice OMR-VPS: Can't get token, try later
Mon Mar  9 06:03:34 2020 user.notice OMR-VPS: Can't get token, try later
Mon Mar  9 06:03:36 2020 user.notice OMR-VPS: Can't get token, try later
Mon Mar  9 06:03:44 2020 user.notice OMR-VPS: Can't get token, try later
Mon Mar  9 06:03:47 2020 user.notice OMR-VPS: Can't get token, try later
Mon Mar  9 06:03:54 2020 daemon.info dnsmasq-dhcp[18015]: DHCPDISCOVER(eth0) f8:95:ea:7a:cd:2c
Mon Mar  9 06:03:54 2020 daemon.info dnsmasq-dhcp[18015]: DHCPOFFER(eth0) 192.168.16.176 f8:95:ea:7a:cd:2c
Mon Mar  9 06:03:54 2020 daemon.info dnsmasq-dhcp[18015]: DHCPREQUEST(eth0) 192.168.16.201 f8:95:ea:7a:cd:2c
Mon Mar  9 06:03:54 2020 daemon.info dnsmasq-dhcp[18015]: DHCPNAK(eth0) 192.168.16.201 f8:95:ea:7a:cd:2c wrong server-ID
Mon Mar  9 06:03:55 2020 user.notice OMR-VPS: Can't get token, try later
Mon Mar  9 06:03:57 2020 user.notice OMR-VPS: Can't get token, try later
Mon Mar  9 06:04:05 2020 user.notice OMR-VPS: Can't get token, try later
Mon Mar  9 06:04:08 2020 user.notice OMR-VPS: Can't get token, try later
Mon Mar  9 06:04:16 2020 user.notice OMR-VPS: Can't get token, try later
Mon Mar  9 06:04:18 2020 user.notice OMR-VPS: Can't get token, try later
Mon Mar  9 06:04:26 2020 user.notice OMR-VPS: Can't get token, try later
Mon Mar  9 06:04:29 2020 user.notice OMR-VPS: Can't get token, try later
 something needs to be changed in this page!!
[image]
[image]

[image]

Ysurac, 2020-03-09:
You have an issue "Network interface duplicated" how did you configure the interfaces ? You use the wizard ?
Also you seems to use v2ray plugin, so obfuscation is enabled. You live in China ?

Archana7700, 2020-03-09:
actually i m using Ethernet which is used for lan as well as wan2 so it is showing as network duplicated ! 
[image]

i didn't  change the configuration at vps Do i need to change that? i live in India

Ysurac, 2020-03-09:
Using same interface, without using macvlan or vlan, will not work well.
Configuration to the VPS is modified by OpenMPTCProuter via the API, but in the log I can see that the API can't be contacted.
I don't think you need obfuscation in India, I think you don't have something like the Chinese "Great Firewall" ?

Archana7700, 2020-03-09:
yes nothing like that

Archana7700, 2020-03-09:
Do i need to configure anything for shadowsocks at the client side in local instance page or in Redir Rules

Ysurac, 2020-03-09:
No, you only need to use OpenMPTCProuter wizard.

Archana7700, 2020-03-09:
in wizard what needs to be configure i have done this much what server key i need to give ??

[image]

Archana7700, 2020-03-09:
i m finding new issue today its showing OMR-VPS :cant get the token ,try later i m unable to figure it out

Archana7700, 2020-03-09:
this is the router setting:
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301,65011'
dscp.@classify[3].comment='OMR vpn'
dscp.@classify[4]=classify
dscp.@classify[4].direction='both'
dscp.@classify[4].proto='udp'
dscp.@classify[4].class='cs6'
dscp.@classify[4].dest_port='65001'
[... 1291 lines cut ...]
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth0' 'wlan0'
vnstat.@vnstat[0].backup='0'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.wifinet0=wifi-iface
wireless.wifinet0.ssid='realme 2 Pro'
wireless.wifinet0.device='radio0'
wireless.wifinet0.mode='sta'
wireless.wifinet0.key='12345678'
wireless.wifinet0.encryption='psk2'
wireless.wifinet0.network='wan1'

Ysurac, 2020-03-09:
Username must be "openmptcprouter" and key is what the install script on the server give you at the end. The key is also available in ```/root/openmptcprouter_config.txt```.

Archana7700, 2020-03-09:
ok thank you its working!

hey now i m unable to see the network overview once it displayed after that now its not displaying

Archana7700, 2020-03-10:
After all this  how aggregation will happen ?when i m trying to download file where will i get to know that aggregation will 
[image]
be happening

Archana7700, 2020-03-10:
when i m seeing in vnstat i couldn't fine the difference which interface is using  more mb which is faster how aggregation is happening

closed 2020-03-11 by Archana7700