Ysurac/openmptcprouter #877: opkg error - 404 Not Found

SriramScorp, 2020-03-10:
Unable to install kernel modules for OMRouter version v0.53.3 (32-bit) and kernel version 4.19.84 (last stable release dated 2020/01/05). Link in /etc/opkg/distfeeds.conf `http://openmptcprouter.com:80/release/4.19/targets/brcm2708/bcm2709/packages` is down. Also, it would be better if you could maintain a repository for kernel modules and other related packages for older releases as well.

Ysurac, 2020-03-10:
I fixed the links.
I can't maintain kernel modules and packages for old releases, because this use lot of disk space. This will then cost money that I don't have.

SriramScorp, 2020-03-10:
Fine. But I'm using the last stable release and it runs on kernel version 4.19.84 whereas the kernel modules in the targets URL of /etc/opkg/distfeeds.conf currently requires kernel 4.19.106.

Ysurac, 2020-03-10:
True, release 0.54 should be out today; I'm making last tests before release.

SriramScorp, 2020-03-10:
Good to know. I was just testing the release candidate and found that MLVPN, which was dropped in v0.53.3, was available once again. Is it being re-introduced? Were you able to make it work without crashing?

Ysurac, 2020-03-10:
Yes, it seems to work without crashing. I think this was related to a dependency.

Ysurac, 2020-03-10:
v0.54 is now released.

closed 2020-03-10 by Ysurac