Ysurac/openmptcprouter #878: VPN tunnel down

Slivikkus, 2020-03-10:
## VPN TUNNEL DOWN

J'ai le "VPN tunnel down", suite à la mise à jour du VPS, le problème viens du script ?
[image]

## Specifications

  - OpenMPTCProuter version: (0.54
  - OpenMPTCProuter VPS version: (0.1013)
  - OpenMPTCProuter platform: (x86_64 ESXI)

Ysurac, 2020-03-10:
Quel est le VPN utilisé ?

Slivikkus, 2020-03-10:
"A Dead Simple VPN"

Slivikkus, 2020-03-10:
Il faut pas que je mette "Glorytun TCP" plutôt ?

Slivikkus, 2020-03-10:
Ah oui c'est bon x)

closed 2020-03-10 by Slivikkus

Ysurac, 2020-03-10:
Il y avait un bug avec DSVPN, j'ai corrigé. En relançant le script sur le VPS ce VPN refonctionnera.

Slivikkus, 2020-03-10:
D'accord, et pour l'IPv6 y'a plus de problème ? (d'ailleurs ça ce configure automatiquement maintenant ?)

Ysurac, 2020-03-10:
Normalement plus de soucis, il faut cocher la case dans le wizard et le tunnel fonctionne.
Il y a encore des améliorations possibles quand un /48 est dispo sur le VPS.

Slivikkus, 2020-03-10:
La configuration ce fait toujours sur le VPS ou le routeur le fait ?

Ysurac, 2020-03-10:
Le VPS doit avoir une IPv6 configurée correctement, sinon tout se fait sur le routeur.

Slivikkus, 2020-03-10:
Et encore une dernière question après j'en ai plus, j'ai le routeur sur une tour maintenant et j'aimerais savoir comment que ça ce passe pour configurer le wifi (via une clé wifi en USB)

Ysurac, 2020-03-10:
En la branchant et en allant dans Réseau->Wifi si le driver est déjà présent. Sinon il faut trouver le driver et l'installer.

Slivikkus, 2020-03-10:
Je vois pas la section wifi mais la clé est bien détecté par le routeur avec le nom suivant "wlan0"