Ysurac/openmptcprouter #879: VPN DOWN DEPUIS VPS 0.1013

bosseb, 2020-03-10:
- OpenMPTCProuter version: 0.54 officielle
  - OpenMPTCProuter VPS version: 0.1013 chez milkywan
  - OpenMPTCProuter platform: x86_64

depuis que j'ai mis à jour le VPS en 0.1013 j'ai un VPN DOWN, j'ai fais le test avec un autre VPS resté en 0.1012 tout refonction
[image]

Ysurac, 2020-03-10:
Quel est le VPN utilisé ?
Le VPS doit être redémarré après la mise à jour et ça semble pas être le cas la.

bosseb, 2020-03-10:
GLORYTUN TCP

Ysurac, 2020-03-10:
Pas de changements après redémarrage du VPS ?

bosseb, 2020-03-10:
j'utilise MILKYWAN et comme le vps n'est pas en mode root de base je ne pouvais pas rebooter

je les ai contactés et maintenant cela marche parfaitement

merci de ton aide

bonne soirée

closed 2020-03-10 by Ysurac