Ysurac/openmptcprouter #880: Question : Comment ajouter le WIFI ?

Slivikkus, 2020-03-10:
## Actual Behavior

J'ai une clé wifi sur mon routeur, la clé est bien détectée par openMPTCP sous le nom de WLAN0 mais pourquoi dans ce cas le menu "Wifi" ne s'affiche pas ?
[image]

## Specifications

  - OpenMPTCProuter version: (0.54)
  - OpenMPTCProuter VPS version: (0.1013)
  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2020-03-10:
Je n'ai jamais testé d'ajouter une clef wifi (je n'en possède pas), il faut peut être ajouté une configuration initiale : https://openwrt.org/docs/guide-user/network/wifi/basic
Si c'est le cas j'essayerais d'ajouter ça automatiquement lorsque qu'une clef wifi est branchée.

Slivikkus, 2020-03-10:
Comment je modifie le fichier /etc/config/wireless ?

Ysurac, 2020-03-10:
Via ssh, en utilisant ```vi```, ```mcedit``` ou en installant ```nano```.

Slivikkus, 2020-03-10:
> root@OpenMPTCProuter:~# wifi up wlan0
> wlan0(): Interface type not supported
> Command failed: Not found
> Command failed: Not found
> wlan0(): Interface type not supported
>

Ysurac, 2020-03-10:
check ```wifi status```

Slivikkus, 2020-03-10:
> root@OpenMPTCProuter:~# wifi status
> {
> 
> }
> root@OpenMPTCProuter:~#

Ysurac, 2020-03-10:
et ```iwinfo``` donne quoi ?

Slivikkus, 2020-03-10:
[image]

Ysurac, 2020-03-10:
et ```iw dev``` ?

Slivikkus, 2020-03-10:
[image]

Ysurac, 2020-03-10:
```dmesg``` affiche bien qu'il détecte quelque chose ?

Slivikkus, 2020-03-10:
Oui

Ysurac, 2020-03-10:
Et il donne quoi sur le sujet ?

Slivikkus, 2020-03-10:
> [ 5097.915160] usb 2-8: new full-speed USB device number 2 using ohci-pci
> [ 5098.231234] hub 2-8:1.0: USB hub found
> [ 5098.234126] hub 2-8:1.0: 3 ports detected
> [ 5098.585165] usb 2-8.1: new low-speed USB device number 3 using ohci-pci
> [ 5098.771908] input: Dell Dell Multimedia Pro Keyboard as /devices/pci0000:00/0000:00:02.0/usb2/2-8/2-8.1/2-8.1:1.0/0003:413C:2011.0001/input/input6
> [ 5098.835410] hid-generic 0003:413C:2011.0001: input,hidraw0: USB HID v1.10 Keyboard [Dell Dell Multimedia Pro Keyboard] on usb-0000:00:02.0-8.1/input0
> [ 5098.848865] input: Dell Dell Multimedia Pro Keyboard Consumer Control as /devices/pci0000:00/0000:00:02.0/usb2/2-8/2-8.1/2-8.1:1.1/0003:413C:2011.0002/input/input7
> [ 5098.915400] input: Dell Dell Multimedia Pro Keyboard System Control as /devices/pci0000:00/0000:00:02.0/usb2/2-8/2-8.1/2-8.1:1.1/0003:413C:2011.0002/input/input8
> [ 5098.916172] hid-generic 0003:413C:2011.0002: input,hidraw1: USB HID v1.10 Device [Dell Dell Multimedia Pro Keyboard] on usb-0000:00:02.0-8.1/input1
> [ 5099.235158] usb 2-8.2: new full-speed USB device number 4 using ohci-pci
> [ 5099.431855] input: Compx 2.4G Receiver as /devices/pci0000:00/0000:00:02.0/usb2/2-8/2-8.2/2-8.2:1.0/0003:25A7:FA23.0003/input/input9
> [ 5099.505552] hid-generic 0003:25A7:FA23.0003: input,hidraw2: USB HID v1.10 Keyboard [Compx 2.4G Receiver] on usb-0000:00:02.0-8.2/input0
> [ 5099.514341] input: Compx 2.4G Receiver Mouse as /devices/pci0000:00/0000:00:02.0/usb2/2-8/2-8.2/2-8.2:1.1/0003:25A7:FA23.0004/input/input10
> [ 5099.514896] input: Compx 2.4G Receiver as /devices/pci0000:00/0000:00:02.0/usb2/2-8/2-8.2/2-8.2:1.1/0003:25A7:FA23.0004/input/input11
> [ 5099.516186] input: Compx 2.4G Receiver Consumer Control as /devices/pci0000:00/0000:00:02.0/usb2/2-8/2-8.2/2-8.2:1.1/0003:25A7:FA23.0004/input/input12
> [ 5099.585406] input: Compx 2.4G Receiver System Control as /devices/pci0000:00/0000:00:02.0/usb2/2-8/2-8.2/2-8.2:1.1/0003:25A7:FA23.0004/input/input13
> [ 5099.585930] hid-generic 0003:25A7:FA23.0004: input,hiddev96,hidraw3: USB HID v1.10 Mouse [Compx 2.4G Receiver] on usb-0000:00:02.0-8.2/input1
> [ 6151.676446] usb 2-8: USB disconnect, device number 2
> [ 6151.676598] usb 2-8.1: USB disconnect, device number 3
> [ 6151.965795] usb 2-8.2: USB disconnect, device number 4
> [ 9426.589230] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
> [ 9427.355603] r8712u 1-6:1.0 wlan0: 1 RCR=0x153f00e
> [ 9427.356489] r8712u 1-6:1.0 wlan0: 2 RCR=0x553f00e
> [ 9429.575461] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
> [ 9432.135447] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
> [16607.988241] IPv6: ADDRCONF(NETDEV_UP): sit0: link is not ready
> [16607.988378] IPv6: ADDRCONF(NETDEV_UP): ifb0: link is not ready
> [16607.988505] IPv6: ADDRCONF(NETDEV_UP): ifb1: link is not ready
> [16607.988631] IPv6: ADDRCONF(NETDEV_UP): teql0: link is not ready
> [16607.988761] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
> [16607.988893] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
> [16607.989102] IPv6: ADDRCONF(NETDEV_UP): eth5: link is not ready
> [16607.989232] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
> [16608.396059] bnx2 0000:05:00.1 eth4: using MSIX
> [16608.396258] IPv6: ADDRCONF(NETDEV_UP): eth4: link is not ready
> [16610.917815] bnx2 0000:05:00.1 eth4: NIC Copper Link is Up, 1000 Mbps full duplex
> [16610.918057] , receive & transmit flow control ON
> [16610.918268] IPv6: ADDRCONF(NETDEV_CHANGE): eth4: link becomes ready
> [16623.321513] nginx invoked oom-killer: gfp_mask=0x6000c0(GFP_KERNEL), nodemask=(null), order=0, oom_score_adj=0
> [16623.321775] nginx cpuset=/ mems_allowed=0
> [16623.321908] CPU: 0 PID: 27172 Comm: nginx Tainted: G         C        4.19.106 #0
> [16623.322158] Hardware name: Acer ASE380/AST180/APM8  /EM61SM/EM61PM         , BIOS R01-C0 06/15/2007
> [16623.322390] Call Trace:
> [16623.322525]  dump_stack+0x6d/0x9a
> [16623.322657]  dump_header+0x65/0x211
> [16623.322786]  ? __delayacct_freepages_end+0x25/0x30
> [16623.322916]  oom_kill_process.cold.12+0xb/0x1e6
> [16623.323046]  ? oom_badness.part.6+0xd/0x130
> [16623.323176]  out_of_memory+0x162/0x250
> [16623.323304]  __alloc_pages_nodemask+0x9f3/0xd80
> [16623.323432]  ? __kmalloc_track_caller+0x40/0x170
> [16623.323560]  skb_page_frag_refill+0x9f/0xc0
> [16623.323687]  sk_page_frag_refill+0x1a/0x80
> [16623.323819]  tcp_sendmsg_locked+0x421/0xd60
> [16623.323949]  tcp_sendmsg+0x27/0x40
> [16623.324079]  inet_sendmsg+0x43/0xb0
> [16623.324210]  sock_sendmsg+0x15/0x20
> [16623.324336]  sock_write_iter+0x84/0xd0
> [16623.324461]  do_iter_readv_writev+0x10a/0x170
> [16623.324587]  do_iter_write+0x70/0x180
> [16623.324714]  vfs_writev+0x6c/0xc0
> [16623.324840]  do_writev+0x4c/0xb0
> [16623.324965]  __x64_sys_writev+0x17/0x20
> [16623.325092]  do_syscall_64+0x57/0x260
> [16623.325153]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
> [16623.325153] RIP: 0033:0x7fa4428032c1
> [16623.325153] Code: Bad RIP value.
> [16623.325153] RSP: 002b:00007ffc74495108 EFLAGS: 00000246 ORIG_RAX: 0000000000000014
> [16623.325153] RAX: ffffffffffffffda RBX: 00007fa442821d48 RCX: 00007fa4428032c1
> [16623.325153] RDX: 0000000000000001 RSI: 00007ffc744951e0 RDI: 0000000000000559
[... 249 lines cut ...]
> [16678.548710] [  12368]     0 12368      282        8    40960        0             0 grep
> [16678.548947] [  12371]     0 12371      278       11    32768        0             0 sh
[... cut ...]
> [16678.572081] TCP: too many orphaned sockets
> [16678.572401] MPTCP: out of memory: force closing socket
> [16678.572650] TCP: too many orphaned sockets
> [16678.572855] TCP: too many orphaned sockets
> [16678.573015] MPTCP: out of memory: force closing socket
> [16678.573231] TCP: too many orphaned sockets
> [16678.573423] TCP: too many orphaned sockets
> [16678.573573] MPTCP: out of memory: force closing socket
> [16678.573785] TCP: too many orphaned sockets
> [16678.573964] TCP: too many orphaned sockets

Ysurac, 2020-03-10:
Y'a bien le wlan0, par contre beaucoup d'erreurs de type "out of memory", il y a combien de mémoire ?

Slivikkus, 2020-03-10:
1Go de mémoire

Ysurac, 2020-03-10:
Quelque chose consomme trop de mémoire, plusieurs VPS sont définis ?
Ou alors beaucoup de connexions ?

Slivikkus, 2020-03-10:
2 connexion, 2 VPS pas plus pas moins (du moins pour l'instant)

Ysurac, 2020-03-10:
ok, c'est certainement le failover entre les 2 VPS, qui utilise NGinx, et ça consomme plus que 1Go de mémoire.

Slivikkus, 2020-03-10:
Ok bah j'ai de la RAM en stock j'en ajouterais, c'est pas un problème

Slivikkus, 2020-03-10:
Je vais faire sans wifi pour l'instant du coup ...

J'ai encore un petit problème , j'ai pas d'accès IPv6 alors que tout est config 
[image]

Ysurac, 2020-03-10:
Le "Direct output" ne me semble pas normal non plus.
Sinon pour IPv6 je n'ai pas testé si on a plusieurs VPS, je ne sais d'ailleurs pas trop comment je pourrais faire que ça fonctionne de façon transparente dans ce cas.

Slivikkus, 2020-03-10:
ça me le fait depuis la version "beta" de la 0.54, le "No IPv6 access" ça le faisais déjà avant que j'ai les deux VPS

closed 2020-03-12 by Slivikkus

tboome33, 2020-04-16:
Bonjour, je serai très intéressé par la reconnaissance automatique des clé WIFI. J'aimerai pouvoir agréger plusieurs partages de téléphone et donc ajouter 4 clés wifi. Je suis allé regarder https://openwrt.org/docs/guide-user/network/wifi/basic  mais j'ai pas tardé à fermer la page. Trop compliqué ! Sinon comment faire un point wifi depuis le Rasberry Pi  ? Par exemple agréger 4 partages de téléphone et avec le wifi intégré au Rasberry Pi faire un point d'accès wifi ?  Merci pour tout