Ysurac/openmptcprouter #881: aggregation issue

Archana7700, 2020-03-11:
After configurations how do i check aggregations is happening not where which i check when i transferring file or downloading i not able to see where aggregation is happening vnstat is not that proper data like while transferring how much data is consumed

Ysurac, 2020-03-11:
In network->MPTCP and bandwidth tab.

Archana7700, 2020-03-11:
when i tried downloading a file from my server i could find that the time duration taken by both of the interface was 5.26mins
but when i tried downloading by using single interface was more than what was it was giving me with both the interface its not aggregating
with wan1 alone it took 5.14mins
with wan2 alone it took 4.55mins 
but with both up it took 5.26mins

Archana7700, 2020-03-11:
and even i had a doubt that by using glorytun -tcp why it is blocking google.com i m not able to access google how do i download file from google n check aggregation is happening.

Ysurac, 2020-03-11:
VPN are by default used for anything that is not TCP, shadowsocks is used for TCP only.

On the router there is ```omr-test-speed``` that download a file from OVH, if you use ```omr-test-speed wan1``` this will download it via wan1, without interface this use the aggregation. This should help to check if aggregation is working. OpenMPTCProuter status should be all green also.

How are you blocked from google ? what is the message if any ?

Archana7700, 2020-03-11:
Thanks for your support!!
firstly i have tried using omr-test-speed what i found is when i turn off the glorytun i m able to see the speed of upload
secondly what i found is individual interfaces are giving me the same speed as compared to when both the interface are up i m not able the find the aggregations or their is not enough contributions when both the interfaces are up 

root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed       Time                  Time                  Time  Current
                                 Dload  Upload   Total   Spent       Left                  Speed
  0 10.0G    0 11.1M    0     0   602k      0  4:50:12     0:00:18              4:49:54                       735k^C

root@OpenMPTCProuter:~# omr-test-speed usb0
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 11.3M    0     0   573k      0  5:04:39  0:00:20  5:04:19  903k^C

Ysurac, 2020-03-11:
Glorytun is not related to omr-test-speed, only shadowsocks is used here.
Test should be for more than 20s, and on each interfaces.

You can also check in Network->MPTCP and MPTCP fullmesh tab that you have ip of every available connections.

All is really green in OpenMPTCProuter status ? With all connections available ?

Archana7700, 2020-03-11:
yes all connections are green but when i connect Glorytun i get an error as 
curl: (6) Couldn't resolve host 'proof.ovh.net'

Ysurac, 2020-03-11:
What is the release of OpenMPTCProuter and VPS script used ?

Archana7700, 2020-03-11:
openMPTCProuter version:
openmptcprouter v0.54rc9 r0+12300-fc65fae304 / LuCI Master git-20.054.72983-5624256 

vps script used: 
wget http://www.openmptcprouter.com/server/ubuntu18.04-x86_64.sh
sudo sh ubuntu18.04-x86_64.sh

Ysurac, 2020-03-11:
You should update to the released version.
And for the VPS I speak about the version too ;)

Archana7700, 2020-03-11:
< OpenMPTCProuter VPS 0.1008 >

Ysurac, 2020-03-11:
This need to be updated.

Archana7700, 2020-03-12:
how to update vps

Ysurac, 2020-03-12:
Run the script again:
```wget -O - http://www.openmptcprouter.com/server/debian10-x86_64.sh | sh```

Archana7700, 2020-03-12:
When i m trying to update the  VPS 0.1013 i m not able to do it , i m getting the following error Please tell me how can i fix this issue 
As u can see that IP [IP: 143.95.32.90 80] i m able to ping it !

erpadmin@mantis:~$ sudo sh ubuntu18.04-x86_64.sh
Get:1 https://packages.microsoft.com/repos/vscode stable InRelease [3959 B]
Hit:2 http://in.archive.ubuntu.com/ubuntu bionic InRelease                                                                                    
Hit:3 http://in.archive.ubuntu.com/ubuntu bionic-updates InRelease                                                                            
Get:4 https://packages.microsoft.com/repos/vscode stable/main amd64 Packages [166 kB]                                                         
Hit:5 http://in.archive.ubuntu.com/ubuntu bionic-backports InRelease                                                                          
Hit:7 https://repo.openmptcprouter.com stretch InRelease                                                                                      
Hit:8 https://download.sublimetext.com apt/stable/ InRelease                                                                                  
Get:9 http://archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB]                                         
Err:6 http://www.getdeb.net/ubuntu yakkety-getdeb InRelease                                               
  403  Forbidden [IP: 143.95.32.90 80]
Hit:10 http://archive.ubuntu.com/ubuntu bionic InRelease                
Get:11 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]
Reading package lists... Done                                 
N: Skipping acquire of configured file 'main/binary-i386/Packages' as repository 'https://repo.openmptcprouter.com stretch InRelease' doesn't support architecture 'i386'
E: Failed to fetch http://archive.getdeb.net/ubuntu/dists/yakkety-getdeb/InRelease  403  Forbidden [IP: 143.95.32.90 80]
E: The repository 'http://archive.getdeb.net/ubuntu yakkety-getdeb InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

  - OpenMPTCProuter version: (v0.54)
  - OpenMPTCProuter VPS version: (1008)
  - OpenMPTCProuter platform: (RPI4)

Archana7700, 2020-03-12:
i m running this script :
wget -O - http://www.openmptcprouter.com/server/ubuntu18.04-x86_64.sh | sh

Ysurac, 2020-03-12:
This doesn't seems to be related to OpenMPTCProuter, the repository "http://archive.getdeb.net/ubuntu/dists/yakkety-getdeb/InRelease" doesn't answer. You should remove it from /etc/apt/

Archana7700, 2020-03-12:
thank you it got updated !

Archana7700, 2020-03-12:
Hi..
Thanks a lot for your support!! , both of the interface is working but individual interface are giving more faster response  than combined effect .
root@OpenMPTCProuter:~# omr-test-speed 
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 14.2M    0     0   537k      0  5:25:04  0:00:27  5:24:37  939k^C
root@OpenMPTCProuter:~# omr-test-speed wlan0
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 14.3M    0     0  1045k      0  2:47:06  0:00:14  2:46:52  939k^C
root@OpenMPTCProuter:~# omr-test-speed usb0
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 14.0M    0     0   791k      0  3:40:49  0:00:18  3:40:31  786k^C

Archana7700, 2020-03-12:
Thanks for your support !!
Both the interface are working fine,but individual interfaces are giving me faster response than the combined effect 

```
> root@OpenMPTCProuter:~# omr-test-speed
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
>   0 10.0G    0 14.2M    0     0   537k      0  5:25:04  0:00:27  5:24:37  939k
> root@OpenMPTCProuter:~# omr-test-speed wlan0
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
>   0 10.0G    0 14.3M    0     0  1045k      0  2:47:06  0:00:14  2:46:52  939k
> root@OpenMPTCProuter:~# omr-test-speed usb0
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
>   0 10.0G    0 14.0M    0     0   791k      0  3:40:49  0:00:18  3:40:31  786k 
```

Ysurac, 2020-03-12:
I would need the config, without keys, from System->OpenMPTCProuter and all parameters tab.

Archana7700, 2020-03-12:
[image]
[image]
[image]

Ysurac, 2020-03-12:
I speak about the "show all settings" tab, this give all config of OpenMPTCProuter.

Archana7700, 2020-03-12:
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].rebind_protection='0'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301,65011'
dscp.@classify[3].comment='OMR vpn'
dscp.@classify[4]=classify
dscp.@classify[4].direction='both'
dscp.@classify[4].proto='udp'
dscp.@classify[4].class='cs6'
dscp.@classify[4].dest_port='65001'
dscp.@classify[4].comment='OMR vpn'
[... 1315 lines cut ...]
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan2' 'wlan0' 'usb0'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.wifinet0=wifi-iface
wireless.wifinet0.ssid='ROGFI'
wireless.wifinet0.device='radio0'
wireless.wifinet0.mode='sta'
wireless.wifinet0.key='pass1234'
wireless.wifinet0.encryption='psk2'
wireless.wifinet0.network='wan1'

Ysurac, 2020-03-12:
wan1 seems to be see as down. Both interface are defined as macvlan of eth0 but defined with another interface, I don't know how this can work. You should remove macvlan in Network->macvlan.
You should also update OpenMPTCProuter to latest release.

Archana7700, 2020-03-12:
yaa wan1 was not connected,but i didnt change anything in network macvlan
[image]

Archana7700, 2020-03-12:
latest release i have downloaded from
here: https://www.openmptcprouter.com/release/4.19/
Latest version: v0.54rc15
As u have mentioned in the issue #864

Ysurac, 2020-03-12:
Latest release is available here: https://www.openmptcprouter.com/download
v0.54 was released now.
For macvlan you should delete it because it's not used in your case.

Archana7700, 2020-03-13:
At server side when i m seeing the status of shadowsocks i could find that its inactive 

/etc/init.d/shadowsocks-libev status
● shadowsocks-libev.service - Shadowsocks-libev Default Server Service
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2020-03-13 15:00:32 IST; 4s ago
     Docs: man:shadowsocks-libev(8)
  Process: 11856 ExecStart=/usr/bin/ss-server -c $CONFFILE $DAEMON_ARGS (code=exited, status=255)
 Main PID: 11856 (code=exited, status=255)

Mar 13 15:00:32 mantis systemd[1]: Started Shadowsocks-libev Default Server Service.
Mar 13 15:00:32 mantis ss-server[11856]:  2020-03-13 15:00:32 INFO: UDP relay enabled
Mar 13 15:00:32 mantis ss-server[11856]:  2020-03-13 15:00:32 INFO: initializing ciphers... chacha20-ietf-poly1305
Mar 13 15:00:32 mantis ss-server[11856]:  2020-03-13 15:00:32 INFO: tcp server listening at 192.168.16.212:65101
Mar 13 15:00:32 mantis ss-server[11856]:  2020-03-13 15:00:32 ERROR: bind: Cannot assign requested address
Mar 13 15:00:32 mantis ss-server[11856]:  2020-03-13 15:00:32 ERROR: failed to bind address
Mar 13 15:00:32 mantis systemd[1]: shadowsocks-libev.service: Main process exited, code=exited, status=255/n/a
Mar 13 15:00:32 mantis systemd[1]: shadowsocks-libev.service: Failed with``

Ysurac, 2020-03-13:
Shadowsocks-libev must not be used. The service used is ```shadowsocks-libev-manager@manager```

gnosoz, 2020-03-21:
I am having similar issues sometimes but today is working everything like a charm.
Having said that, the results are pretty peculiar especially on omr-test-speed side...

**A) from router i see as follows**
```
Connecting to host bouygues.iperf.fr, port 5202
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   128 MBytes   107 Mbits/sec    2             sender
[  5]   0.00-10.09  sec   100 MBytes  83.2 Mbits/sec                  receiver
```

**B) from vps i see as follows**
```
iperf3 -c bouygues.iperf.fr -p 5202
Connecting to host bouygues.iperf.fr, port 5202
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec   124 MBytes   104 Mbits/sec  554             sender
[  5]   0.00-10.05  sec   116 MBytes  96.5 Mbits/sec                  receiver
```
When i run a speed test through omr i get completely out of place results
```
**~# omr-test-speed eth1**
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  100M  100  100M    0     0  11.3M      0  0:00:08  0:00:08 --:--:-- 12.3M
**~# omr-test-speed eth2**
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 59  100M   59 59.8M    0     0  1001k      0  0:01:42  0:01:01  0:00:41 1162k
**~# omr-test-speed eth3**
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 32  100M   32 32.7M    0     0  1269k      0  0:01:20  0:00:26  0:00:54 1360k
**~# omr-test-speed** 
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  100M  100  100M    0     0  8515k      0  0:00:12  0:00:12 --:--:-- 10.3M
```

[speed-test OVH](https://pic.nperf.com/r/3245626230744193-IP0jmzhe.png)

Any thoughts on why curl is not giving real results and how to better the omr-test-speed procedure?

Ysurac, 2020-03-21:
Don't forget that curl give result in byte per second, so 10.3*8=80.4Mbps. Not so bad. And test on eth1 and aggregated are too short to say anything more.

Archana7700, 2020-03-24:
Hi,
i m still facing the same issue when i m disconnecting vpn only than i can check the omr test speed i dont know why it is happening so how to resolve it
curl: (6) Couldn't resolve host 'proof.ovh.net'

Ysurac, 2020-03-24:
You have latest OMR release and latest VPS script ? Macvlan interfaces are deleted in Network->Macvlan ?

Archana7700, 2020-03-24:
yes i have updated today, VPS i m not able to update i m getting error as unable to establish ssl connection i have even tried by disabling the vpn but still same thing

Archana7700, 2020-03-24:
also i have deleted macvlan i could not find anything like before but i can see these two lines in setting 
ucitrack.@macvlan[0]=macvlan
ucitrack.@macvlan[0].init='macvlan'

Ysurac, 2020-03-24:
The VPS must be also updated.
Can you paste me the error ? Check also date on the VPS.

Archana7700, 2020-03-24:
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 269 not upgraded.
URL transformed to HTTPS due to an HSTS policy
--2020-03-24 11:52:36--  https://github.com/shadowsocks/shadowsocks-libev/releases/download/v3.3.3/shadowsocks-libev-3.3.3.tar.gz
Resolving github.com (github.com)... 13.234.210.38
Connecting to github.com (github.com)|13.234.210.38|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/7177269/a5de2500-fbf1-11e9-943c-bbcf9194ceac?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200324%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200324T062236Z&X-Amz-Expires=300&X-Amz-Signature=9c212f2b74b300ae64a99b87dc6e2ed7f7d31b8a43546041cb80c57d4f6399e9&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dshadowsocks-libev-3.3.3.tar.gz&response-content-type=application%2Foctet-stream [following]
--2020-03-24 11:52:37--  https://github-production-release-asset-2e65be.s3.amazonaws.com/7177269/a5de2500-fbf1-11e9-943c-bbcf9194ceac?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200324%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200324T062236Z&X-Amz-Expires=300&X-Amz-Signature=9c212f2b74b300ae64a99b87dc6e2ed7f7d31b8a43546041cb80c57d4f6399e9&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dshadowsocks-libev-3.3.3.tar.gz&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)... 52.216.204.155
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.204.155|:443... connected.
Unable to establish SSL connection.

Archana7700, 2020-03-24:
I can show u the graphs how much bandwidth i m able to get 
First one is of wan1
second is of wan2 
wan2 is very negligible and i also have a doubt that my wan2 is nothing but usb0 that is i have given connection from my phone via usb tethering is that an issue ?

  
[image]

Ysurac, 2020-03-24:
Yuo can try, on the VPS, to run ```update-ca-certificates -f``` (this updates certificates) and run the script again.
For wan2, there is no issues with usb tethering but ISP can limit speed or some other 4G speed limits.

Archana7700, 2020-03-24:
still i m getting same error after updating also

Ysurac, 2020-03-24:
What is the VPS provider ?

Archana7700, 2020-03-24:
sorry i didnt get it

Archana7700, 2020-03-27:
Hi 
this is the status of my aggregation and  only shadowsocks is running no vps is running will it effect my aggregation? 

[image]

Ysurac, 2020-03-27:
No VPS is running ? Then aggregation can't work.
Also an interface need to be set as master in Network->MPTCP
On the version reported on status page, I see an old release and not booted on MPTCP kernel, this can't work.

Archana7700, 2020-03-27:
first i use to use vps as glorytun tcp but i cant check omr test speed with vps running

Archana7700, 2020-03-27:
now glortun tcp is running but vps i m not able to install the new version ;(

Ysurac, 2020-03-27:
MPTCP kernel is not running on VPS, this can't work at all.
Check date time on your VPS, if you VPS still can't access some https site (like the previous github link) then your VPS have a big problem not related to OpenMPTCProuter.

Archana7700, 2020-03-27:
sorry i mean vpn was not running at that time not vps  sorry by mistake i wrote vps instead of vpn

Archana7700, 2020-03-27:
i shall reset up my server n try again

Ysurac, 2020-03-27:
Yes

closed 2020-05-18 by Ysurac