Ysurac/openmptcprouter #882: passage sur GLORYTUN UDP : VPN Tunnel DOWN

bosseb, 2020-03-11:
- OpenMPTCProuter version: 0.54 Officiel
  - OpenMPTCProuter VPS version: 0.1013 (vps MILKYWAN)
  - OpenMPTCProuter platform: x86_64

quand je passe sur glorytun udp, je passe en VPN tunnel DOWN, et l'accès distant de mon plex ne fonctionne plus ...
j'ai redémarré vps et OMR et pas mieux !

[image]

Ysurac, 2020-03-11:
Il faut voir dans Etat->Journal Système ce qu'il y a.
Et sur le VPS, un ```ps aux | grep glory``` permettra de savoir si glorytun-udp est bien lancé.

bosseb, 2020-03-11:
ci dessous le journal système

[journal syst.txt](https://github.com/Ysurac/openmptcprouter/files/4320715/journal.syst.txt)

bosseb, 2020-03-11:
sebastien@sebastieng:~$ ps aux | grep glory
root       429  0.0  0.0   2692   904 ?        SLs  21:03   0:00 glorytun bind 0                .0.0.0 65001 keyfile /etc/glorytun-udp/tun0.key dev gt-udp-tun0 chacha persist
root       437  0.0  0.0   2960   928 ?        Ss   21:03   0:00 glorytun-tcp li                stener keyfile /etc/glorytun-tcp/tun0.key dev gt-tun0 port 65001 mptcp chacha20                 retry count -1 const 5000000 timeout 10000 keepalive count 5 idle 20 interval 2                 buffer-size 32768 multiqueue
sebasti+ 22144  0.0  0.0   6144   892 pts/0    S+   21:24   0:00 grep glory
sebastien@sebastieng:~$

flimo44, 2020-03-11:
Bonjour , 
j'ai le problème inverse depuis la 0.54. J'ai dû passer en UDP car ça ne montait plus en TCP.

Ysurac, 2020-03-11:
Il me faudrait la config, sans les clefs (dans Système->OpenMPTCProuter, onglet "tous les paramètres")

Ysurac, 2020-03-11:
@flimo44 ouvre ta propre issue du coup (et vérifie que tu utilises bien la dernière version du script VPS)

bosseb, 2020-03-11:
[tous parametres.txt](https://github.com/Ysurac/openmptcprouter/files/4320847/tous.parametres.txt)

j'ai remplacé les keys par key
password par password et token par token

;)

Ysurac, 2020-03-11:
Je ne vois pas de soucis dans la config la.
Il faudrait voir le contenu de ```/etc/glorytun-udp/tun0``` (la clef doit correspondre, et il doit indiquer chacha20)
tu peux tester un ```ping 10.255.254.1``` depuis le routeur via SSH aussi ou un ```ping 10.255.254.2``` depuis le VPS.

bosseb, 2020-03-11:
pour tune0
BIND=0.0.0.0
BIND_PORT=65001
DEV=tun0
OPTIONS="chacha persist"

bosseb, 2020-03-11:
root@OpenMPTCProuter:~# ping 10.255.254.1
PING 10.255.254.1 (10.255.254.1) 56(84) bytes of data.

Ysurac, 2020-03-11:
Etrange, je ne vois pas où peut être le soucis...
Sur le routeur, que donne ```glorytun-udp path``` via SSH ?

bosseb, 2020-03-11:
path UP
  status:  DEGRADED
  bind:    192.168.6.5 port 5000
  public:  - port 0
  peer:    45.13.104.175 port 65001
  mtu:     0 bytes
  rtt:     0.000 ms
  rttvar:  0.000 ms
  rate:    fixed
  beat:    100 ms
  tx:
    rate:  125000000 bytes/sec
    loss:  0 percent
    total: 2365 packets
  rx:
    rate:  125000000 bytes/sec
    loss:  0 percent
    total: 0 packets
path UP
  status:  DEGRADED
  bind:    192.168.2.5 port 5000
  public:  - port 0
  peer:    45.13.104.175 port 65001
  mtu:     0 bytes
  rtt:     0.000 ms
  rttvar:  0.000 ms
  rate:    fixed
  beat:    100 ms
  tx:
    rate:  125000000 bytes/sec
    loss:  0 percent
    total: 2365 packets
  rx:
    rate:  125000000 bytes/sec
    loss:  0 percent
    total: 0 packets
path UP
  status:  DEGRADED
  bind:    192.168.4.5 port 5000
  public:  - port 0
  peer:    45.13.104.175 port 65001
  mtu:     0 bytes
  rtt:     0.000 ms
  rttvar:  0.000 ms
  rate:    fixed
  beat:    100 ms
  tx:
    rate:  125000000 bytes/sec
    loss:  0 percent
    total: 2365 packets
  rx:
    rate:  125000000 bytes/sec
    loss:  0 percent
    total: 0 packets

bosseb, 2020-03-11:
tu m'indiquais dans un message précédent : Il faudrait voir le contenu de /etc/glorytun-udp/tun0 (la clef doit correspondre, et il doit indiquer chacha20)
cependant dans ce fichier pas de clé et pas de chacha20 regarde : 
BIND=0.0.0.0
BIND_PORT=65001
DEV=tun0
OPTIONS="chacha persist"

Ysurac, 2020-03-12:
La clef est bien dans tun0.key.
La configuration est bonne.
Que donne un ```ip a show gt-udp-tun0``` ?

bosseb, 2020-03-12:
je fais ça où ? VPS ou OMR ?

Ysurac, 2020-03-12:
sur le VPS pardon

bosseb, 2020-03-12:
sebastien@sebastieng:~$ ip a show gt-udp-tun0
4: gt-udp-tun0: <POINTOPOINT,MULTICAST,NOARP,80000> mtu 1500 qdisc noop state DO                                                                                                                                   WN group default qlen 500
    link/none

Ysurac, 2020-03-12:
Ah bah voila le soucis.
Le fichier ```/etc/glorytun-udp/post.sh``` est bien présent ?
Tu peux me donenr le contenu du ```/lib/systemd/system/glorytun-udp@.service``` ?
Tu peux tenter un ```systemctl daemon-reload``` puis un ```systemctl restart glorytun-udp@gt-udp-tun0``` ?

bosseb, 2020-03-12:
1//etc/glorytun-udp/post.sh est bien présent ? oui il est présent

bosseb, 2020-03-12:
/lib/systemd/system/glorytun-udp@.service

[Unit]
Description=Glorytun UDP on %I
After=network.target network-online.target

[Service]
Type=simple
Restart=always
EnvironmentFile=/etc/glorytun-udp/%i
ExecStart=/usr/local/bin/glorytun-udp-run /etc/glorytun-udp/%i
ExecStartPost=-/etc/glorytun-udp/post.sh /etc/glorytun-udp/%i
CapabilityBoundingSet=CAP_NET_ADMIN

[Install]
WantedBy=multi-user.target

bosseb, 2020-03-12:
j'oublie un détail, depuis hier soir j'ai repassé ma config en GLORYTUN TCP ... ça foire les éléments que je t'envoie ?

Ysurac, 2020-03-12:
Non. Tout est bon jusque là.

bosseb, 2020-03-12:
sebastien@sebastieng:~$ systemctl daemon-reload
Failed to reload daemon: The name org.freedesktop.PolicyKit1 was not provided by                                                                                                                                    any .service files

Ysurac, 2020-03-12:
```sudo systemctl daemon-reload``` si tu es pas root

bosseb, 2020-03-12:
pas de message d'erreur sur les 2 dernières avec sudo

Ysurac, 2020-03-12:
et que donne ```ip a show gt-udp-tun0``` maintenant ?

bosseb, 2020-03-12:
4: gt-udp-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq state UP group default qlen 500
    link/none
    inet 10.255.254.1/30 brd 10.255.254.3 scope global gt-udp-tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::8adb:63fe:29b3:3400/64 scope link stable-privacy
       valid_lft forever preferred_lft forever

Ysurac, 2020-03-12:
C'est beaucoup mieux.

bosseb, 2020-03-12:
donc je dois faire quoi pour que cela marche en GLORYTUN UDP ?

Ysurac, 2020-03-12:
Tu peux essayer de repasser dessus

bosseb, 2020-03-12:
pas mieux il y a  tjrs marqué vnp tunnel down

Ysurac, 2020-03-12:
Je sèche la... Tu utilises bien le wizard pour changer de VPN ?

bosseb, 2020-03-12:
yes

bosseb, 2020-03-12:
oh attends ça viens de passer en OK (je suis connecté à distance)

bosseb, 2020-03-12:
donc tout est ok maintenant avec glorytun udp , donc je dois faire quoi en fait pour que cela marche ? j'ai une autre agrégation au bureau et j'ai le même soucis

Ysurac, 2020-03-12:
```sudo systemctl daemon-reload``` puis un ```sudo systemctl restart glorytun-udp@tun0```
J'ai ajouté le daemon-reload dans le script d'installation du VPS.

bosseb, 2020-03-12:
il faut relancer une MAJ du VPS ?

Ysurac, 2020-03-12:
Soit le daemon-reload, soit màj. Mais autant faire le daemon reload si elle est déjà à jour.

bosseb, 2020-03-12:
en tout cas merci beaucoup pour l'aide c'est top !!!

RH2WTF, 2020-03-12:
Bonjour,

pour info je rencontre exactement le même problème, la solution proposer ci dessus fonctionne : 

sudo systemctl daemon-reload puis un sudo systemctl restart glorytun-udp@tun0

cependant à chaque redémarrage du VPS, le problème survient de nouveau.

y a t'il une solution définitive pour résoudre cette problématique ?

Merci pour le support @Ysurac , t'es au top !!!

Ysurac, 2020-03-12:
C'est corrigé dans le script VPS v0.1014 (un script tourne en permanence pour vérifier l'état de glorytun-udp).

Eltarius, 2020-03-13:
> 
> 
> C'est corrigé dans le script VPS v0.1014 (un script tourne en permanence pour vérifier l'état de glorytun-udp).

Je rajoute une petite question sur cette issue que j'ai également depuis le passage à la dernière version. J'ai egalement eu besoin de relancer le daemon et de restart glorytun-udp après la mise à jour en VPS v0.1014 alors que d'après toi un script tourne en permanence pour vérifier.
Maintenant j'ai:
`mtu 1500 qdisc fq state UP group default qlen 500`

Cependant sur mon routeur j'ai toujours tun0 DOWN.
[image]

Si tu veux que j'ouvre une autres issue n'hésite pas.

Ysurac, 2020-03-13:
Il me faudrait le résultat de ```ip a show gt-udp-tun0``` et de ```glorytun-udp show dev gt-udp-tun0```

Eltarius, 2020-03-13:
`8: gt-udp-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq state UP group default qlen 500
    link/none
    inet 10.255.254.1/30 brd 10.255.254.3 scope global gt-udp-tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::28ad:f2f1:e4f5:1622/64 scope link stable-privacy
       valid_lft forever preferred_lft forever`

`-bash: glorytun-udp : commande introuvable`

Ysurac, 2020-03-13:
oui pardon, c'est ```glorytun show dev gt-udp-tun0``` sur le VPS

Eltarius, 2020-03-13:
`server gt-udp-tun0:
  pid:    20679
  bind:   0.0.0.0 port 65001
  mtu:    0
  cipher: chacha20poly1305`

Ysurac, 2020-03-14:
Visiblement glorytun-udp tourne correctement.

Eltarius, 2020-03-14:
Donc ceci est un bug d'affichage coté routeur:
[image]

xillibit, 2020-03-18:
Je viens de passer le router en 0.56 et le VPS en 0.1015 et j'ai aussi le VPN tunnel down

Sur le VPS quand je fais un `glorytun show dev gt-udp-tun0` ça m'affiche : 

[image]

Ysurac, 2020-03-18:
En 0.54.6 Glorytun UDP est mis à jour vers la dernière version. Est-ce que le débit de chaque interface est bien présent dans le wizard sur le routeur ?

xillibit, 2020-03-18:
C'est ça le wizard ? Pour LAN2 et WAN3 je voudrais les supprimer mais je n'y arrive pas

[image]

Ysurac, 2020-03-18:
Non, ça c'est l'affichage de l'état/status, le wizard c'est l'onglet juste avant.
Pour LAN2 et wan3 il suffit de désactiver le multipath sur ces interfaces dans Réseau->MPTCP

xillibit, 2020-03-18:
J'ai 0 : 

[image]

J'ai ça maintenant dans l'état en ayant voulu rajouter des étiquettes sur chaque interface, donc j'ai fait Sauvegarder et appliquer sur le wizard : 

[image]

Ysurac, 2020-03-19:
Essaye en définissant les valeurs.

xillibit, 2020-03-19:
J'ai mis les valeurs suivantes et en en redémarrant le routeur j'ai plus le VPN down, hier soir j'avais aussi redémarré l'instance du serveur : 

[image]

philoun8, 2020-07-30:
Bonjour a tous
Pour moi c'est des jours VPN tunnel DOWNet d'autres jours non!
le hic c'est que quand c'est tunnel down, les ports ne sont plus ouverts depuis l'exterieur!!!

Ysurac, 2020-07-30:
Ouvre ta propre issue, car il me faudra toute les infos du template (versions etc...).

closed 2020-12-03 by github-actions[bot]