Ysurac/openmptcprouter #885: PLEX ne fonctionne plus depuis mise a jour ! ver :  0.1013

skapogo, 2020-03-12:
## Specifications

  - OpenMPTCProuter version: 0.54.1 
  - OpenMPTCProuter VPS version:  0.1013
  - OpenMPTCProuter platform: RPI3 B+

Bonjour, 

Depuis la mise a jour impossible de faire fonctionner Plex.
Je me suis aperçu qu'il était impossible d'utiliser le port forwarding depuis la mise à jour.
J'ai essayé de downgrade OpenMPTCProuter version 0.53 mais le problème persiste.
Le problème vient du soft VPS et il est impossible de le downgrade.
Pouvez vous me fournir un script pour reinstaller le VPS sur la version précédente merci d'avance en attendant une correction du problème.

Merci encore pour votre travail !

[image]
[image]

Ysurac, 2020-03-12:
Il faut vider le cache du navigateur.

skapogo, 2020-03-12:
ok j'ai vider le cache on voit de nouveau les IP de desitnation
par contre PLEX bloque toujours alors que avant la mise a jour tout était ok

Ysurac, 2020-03-12:
Le VPN fonctionne correctement ?
Dans Etat->Journal Système il y a quoi ?

skapogo, 2020-03-12:
[image]

Journal :
Thu Mar 12 14:19:26 2020 user.notice OMR-VPS: Can't get token, try later
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/dsvpn
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/mptcpovervpn
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: add serial interface ttyI40: event processed
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/omr6in4
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI40, subsystem=tty
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/omr-quota
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/omr-tracker
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/openmptcprouter
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/wireless reload dependency on /etc/config/network
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: add serial interface ttyI41: event processed
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI41, subsystem=tty
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/firewall reload dependency on /etc/config/luci-splash
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/firewall reload dependency on /etc/config/qos
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/firewall reload dependency on /etc/config/miniupnpd
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/firewall reload dependency on /etc/config/sqm
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: add serial interface ttyI42: event processed
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI42, subsystem=tty
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/dhcp reload dependency on /etc/config/odhcpd
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: add serial interface ttyI43: event processed
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI43, subsystem=tty
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up non-init /etc/config/fstab reload handler: /sbin/block mount
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/system reload trigger for non-procd /etc/init.d/led
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: add serial interface ttyI44: event processed
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI44, subsystem=tty
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/system reload dependency on /etc/config/luci_statistics
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/system reload dependency on /etc/config/dhcp
Thu Mar 12 14:19:26 2020 user.notice firewall: Reloading firewall due to ifup of wan1 (wan1)
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: add serial interface ttyI45: event processed
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/glorytun reload dependency on /etc/config/glorytun-udp
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI45, subsystem=tty
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/glorytun reload dependency on /etc/config/openmptcprouter-vps
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: add serial interface ttyI46: event processed
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI46, subsystem=tty
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:26 2020 user.notice OMR-VPS: Can't get token, try later
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: add serial interface ttyI47: event processed
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI47, subsystem=tty
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: add serial interface ttyI48: event processed
Thu Mar 12 14:19:26 2020 user.notice ucitrack: Setting up /etc/config/shadowsocks-libev reload dependency on /etc/config/omr-tracker
Thu Mar 12 14:19:26 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI48, subsystem=tty
Thu Mar 12 14:19:27 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
[... cut ...]
Thu Mar 12 14:19:57 2020 daemon.info unbound: [6640:0] info:    0.032768    0.065536 1
Thu Mar 12 14:19:57 2020 daemon.info unbound: [6640:0] info:    0.131072    0.262144 3
Thu Mar 12 14:19:57 2020 daemon.info unbound: [6640:0] info:    0.262144    0.524288 3
Thu Mar 12 14:19:57 2020 daemon.info unbound: [6640:0] info:    0.524288    1.000000 3
Thu Mar 12 14:19:57 2020 daemon.notice unbound: [14497:0] notice: init module 0: iterator
Thu Mar 12 14:19:58 2020 daemon.info unbound: [14497:0] info: start of service (unbound 1.10.0).

skapogo, 2020-03-12:
il en manquait journal complet :
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty24, subsystem=tty
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty52, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty52
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty52, subsystem=tty
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: add serial interface ttyI63: event processed
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI63, subsystem=tty
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty14, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty14
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty14, subsystem=tty
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty6, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty6
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty6, subsystem=tty
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: add network interface teql0: event processed
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: event reported: action=add, name=teql0, subsystem=net
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty42, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty42
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty42, subsystem=tty
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty32, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty32
Thu Mar 12 14:19:28 2020 user.notice mptcp: Reloading mptcp config due to ifup of loopback (lo)
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty32, subsystem=tty
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty60, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty60
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty60, subsystem=tty
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty22, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty22
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty22, subsystem=tty
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty50, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty50
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty50, subsystem=tty
Thu Mar 12 14:19:28 2020 daemon.notice wpa_supplicant[5907]: Successfully initialized wpa_supplicant
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: add network interface wlan0: event processed
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty12, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty12
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: event reported: action=add, name=wlan0, subsystem=net
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty12, subsystem=tty
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty4, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty4
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty4, subsystem=tty
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty40, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty40
Thu Mar 12 14:19:28 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty40, subsystem=tty
[... cut ...]
Thu Mar 12 14:20:12 2020 daemon.info vnstatd[23735]: Info: vnStat daemon 2.4 started. (pid:23735 uid:0 gid:0 64-bit)
Thu Mar 12 14:20:12 2020 daemon.info vnstatd[23735]: Info: Monitoring (8): wlan0 (1000 Mbit) wan2 (1000 Mbit) wan1 (1000 Mbit) tun0 (10 Mbit) teql0 (1000 Mbit) ifb1 (1000 Mbit) ifb0 (1000 Mbit) eth0 (1000 Mbit)
Thu Mar 12 14:22:37 2020 daemon.err /usr/bin/ss-redir[23421]: remote recv: Operation timed out
Thu Mar 12 14:22:37 2020 daemon.err /usr/bin/ss-redir[23421]: remote recv: Operation timed out

Ysurac, 2020-03-12:
Quel est le VPN utilisé ?
Dans ```/etc/shorewall/rules``` sur le VPS il y a bien les régles DNAT pour PLEX ajoutées ?
Toujours sur le VPS, quel est le contenu de ```/etc/shorewall/params.vpn``` ?

skapogo, 2020-03-12:
> Quel est le VPN utilisé ?
le vpn par défault je n'ai rien modifié dans OpenMPTCProuter

[image]
[image]

Ysurac, 2020-03-12:
Rien plus bas dans le fichier rules ? car la il n'y a effectivement aucune redirection.
Tu peux me donner le résultat de ```uci show firewall``` via SSH sur le routeur ?

skapogo, 2020-03-12:
> Rien plus bas dans le fichier rules ?
non rien de plus , je précise le parefeu ovh est desactivé, et ca fonctionnait avant la mise a jour.

uci show firewall :
 
root@OpenMPTCProuter:~# uci show firewall
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].disable_ipv6='1'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].network='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[0].mtu_fix='1'
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
firewall.@zone[1].network='wan' 'wan6' 'wan1'
firewall.@zone[1].input='REJECT'
firewall.@zone[1].output='ACCEPT'
firewall.@zone[1].forward='REJECT'
firewall.@zone[1].masq='1'
firewall.@zone[1].mtu_fix='1'
firewall.@forwarding[0]=forwarding
firewall.@forwarding[0].src='lan'
firewall.@forwarding[0].dest='wan'
firewall.@rule[0]=rule
firewall.@rule[0].name='Allow-DHCP-Renew'
firewall.@rule[0].src='wan'
firewall.@rule[0].proto='udp'
firewall.@rule[0].dest_port='68'
firewall.@rule[0].target='ACCEPT'
firewall.@rule[0].family='ipv4'
firewall.@rule[1]=rule
firewall.@rule[1].name='Allow-Ping'
firewall.@rule[1].src='wan'
firewall.@rule[1].proto='icmp'
firewall.@rule[1].icmp_type='echo-request'
firewall.@rule[1].family='ipv4'
firewall.@rule[1].target='ACCEPT'
firewall.@rule[2]=rule
firewall.@rule[2].name='Allow-IGMP'
firewall.@rule[2].src='wan'
firewall.@rule[2].proto='igmp'
firewall.@rule[2].family='ipv4'
firewall.@rule[2].target='ACCEPT'
firewall.@rule[3]=rule
firewall.@rule[3].name='Allow-DHCPv6'
firewall.@rule[3].src='wan'
firewall.@rule[3].proto='udp'
firewall.@rule[3].src_ip='fc00::/6'
firewall.@rule[3].dest_ip='fc00::/6'
firewall.@rule[3].dest_port='546'
firewall.@rule[3].family='ipv6'
firewall.@rule[3].target='ACCEPT'
firewall.@rule[4]=rule
firewall.@rule[4].name='Allow-MLD'
firewall.@rule[4].src='wan'
firewall.@rule[4].proto='icmp'
firewall.@rule[4].src_ip='fe80::/10'
firewall.@rule[4].icmp_type='130/0' '131/0' '132/0' '143/0'
firewall.@rule[4].family='ipv6'
firewall.@rule[4].target='ACCEPT'
firewall.@rule[5]=rule
firewall.@rule[5].name='Allow-ICMPv6-Input'
firewall.@rule[5].src='wan'
[... 74 lines cut ...]
firewall.omr_server.reload='1'
firewall.allow_dhcp_request_vpn=rule
firewall.allow_dhcp_request_vpn.name='Allow-DHCP-Request-VPN'
firewall.allow_dhcp_request_vpn.src='vpn'
firewall.allow_dhcp_request_vpn.proto='udp'
firewall.allow_dhcp_request_vpn.dest_port='67'
firewall.allow_dhcp_request_vpn.target='ACCEPT'
firewall.allow_dhcp_request_vpn.family='ipv4'
firewall.miniupnpd=include
firewall.miniupnpd.type='script'
firewall.miniupnpd.path='/usr/share/miniupnpd/firewall.include'
firewall.miniupnpd.family='any'
firewall.miniupnpd.reload='1'
firewall.ss_rules=include
firewall.ss_rules.path='/etc/firewall.ss-rules'
firewall.ss_rules.reload='1'
firewall.@redirect[0]=redirect
firewall.@redirect[0].target='DNAT'
firewall.@redirect[0].name='PLEX'
firewall.@redirect[0].proto='tcp udp'
firewall.@redirect[0].src='vpn'
firewall.@redirect[0].src_dport='32400'
firewall.@redirect[0].dest='lan'
firewall.@redirect[0].dest_ip='192.168.1.120'
firewall.@redirect[0].dest_port='32400'

Ysurac, 2020-03-12:
Etrange ça semble bon. Tu peux me donner le résultat de ```uci show openmptcprouter``` via SSH sur le routeur (en virant la clef) ?

skapogo, 2020-03-12:
root@OpenMPTCProuter:~# uci show openmptcprouter
openmptcprouter.settings=settings
openmptcprouter.settings.enabled='1'
openmptcprouter.settings.disable_ipv6='1'
openmptcprouter.settings.check_ipv4_website='http://ip.openmptcprouter.com'
openmptcprouter.settings.check_ipv6_website='http://ipv6.openmptcprouter.com'
openmptcprouter.settings.status_vps_timeout='2'
openmptcprouter.settings.status_getip_timeout='2'
openmptcprouter.settings.ha='0'
openmptcprouter.settings.shadowsocks_disable='0'
openmptcprouter.settings.firstboot='0'
openmptcprouter.settings.master='change'
openmptcprouter.settings.version='0.54.1'
openmptcprouter.settings.vpn='dsvpn'
openmptcprouter.vps=server
openmptcprouter.vps.username='openmptcprouter'
openmptcprouter.vps.master='1'
openmptcprouter.vps.backup='0'
openmptcprouter.vps.port='65500'
openmptcprouter.vps.password='BB8D1817FF412F14E6BB5059257960555F094A0F05C7E8B0A1C49D42C06B7AF8'
openmptcprouter.vps.ip='51.254.126.233'
openmptcprouter.vps.redirect_ports='0'
openmptcprouter.vps.pihole='0'
openmptcprouter.vps.kernel='4.19.104-mptcp'
openmptcprouter.vps.machine='x86_64'
openmptcprouter.vps.omr_version='0.1013'
openmptcprouter.vps.available_vpn='glorytun_tcp' 'glorytun_udp' 'dsvpn' 'openvpn' 'mlvpn'
openmptcprouter.vps.get_config='0'
openmptcprouter.vps.admin_error='0'
openmptcprouter.vps.token='eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJzdWIiOiJvcGVubXB0Y3Byb3V0ZXIiLCJleHAiOjE1ODQxMTk3MzB9.HvE_bCDX7SnPaeAq7gXRKAqKhayU991nzxpajuxFCWA'
openmptcprouter.vps.lastchange='1584033334'
openmptcprouter.omr=router
openmptcprouter.omr.shadowsocks='up'
openmptcprouter.omr.detected_public_ipv4='51.254.126.233'
openmptcprouter.omr.detected_ss_ipv4='51.254.126.233'
openmptcprouter.wan1=interface
openmptcprouter.wan1.metric='3'
openmptcprouter.wan1.multipathvpn='0'
openmptcprouter.wan1.mptcp_status='MPTCP enabled'
openmptcprouter.wan1.asn='FREEM'
openmptcprouter.wan1.mtu='1500'
openmptcprouter.wan1.state='up'
openmptcprouter.wan1.publicip='37.164.100.239'
openmptcprouter.wan1.multipath='on'
openmptcprouter.wan1.lc='1584033296'
openmptcprouter.wan1.latency_previous='73'
openmptcprouter.wan1.latency='48'
openmptcprouter.wan2=interface
openmptcprouter.wan2.multipathvpn='0'
openmptcprouter.wan2.metric='6'
openmptcprouter.wan2.mptcp_status='MPTCP enabled'
openmptcprouter.wan2.publicip='88.123.25.222'
openmptcprouter.wan2.asn='PROXAD'
openmptcprouter.wan2.mtu='1500'
openmptcprouter.wan2.multipath='backup'
openmptcprouter.wan2.state='up'
openmptcprouter.wan2.manufacturer='huawei'
openmptcprouter.wan2.lc='1584033329'
openmptcprouter.wan2.latency='22'
openmptcprouter.wan2.latency_previous='22'
openmptcprouter.omrvpn=interface
openmptcprouter.omrvpn.multipath='off'
openmptcprouter.omrvpn.metric='4'
openmptcprouter.omrvpn.state='down'
openmptcprouter.latest_versions=latest_versions
openmptcprouter.latest_versions.vps='0.1013'
openmptcprouter.latest_versions.omr='0.54.1'
openmptcprouter.latest_versions.lc='1584033296'
root@OpenMPTCProuter:~#

skapogo, 2020-03-12:
openmptcprouter.settings=settings
openmptcprouter.settings.enabled='1'
openmptcprouter.settings.disable_ipv6='1'
openmptcprouter.settings.check_ipv4_website='http://ip.openmptcprouter.com'
openmptcprouter.settings.check_ipv6_website='http://ipv6.openmptcprouter.com'
openmptcprouter.settings.status_vps_timeout='2'
openmptcprouter.settings.status_getip_timeout='2'
openmptcprouter.settings.ha='0'
openmptcprouter.settings.shadowsocks_disable='0'
openmptcprouter.settings.firstboot='0'
openmptcprouter.settings.master='change'
openmptcprouter.settings.version='0.54.1'
openmptcprouter.settings.vpn='dsvpn'
openmptcprouter.vps=server
openmptcprouter.vps.username='openmptcprouter'
openmptcprouter.vps.master='1'
openmptcprouter.vps.backup='0'
openmptcprouter.vps.port='65500'
openmptcprouter.vps.password='BB8D1817FF412F14E6BB5059257960555F094A0F05C7E8B0A1C49D42C06B7AF8'
openmptcprouter.vps.ip='51.254.126.233'
openmptcprouter.vps.redirect_ports='0'
openmptcprouter.vps.pihole='0'
openmptcprouter.vps.kernel='4.19.104-mptcp'
openmptcprouter.vps.machine='x86_64'
openmptcprouter.vps.omr_version='0.1013'
openmptcprouter.vps.available_vpn='glorytun_tcp' 'glorytun_udp' 'dsvpn' 'openvpn' 'mlvpn'
openmptcprouter.vps.get_config='0'
openmptcprouter.vps.admin_error='0'
openmptcprouter.vps.token='eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJzdWIiOiJvcGVubXB0Y3Byb3V0ZXIiLCJleHAiOjE1ODQxMTk3MzB9.HvE_bCDX7SnPaeAq7gXRKAqKhayU991nzxpajuxFCWA'
openmptcprouter.vps.lastchange='1584033334'
openmptcprouter.omr=router
openmptcprouter.omr.shadowsocks='up'
openmptcprouter.omr.detected_public_ipv4='51.254.126.233'
openmptcprouter.omr.detected_ss_ipv4='51.254.126.233'
openmptcprouter.wan1=interface
openmptcprouter.wan1.metric='3'
openmptcprouter.wan1.multipathvpn='0'
openmptcprouter.wan1.mptcp_status='MPTCP enabled'
openmptcprouter.wan1.asn='FREEM'
openmptcprouter.wan1.mtu='1500'
openmptcprouter.wan1.state='up'
openmptcprouter.wan1.publicip='37.164.100.239'
openmptcprouter.wan1.multipath='on'
openmptcprouter.wan1.lc='1584033296'
openmptcprouter.wan1.latency_previous='73'
openmptcprouter.wan1.latency='48'
openmptcprouter.wan2=interface
openmptcprouter.wan2.multipathvpn='0'
openmptcprouter.wan2.metric='6'
openmptcprouter.wan2.mptcp_status='MPTCP enabled'
openmptcprouter.wan2.publicip='88.123.25.222'
openmptcprouter.wan2.asn='PROXAD'
openmptcprouter.wan2.mtu='1500'
openmptcprouter.wan2.multipath='backup'
openmptcprouter.wan2.state='up'
openmptcprouter.wan2.manufacturer='huawei'
openmptcprouter.wan2.lc='1584033329'
openmptcprouter.wan2.latency='22'
openmptcprouter.wan2.latency_previous='22'
openmptcprouter.omrvpn=interface
openmptcprouter.omrvpn.multipath='off'
openmptcprouter.omrvpn.metric='4'
openmptcprouter.omrvpn.state='down'
openmptcprouter.latest_versions=latest_versions
openmptcprouter.latest_versions.vps='0.1013'
openmptcprouter.latest_versions.omr='0.54.1'
openmptcprouter.latest_versions.lc='1584033296'

Ysurac, 2020-03-12:
Il y a un bug, j'ai corrigé et c'est en cours de compilation.

skapogo, 2020-03-12:
merci encore pour ton tout ta solution est vraiment super pratique.

du coup il faut mettre a jour le VPS ?

Ysurac, 2020-03-12:
Non, ça sera du côté du routeur. La version 0.54.2.
Mais il faut lui laisser le temps de compiler.

skapogo, 2020-03-12:
je viens de mettre a jour toujours le même problème. le port Forwarding ne fonctionne pas

skapogo, 2020-03-12:
[image]

skapogo, 2020-03-13:
Merci pour ton patch Ysurac, je viens d'essayer ça fonctionne par contre bizarrement dans l'administration du serveur Plex il est  dit que ça ne fonctionne pas. 
Après de multiples essais l'accès a distance est bien  fonctionnel 
[image]

skapogo, 2020-03-13:
alors que avant la mise à jour tout était au vert

skapogo, 2020-03-13:
tout est passé au vert aujourd'hui Merci encore

closed 2020-03-13 by skapogo