Ysurac/openmptcprouter #887: No way to backup or update

patimages, 2020-03-12:
Hi, 
I cannot backup on 0.54.0 (404 error: Unable to dispatch: /cgi-bin/luci/admin/system/undefined/cgi-backup) and cannot update to 0.54.1 (XHR request timed out) : Do I need to reinstall ? tricky without backup .... Can I update via command line ? If yes, how ? 

Thanks !

bnhf, 2020-03-12:
I'm also on the x86-64 version of OMR 0.54.0, but was able to both backup locally and to the server. I used the ext4 version of the image though.

Do you have a complicated setup on your OMR device? I did quite a few builds from scratch when I was first trying to get my VLAN based setup running on an RPi4 -- and it was a pretty quick process.  The IP address of the VPS, the server key, and the particulars of my WAN interfaces was all I needed. Yours could be more involved of course.

Ysurac, 2020-03-13:
You should try to clear your browser cache (this seems to solve most issues).

patimages, 2020-03-13:
Clearing the cache solve it ! Merci and thanks !

closed 2020-03-13 by Ysurac