Ysurac/openmptcprouter #888: BUG : Bug depuis la maj vers la 0.54.2

Slivikkus, 2020-03-12:
## Actual Behavior
J'ai mis à jour il y a quelques minutes vers la version 0.1014 côté VPS et 0.54.2 côté routeur mais j'ai des bug depuis ...
[image]
 
il considère aussi que une de mes connexion est down alors qu'elle fonctionne (ça clignote même)
[image]

## Specifications

  - OpenMPTCProuter version: (0.54.2)
  - OpenMPTCProuter VPS version: (0.1014)
  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2020-03-13:
Il faut voir dans Etat->Journal Système la raison du soucis.

Slivikkus, 2020-03-13:
> Fri Mar 13 10:05:49 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:05:49 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:05:49 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 10:05:50 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:05:50 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:05:50 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:05:51 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:05:51 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:05:51 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:05:52 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:05:52 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:05:52 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:05:53 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 10:05:53 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:05:53 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:05:53 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:05:54 2020 daemon.err /usr/bin/ss-local[17633]: remote_recv_cb_recv: Connection reset by peer
> Fri Mar 13 10:05:54 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:05:54 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:05:54 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:05:55 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:05:55 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:05:55 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:05:56 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:05:56 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:05:56 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:05:57 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:05:57 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:05:57 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:05:58 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:05:58 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:05:58 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:05:59 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:06:00 2020 daemon.err glorytun[6913]: read: Connection reset by peer
> Fri Mar 13 10:06:00 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:06:00 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:06:01 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:06:01 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:06:01 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:06:01 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 10:06:02 2020 daemon.err /usr/bin/ss-local[17633]: remote_recv_cb_recv: Connection reset by peer
> Fri Mar 13 10:06:02 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:06:02 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:06:02 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:06:03 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:06:03 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:06:03 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:06:04 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:06:04 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:06:04 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:06:05 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:06:05 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:06:05 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:06:05 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 10:06:06 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:06:06 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:06:06 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:06:07 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:06:07 2020 daemon.err glorytun[6913]: read: Connection reset by peer
> Fri Mar 13 10:06:07 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:06:07 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:06:08 2020 daemon.err /usr/bin/ss-local[17633]: remote_recv_cb_recv: Connection reset by peer
> Fri Mar 13 10:06:08 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:06:12 2020 daemon.err glorytun[6913]: read: Connection reset by peer
> Fri Mar 13 10:06:12 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
[... cut ...]
> Fri Mar 13 10:10:40 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:10:40 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:10:41 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: connected
> Fri Mar 13 10:10:41 2020 daemon.err glorytun[6913]: 127.0.0.1.65001: key exchange failed
> Fri Mar 13 10:10:41 2020 daemon.info glorytun[6913]: STOPPED tun0
> Fri Mar 13 10:10:42 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)

Ysurac, 2020-03-13:
Le soucis est que l'API sur le serveur ne répond pas. Il faut voir dans ```/var/log/daemon.log``` si il y a une erreur.
Le serveur a été redémarré après la mise à jour ?

Slivikkus, 2020-03-13:
Oui le vps à été redémarrer après la maj

Slivikkus, 2020-03-13:
[image]
C'est tout ce que je peux montrer

Ysurac, 2020-03-13:
essaye un ```systemctl restart omr-admin``` sur le VPS

Slivikkus, 2020-03-13:
ça change rien :/

Ysurac, 2020-03-13:
que donne un ```curl -k https://127.0.0.1:65500/``` sur le VPS ?

Slivikkus, 2020-03-13:
[image]

Ysurac, 2020-03-13:
ok, donc la partie admin est bien lancée mais non joignable depuis le routeur.
Tu utilises plusieurs VPS visiblement, il va falloir que je re-teste ça...

Slivikkus, 2020-03-13:
Oui deux VPS, ça pose problème ?

Ysurac, 2020-03-13:
Dans le cas présent ça semble. Il faut que je refasse les tests avec plusieurs VPS, j'ai peut être fait des modifications qui font planter ce cas.

Slivikkus, 2020-03-13:
à mon avis la maj du VPS à du merder parce-que j'ai retiré le second VPS mais même problème et VPS injoignable par SSH

Slivikkus, 2020-03-13:
[image]
Finalement c'était pas une bonne idée de remettre à zéro le VPS

Ysurac, 2020-03-13:
Je n'ai aucun soucis avec plusieurs VPS, donc le soucis est ailleurs.
Il me refaudrait les logs sur le routeur et le VPS. Le VPN semble fonctionner maintenant, mais pas shadowsocks ni l'API.

Slivikkus, 2020-03-13:
> Fri Mar 13 19:47:01 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:47:05 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:47:06 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:47:09 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:47:10 2020 user.notice omr6in4: Set network for OMR 6in4 to local 10.255.255.2 peer 10.255.255.1
> Fri Mar 13 19:47:11 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:47:16 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:47:18 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:47:21 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:47:22 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:47:25 2020 user.notice omr6in4: Set network for OMR 6in4 to local 10.255.255.2 peer 10.255.255.1
> Fri Mar 13 19:47:27 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:47:29 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:47:32 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:47:33 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:47:37 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:47:40 2020 user.notice omr6in4: Set network for OMR 6in4 to local 10.255.255.2 peer 10.255.255.1
> Fri Mar 13 19:47:42 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:47:43 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:47:45 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:47:48 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:47:54 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:47:54 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:47:55 2020 user.notice omr6in4: Set network for OMR 6in4 to local 10.255.255.2 peer 10.255.255.1
> Fri Mar 13 19:47:56 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:47:59 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:48:04 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:48:05 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:48:09 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:48:10 2020 user.notice omr6in4: Set network for OMR 6in4 to local 10.255.255.2 peer 10.255.255.1
> Fri Mar 13 19:48:11 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:48:14 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:48:17 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:48:19 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:48:22 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:48:24 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:48:25 2020 user.notice omr6in4: Set network for OMR 6in4 to local 10.255.255.2 peer 10.255.255.1
> Fri Mar 13 19:48:28 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:48:29 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:48:34 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:48:35 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:48:40 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:48:40 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:48:40 2020 user.notice omr6in4: Set network for OMR 6in4 to local 10.255.255.2 peer 10.255.255.1
> Fri Mar 13 19:48:45 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:48:46 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:48:50 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:48:51 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 19:48:55 2020 user.notice omr6in4: Set network for OMR 6in4 to local 10.255.255.2 peer 10.255.255.1
> Fri Mar 13 19:48:55 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 19:48:58 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
[... cut ...]
> Fri Mar 13 20:14:26 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 20:14:27 2020 user.notice omr6in4: Set network for OMR 6in4 to local 10.255.255.2 peer 10.255.255.1
> Fri Mar 13 20:14:29 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Fri Mar 13 20:14:32 2020 daemon.err /usr/bin/ss-local[15027]: getpeername: Socket not connected
> Fri Mar 13 20:14:35 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)

Ysurac, 2020-03-13:
il faudrait un ```ps aux | grep ss``` sur le VPS.
Et il me faudra la configuration (sans les clefs) présente dans Système->OpenMPTCProuter et onglet "'Voir tous les paramètres"

Slivikkus, 2020-03-13:
> dhcp.@dnsmasq[0]=dnsmasq
> dhcp.@dnsmasq[0].domainneeded='1'
> dhcp.@dnsmasq[0].localise_queries='1'
> dhcp.@dnsmasq[0].rebind_protection='1'
> dhcp.@dnsmasq[0].rebind_localhost='1'
> dhcp.@dnsmasq[0].local='/lan/'
> dhcp.@dnsmasq[0].domain='lan'
> dhcp.@dnsmasq[0].expandhosts='1'
> dhcp.@dnsmasq[0].authoritative='1'
> dhcp.@dnsmasq[0].readethers='1'
> dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
> dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
> dhcp.@dnsmasq[0].localservice='1'
> dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/use-application-dns.net/'
> dhcp.@dnsmasq[0].noresolv='1'
> dhcp.@dnsmasq[0].nonegcache='1'
> dhcp.@dnsmasq[0].rebind_domain='plex.direct'
> dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
> dhcp.lan=dhcp
> dhcp.lan.interface='lan'
> dhcp.lan.start='100'
> dhcp.lan.limit='150'
> dhcp.lan.leasetime='12h'
> dhcp.lan.force='1'
> dhcp.lan.ra_slaac='1'
> dhcp.lan.ra_flags='managed-config' 'other-config'
> dhcp.lan.ra='server'
> dhcp.lan.ra_preference='high'
> dhcp.lan.ra_management='1'
> dhcp.lan.dhcpv6='server1'
> dhcp.lan.ra_default='1'
> dhcp.wan=dhcp
> dhcp.wan.interface='wan'
> dhcp.wan.ignore='1'
> dhcp.odhcpd=odhcpd
> dhcp.odhcpd.maindhcp='0'
> dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
> dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
> dhcp.odhcpd.loglevel='4'
> dhcp.@host[0]=host
> dhcp.@host[0].mac='00:0C:29:79:B0:CD'
> dhcp.@host[0].name='adky-vps-linux-1'
> dhcp.@host[0].dns='1'
> dhcp.@host[0].ip='192.168.100.115'
> dhcp.@host[1]=host
> dhcp.@host[1].mac='00:0C:29:79:54:25'
> dhcp.@host[1].name='adkynet.com'
> dhcp.@host[1].dns='1'
> dhcp.@host[1].ip='192.168.100.139'
> dhcp.@host[2]=host
> dhcp.@host[2].mac='00:0C:29:A0:98:BF'
> dhcp.@host[2].name='MailServer'
> dhcp.@host[2].dns='1'
> dhcp.@host[2].ip='192.168.100.140'
> dropbear.@dropbear[0]=dropbear
> dropbear.@dropbear[0].PasswordAuth='on'
> dropbear.@dropbear[0].RootPasswordAuth='on'
> dropbear.@dropbear[0].Port='22'
> dscp.@classify[0]=classify
> dscp.@classify[0].direction='both'
> dscp.@classify[0].proto='icmp'
> dscp.@classify[0].class='cs5'
> dscp.@classify[0].comment='ICMP'
> dscp.@classify[1]=classify
> dscp.@classify[1].direction='both'
> dscp.@classify[1].proto='udp'
> dscp.@classify[1].class='cs5'
> dscp.@classify[1].src_port='53,123,5353'
> dscp.@classify[1].comment='DNS udp and NTP'
> dscp.@classify[2]=classify
[... 1424 lines cut ...]
> upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
> upnpd.config.igdv1='0'
> upnpd.@perm_rule[0]=perm_rule
> upnpd.@perm_rule[0].action='allow'
> upnpd.@perm_rule[0].ext_ports='1024-65535'
> upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
> upnpd.@perm_rule[0].int_ports='1024-65535'
> upnpd.@perm_rule[0].comment='Allow high ports'
> upnpd.@perm_rule[1]=perm_rule
> upnpd.@perm_rule[1].action='deny'
> upnpd.@perm_rule[1].ext_ports='0-65535'
> upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
> upnpd.@perm_rule[1].int_ports='0-65535'
> upnpd.@perm_rule[1].comment='Default deny'
> vnstat.@vnstat[0]=vnstat
> vnstat.@vnstat[0].interface='eth2' 'eth3' 'eth2'
> wireless.@wifi-iface[0]=wifi-iface
> wireless.@wifi-iface[0].device='wlan0'
> wireless.@wifi-iface[0].mode='ap'
> wireless.@wifi-iface[0].ssid='AdKyNet.com'
> wireless.@wifi-iface[0].network='lan'
> wireless.@wifi-iface[0].encryption='psk2'
> wireless.@wifi-iface[0].key='StaffAdKyNet.com'
> wireless.@wifi-iface[0].ieee80211w='0'
> wireless.@wifi-iface[0].wps_pushbutton='1'

Ysurac, 2020-03-13:
Il y a une configuration étrange avec shadowsocks : IPv6 est désactivé globalement mais shadowsocks écoute en IPv6 ce qui peut poser un soucis.

Slivikkus, 2020-03-13:
Ah je peux régler ça ? Aussi je vois tout en bas de tout les paramètres que le wifi est config mais que ça n'a pas l'air de fonctionner ?

Ysurac, 2020-03-13:
Dans l'interface de shadowsocks, tu peux définir les ss_redir avec une adresse "0.0.0.0" au lieu de "::"

Slivikkus, 2020-03-13:
[image]
Tout est ok après réinstallation complète des deux côté

closed 2020-03-13 by Slivikkus