Ysurac/openmptcprouter #891: MPTCP over VPN not working

ccmks, 2020-03-13:
## Expected Behavior
Expect to get MPTCP work over VPN for MPTCP blocked connection
  

## Actual Behavior
Doesn't get MPTCP to work over VPN, or I might be missing some settings

## Steps to Reproduce the Problem

  1. Flash my Banana Pi RP2 with v0.53.3 r0+11546-a74095c68c (I never get to work with v0.54 because the device will be in reboot loop when flashing to v0.54)
  2. Put the server key and VPS IP address and set all WAN to be DHCP
  3. 1st WAN works with MPTCP, but the 2nd WAN is not working with MPTCP, however, it doesn't say the MPTCP is being blocked on the status page. But still doesn't do aggregation when check on bandwidth while performing download test on PC

## Specifications

  - OpenMPTCProuter version: v0.53.3 r0+11546-a74095c68c
  - OpenMPTCProuter VPS version: 0.1014 4.19.104-mptcp
  - OpenMPTCProuter platform: Banana Pi RP2

[image]

Ysurac, 2020-03-13:
No support on old release, so I'm checking why OpenMPTCProuter reboot on BPI-R2.

ccmks, 2020-03-14:
I got the MPTCP working over VPN, just didn't see the extra setting on each
WAN in the first place.

But still I can't flash the v0.54 to Banana Pi R2.

Keeps rebooting in loop

Ysurac, 2020-03-14:
I work on free time, so no. For now I can't make it work with current kernel.

closed 2020-09-14 by github-actions[bot]