Ysurac/openmptcprouter #892: problème de déconnexion

Seb944, 2020-03-13:
OpenMPTCProuter version: (0.54.2)
OpenMPTCProuter VPS version: (0.1014)
OpenMPTCProuter platform: (x86_64)

J'ai beaucoup de problèmes de déconnexion.

Voici le journal système.

Merci

Fri Mar 13 18:45:41 2020 daemon.err /usr/bin/ss-local[23856]: getpeername: Socket not connected
Fri Mar 13 18:45:48 2020 daemon.err /usr/bin/ss-redir[23857]: remote recv: Operation timed out
Fri Mar 13 18:45:48 2020 daemon.err /usr/bin/ss-redir[23858]: remote recv: Operation timed out
Fri Mar 13 18:45:48 2020 daemon.err /usr/bin/ss-redir[23857]: remote recv: Operation timed out
Fri Mar 13 18:45:49 2020 daemon.err /usr/bin/ss-redir[23859]: remote recv: Operation timed out
Fri Mar 13 18:45:51 2020 daemon.err /usr/bin/ss-redir[23859]: remote recv: Operation timed out
Fri Mar 13 18:45:51 2020 daemon.err /usr/bin/ss-redir[23857]: remote recv: Operation timed out
Fri Mar 13 18:45:52 2020 daemon.err /usr/bin/ss-redir[23860]: remote recv: Operation timed out
Fri Mar 13 18:45:59 2020 daemon.err /usr/bin/ss-local[23856]: getpeername: Socket not connected
Fri Mar 13 18:46:03 2020 daemon.err /usr/bin/ss-redir[23857]: remote recv: Operation timed out
Fri Mar 13 18:46:05 2020 daemon.err /usr/bin/ss-local[23856]: getpeername: Socket not connected
Fri Mar 13 18:46:12 2020 daemon.err /usr/bin/ss-local[23856]: getpeername: Socket not connected
Fri Mar 13 18:46:20 2020 daemon.err /usr/bin/ss-local[23856]: getpeername: Socket not connected
Fri Mar 13 18:46:23 2020 daemon.err /usr/bin/ss-local[23856]: getpeername: Socket not connected
Fri Mar 13 18:46:23 2020 daemon.err /usr/bin/ss-redir[23858]: remote recv: Operation timed out
Fri Mar 13 18:46:29 2020 daemon.err /usr/bin/ss-redir[23859]: remote recv: Operation timed out
Fri Mar 13 18:46:34 2020 daemon.err /usr/bin/ss-local[23856]: getpeername: Socket not connected
Fri Mar 13 18:46:35 2020 daemon.err /usr/bin/ss-redir[23858]: remote recv: Operation timed out
Fri Mar 13 18:46:43 2020 daemon.err /usr/bin/ss-redir[23859]: remote recv: Operation timed out
Fri Mar 13 18:46:46 2020 daemon.err /usr/bin/ss-redir[23857]: remote recv: Operation timed out
Fri Mar 13 18:46:47 2020 daemon.err /usr/bin/ss-redir[23860]: remote recv: Operation timed out
Fri Mar 13 18:46:52 2020 daemon.err /usr/bin/ss-local[23856]: getpeername: Socket not connected

Seb944, 2020-03-13:
complément de log

Fri Mar 13 18:51:34 2020 daemon.err /usr/bin/ss-local[23856]: getpeername: Socket not connected
Fri Mar 13 18:51:34 2020 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact via http 88.191.250.176, 151.101.129.164, 1.1.1.1)
Fri Mar 13 18:51:34 2020 user.notice Shadowsocks: Rules DOWN
Fri Mar 13 18:51:44 2020 daemon.err /usr/bin/ss-local[23856]: getpeername: Socket not connected
Fri Mar 13 18:51:52 2020 daemon.err /usr/bin/ss-local[23856]: getpeername: Socket not connected
Fri Mar 13 18:52:00 2020 daemon.err /usr/bin/ss-local[23856]: getpeername: Socket not connected
Fri Mar 13 18:52:06 2020 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 103.224.182.242)
Fri Mar 13 18:52:06 2020 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Fri Mar 13 18:52:06 2020 user.notice Shadowsocks: Rules UP
Fri Mar 13 18:52:06 2020 user.notice Shadowsocks: Reload omr-bypass rules
Fri Mar 13 18:52:06 2020 user.notice omr-bypass: Starting OMR-ByPass...
Fri Mar 13 18:52:06 2020 user.notice omr-bypass: Reload dnsmasq...
Fri Mar 13 18:52:06 2020 daemon.info dnsmasq[29515]: read /etc/hosts - 4 addresses
Fri Mar 13 18:52:06 2020 daemon.info dnsmasq[29515]: read /tmp/hosts/dhcp.cfg01411c - 3 addresses
Fri Mar 13 18:52:06 2020 daemon.info dnsmasq-dhcp[29515]: read /etc/ethers - 0 addresses
Fri Mar 13 18:52:06 2020 user.notice omr-bypass: OMR-ByPass is running
Fri Mar 13 18:52:10 2020 daemon.err /usr/bin/ss-redir[23860]: remote recv: Operation timed out
Fri Mar 13 18:52:13 2020 daemon.err /usr/bin/ss-redir[23857]: remote recv: Operation timed out

Ysurac, 2020-03-13:
Il me faudrait aussi le fichier ```/var/log/daemon.log``` sur le VPS.
Le VPS a bien été redémarré après la mise à jour ?

Seb944, 2020-03-13:
Mar 13 19:11:15 SebVPS ss-server[538]: remote recv: Connection reset by peer
Mar 13 19:11:18 SebVPS ss-server[538]: remote recv: Connection reset by peer
Mar 13 19:11:54 SebVPS ss-server[538]: remote recv: Connection reset by peer
Mar 13 19:12:35 SebVPS omr-admin.py[483]: DEBUG:    password true
Mar 13 19:12:35 SebVPS omr-admin.py[483]: DEBUG:    Get config...
Mar 13 19:12:35 SebVPS omr-admin.py[483]: DEBUG:    Get config... shadowsocks
Mar 13 19:12:35 SebVPS omr-admin.py[483]: DEBUG:    Get config... glorytun
Mar 13 19:12:35 SebVPS omr-admin.py[483]: DEBUG:    Get config... dsvpn
Mar 13 19:12:35 SebVPS omr-admin.py[483]: DEBUG:    Get config... iperf3
Mar 13 19:12:35 SebVPS omr-admin.py[483]: DEBUG:    Get config... openvpn
Mar 13 19:12:35 SebVPS omr-admin.py[483]: DEBUG:    Get config... mlvpn
Mar 13 19:12:35 SebVPS omr-admin.py[483]: DEBUG:    Get config... mptcp
Mar 13 19:12:35 SebVPS omr-admin.py[483]: DEBUG:    Get config... ipv6
Mar 13 19:12:35 SebVPS omr-admin.py[483]: DEBUG:    get server IPv4
Mar 13 19:12:35 SebVPS omr-admin.py[483]: DEBUG:    get hostname
Mar 13 19:12:35 SebVPS omr-admin.py[483]: DEBUG:    Get config: done
Mar 13 19:12:36 SebVPS omr-admin.py[483]: DEBUG:    password true
Mar 13 19:12:36 SebVPS omr-admin.py[483]: DEBUG:    password true
Mar 13 19:12:36 SebVPS omr-admin.py[483]: DEBUG:    Get config...
Mar 13 19:12:36 SebVPS omr-admin.py[483]: DEBUG:    Get config... shadowsocks
Mar 13 19:12:36 SebVPS omr-admin.py[483]: DEBUG:    Get config... glorytun
Mar 13 19:12:36 SebVPS omr-admin.py[483]: DEBUG:    Get config... dsvpn
Mar 13 19:12:36 SebVPS omr-admin.py[483]: DEBUG:    Get config... iperf3
Mar 13 19:12:36 SebVPS omr-admin.py[483]: DEBUG:    Get config... openvpn
Mar 13 19:12:36 SebVPS omr-admin.py[483]: DEBUG:    Get config... mlvpn
Mar 13 19:12:36 SebVPS omr-admin.py[483]: DEBUG:    Get config... mptcp
Mar 13 19:12:36 SebVPS omr-admin.py[483]: DEBUG:    Get config... ipv6
Mar 13 19:12:36 SebVPS omr-admin.py[483]: DEBUG:    get server IPv4
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    get hostname
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config: done
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config...
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config... shadowsocks
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config... glorytun
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config... dsvpn
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config... iperf3
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config... openvpn
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config... mlvpn
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config... mptcp
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config... ipv6
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    get server IPv4
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    get hostname
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config: done
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config...
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config... shadowsocks
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config... glorytun
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config... dsvpn
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config... iperf3
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config... openvpn
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config... mlvpn
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config... mptcp
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config... ipv6
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    get server IPv4
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    get hostname
Mar 13 19:12:37 SebVPS omr-admin.py[483]: DEBUG:    Get config: done
Mar 13 19:12:38 SebVPS omr-admin.py[483]: DEBUG:    Get config...
Mar 13 19:12:38 SebVPS omr-admin.py[483]: DEBUG:    Get config... shadowsocks
Mar 13 19:12:38 SebVPS omr-admin.py[483]: DEBUG:    Get config... glorytun
Mar 13 19:12:38 SebVPS omr-admin.py[483]: DEBUG:    Get config... dsvpn
Mar 13 19:12:38 SebVPS omr-admin.py[483]: DEBUG:    Get config... iperf3
Mar 13 19:12:38 SebVPS omr-admin.py[483]: DEBUG:    Get config... openvpn
Mar 13 19:12:38 SebVPS omr-admin.py[483]: DEBUG:    Get config... mlvpn
Mar 13 19:12:38 SebVPS omr-admin.py[483]: DEBUG:    Get config... mptcp
Mar 13 19:12:38 SebVPS omr-admin.py[483]: DEBUG:    Get config... ipv6
Mar 13 19:12:38 SebVPS omr-admin.py[483]: DEBUG:    get server IPv4
Mar 13 19:12:38 SebVPS omr-admin.py[483]: DEBUG:    get hostname
Mar 13 19:12:38 SebVPS omr-admin.py[483]: DEBUG:    Get config: done
Mar 13 19:15:36 SebVPS systemd[1]: Created slice User Slice of root.
Mar 13 19:15:36 SebVPS systemd[1]: Starting User Manager for UID 0...
Mar 13 19:15:36 SebVPS systemd[1]: Started Session 167 of user root.
Mar 13 19:15:36 SebVPS systemd[11033]: Listening on GnuPG cryptographic agent an                                                                                                                                                             d passphrase cache.
Mar 13 19:15:36 SebVPS systemd[11033]: Listening on GnuPG network certificate ma                                                                                                                                                             nagement daemon.
[... cut ...]
Mar 13 19:15:43 SebVPS systemd-networkd[332]: gt-udp-tun0: Lost carrier
Mar 13 19:15:43 SebVPS glorytun-udp-run[11235]: running on device gt-udp-tun0 as                                                                                                                                                              pid 11235
Mar 13 19:15:43 SebVPS post.sh[11236]: RTNETLINK answers: File exists
Mar 13 19:15:43 SebVPS systemd[1]: Started Glorytun UDP on tun0.
Mar 13 19:15:43 SebVPS systemd-networkd[332]: gt-udp-tun0: Gained carrier

Oui je reboot toujours le vps

Ysurac, 2020-03-13:
Un peu cours le log sur le VPS, il m'en faudrait un peu plus, notamment ce qui peut concerner ss-manager/ss-server.
Et que donne un ```ps aux | grep ss``` sur le VPS ?

Seb944, 2020-03-13:
root@SebVPS:~# ps aux | grep ss
root       526  0.0  0.1   9832  2948 ?        Ss   Mar12   0:01 /usr/bin/ss-manager -c /etc/shadowsocks-libev/manager.json
root       538  0.0  0.4  17828  9236 ?        Ss   Mar12   0:53 ss-server --manager-address 127.0.0.1:8839 -f /root/.shadowsocks/.shadowsocks_65101.pid -c /root/.shadowsocks/.shadowsocks_65101.conf --acl /etc/shadowsocks-libev/local.acl -t 600 -u --fast-open --reuse-port --no-delay --mptcp -6 -s 0.0.0.0
message+   591  0.0  0.2  10908  4288 ?        Ss   Mar12   0:00 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only
root      1005  0.0  0.2  19244  5892 ?        Ss   Mar12   0:00 /usr/sbin/sshd -D
root     14452  0.1  0.3  42500  7176 ?        Ss   19:30   0:00 sshd: root@pts/0
root     14698  0.0  0.0   6080   888 pts/0    S+   19:30   0:00 grep ss

Ysurac, 2020-03-13:
je ne vois pas de soucis.
La latence entre le VPS et chaque connexion est de combien ? (visible dans la page status du routeur).
Un seul VPS est configuré ?

Seb944, 2020-03-13:
Oui un seul vps de configuré.
Je suis a 25ms sur le master et 33ms sur l'autre

closed 2020-09-14 by github-actions[bot]