Ysurac/openmptcprouter #894: DMZ - Forward all port from VPS directly to client behind OpenMPTCPRouter

ccmks, 2020-03-14:
How can I forward all port from VPS directly to a client behind OpenMPTCPRouter?

Usually, on consumer router, we will just look for DMZ which open all port for specific client behind the router.

OpenMPTCProuter version: v0.53.3 r0+11546-a74095c68c
OpenMPTCProuter VPS version: 0.1014 4.19.104-mptcp
OpenMPTCProuter platform: Banana Pi R2

Eltarius, 2020-03-15:
Like this ????
[image]

(Network > Firewall > Port forwards)

Try it, it seem work

ccmks, 2020-03-15:
Doesn't work for me. I don't have option to select protocol as "any"

Eltarius, 2020-03-15:
For me:
[image]

Can you print your screen?
I'm on v0.54.2, perhaps v0.53.3 does not have this option

ccmks, 2020-03-15:
Here is mine

[image]

I can update to v0.54 because I am on Banana Pi R2 which will keep booting in loop after flashing the v0.54

maxou56, 2020-03-22:
If "Any" is not available.
Select for example: TCP + UDP and for external port "1-64999"
And check "Redirects all ports from server to OpenMPTCProuter" (System - OpenMPTCProuter - advanced settings)

ccmks, 2020-03-23:
Tried that too, still not working. I guess the Banana Pi version is not
working well

closed 2020-09-14 by github-actions[bot]