Ysurac/openmptcprouter #896: Déconnexion VPS...

Slivikkus, 2020-03-14:
## Actual Behavior
Visiblement le VPS est assez capricieux en ce moment ...

Résultat grep ss 

## Specifications

  - OpenMPTCProuter version: (Last is not a version)
  - OpenMPTCProuter VPS version: (Last is not a version)
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64)

Slivikkus, 2020-03-14:
[image]
ça fonctionne toujours pas ça veux donc pas revenir, j'aurais toujours des pb avec ce VPS moi 🤔🤔

Ysurac, 2020-03-14:
Il faudrait vérifier la clef.

Slivikkus, 2020-03-14:
Bah pourtant ça fonctionnais bien hier soir et ça à couper pendant la nuit

Ysurac, 2020-03-14:
Tu peux vérifier le /var/log/daemon.log sur le VPS

Slivikkus, 2020-03-14:
Je l'ai envoyer justement et à 10h de ce que je vois la connexion à été reset

Slivikkus, 2020-03-14:
La connexion au VPS ne ce fait plus je pense que je vais devoir le reset à nouveau

Ysurac, 2020-03-14:
Visiblement l'API répond une fois sur deux. Après je vois beaucoup de soucis avec eth4...

Slivikkus, 2020-03-14:
eth4 est justement l'interface ou part la connexion pour tout les appareils chez moi

Ysurac, 2020-03-14:
Elle est en DHCP ?

Slivikkus, 2020-03-14:
Oui pour le coup, faut bien attribué les IP à mes machines :D

Slivikkus, 2020-03-14:
[image]

Slivikkus, 2020-03-14:
C'est une carte ethernet avec 4 ports rj :)

Ysurac, 2020-03-14:
Il y a ```192.168.100.129 e8:39:35:bc:55:84``` plusieurs fois dans le log qui semble du mal à obtenir une IP.

Slivikkus, 2020-03-14:
Aucun appareil sur cet ip étrangement ...

Slivikkus, 2020-03-14:
[image]
Visiblement des réinstallation du VPS change absolument rien

Ysurac, 2020-03-15:
Que donne un ```ip r``` et un ```ip a``` sur le routeur via SSH ?

Slivikkus, 2020-03-15:
[image]

Ysurac, 2020-03-15:
Et si tu désactives le VPN (en mettant sur "none" dans le wizard en cochant la case paramètres avancés), est-ce que shadowsocks monte ?

Slivikkus, 2020-03-15:
[image]

Ysurac, 2020-03-15:
que donne un ```curl -k https://<ipduserveur>:65500/``` depuis le routeur ?
Et un ```ping <ipduserveur>``` toujours depuis le routeur via SSH ?

Slivikkus, 2020-03-15:
[image]

Slivikkus, 2020-03-15:
[image]

Ysurac, 2020-03-15:
Il n'y a pas un firewall fournis par le fournisseur pouvant bloquer les ports sur le VPS ?
Il est chez qui ?

Slivikkus, 2020-03-15:
OVH, j'ai pas activé le pare feu.

Ysurac, 2020-03-15:
Faudrait essayer une réinstallation et une reconfiguration, car je vois pas d'où peut venir le soucis.

Slivikkus, 2020-03-15:
Même problème après réinstallation on va encore essayer de réinstaller le VPS pour voir

Ysurac, 2020-03-15:
Tu peux me recopier la configuration ? (Dans le menu Système->OpenMPTCProuter puis onglet tous les paramètres)
Je vais l'importer sur une de mes installations pour essayer de trouver le soucis.

Ysurac, 2020-03-15:
Hum, même mieux, tu peux faire une sauvegarde de la configuration dans Système->Sauvegarde/Mise à jour du micrologiciel puis "construire l'archive" et me l'envoyer par mail sur contact@openmptcprouter.com ? Je pourrais tester directement ainsi ?

Slivikkus, 2020-03-15:
C'est envoyer :)

Ysurac, 2020-03-15:
Bien reçu. je regarderais ça demain.

Slivikkus, 2020-03-15:
no problémo

Slivikkus, 2020-03-16:
Des nouvelles ? :)

Ysurac, 2020-03-16:
Ton serveur est toujours avec la même IP commençant par 54.38 ?
Si c'est le cas, l'API sur le VPS est effectivement non joignable. Donc le soucis serait du côté du VPS.

Slivikkus, 2020-03-16:
Oui l'ip n'a pas changée, j'ai réinstaller deux fois le VPS mais ça m'a toujours fait le même problème

Ysurac, 2020-03-16:
Sur le VPS : ```ps aux | grep omr```, ```ps aux | grep ss``` et il me faudrait tout le fichier /var/log/daemon.log (sur pastebin par exemple).
Il est sur Debian ?

Slivikkus, 2020-03-16:
Du coup quelle est la solution pour ça ?

Slivikkus, 2020-03-16:
VPS sous Debian 10

[image]
 

J'ai pas la possibilité de mettre le fichier daemon, j'ai aucun accès au SSH

Ysurac, 2020-03-16:
Comment ça tu as aucun accès au SSH ? Si tu veux dire mettre le fichier sur pastebin, tu peux utiliser ça : https://wiki.debian.org/fr/Pastebinit

Slivikkus, 2020-03-16:
[image]

Slivikkus, 2020-03-16:
https://paste.debian.net/1135177

Ysurac, 2020-03-16:
Ok, donc du coup ça semble un soucis de firewall.
Dans ```/etc/shorewall/params.net``` l'interface correspond bien à celle portant l'IP publique (visible via ```ip a```) ?

Slivikkus, 2020-03-16:
OUi

Ysurac, 2020-03-16:
il me faudrait le ```iptables-save``` sur le VPS.

Slivikkus, 2020-03-16:
Je l'envoi comment ?

Ysurac, 2020-03-16:
Pareil que le paste précédent

Slivikkus, 2020-03-16:
il peux pas l'envoyer

Ysurac, 2020-03-16:
```iptables-save | pastebinit``` ou un truc du genre

Slivikkus, 2020-03-16:
https://paste.debian.net/1135179

Ysurac, 2020-03-16:
```-A net_dnat -p tcp -m tcp --dport 49152:65534 -j DNAT --to-destination 10.255.255.2
-A net_dnat -p udp -m udp --dport 49152:65534 -j DNAT --to-destination 10.255.255.2```
C'est mauvais ça. Les portes au dessus de 65000 ne doivent pas être redirigé. Donc il y a un soucis.
Il me faudrait le fichier ```/etc/shorewall/rules```

Slivikkus, 2020-03-16:
Moi je touche rien du tout, je préfère ne rien toucher juste éxécuter le script et reboot le VPS (sans config IPv6 activée côté routeur et VPS)

https://paste.debian.net/1135180

Ysurac, 2020-03-16:
Il y aurait une régle sur le routeur redirigeant les ports 49152-65534 en TCP et UDP.
Si la régle n'existe plus sur le routeur, tu peux virer les 2 lignes avec DNAT.... et 49152 dans le fichier ```/etc/shorewall/rules``` (avec nano ou autre) puis faire un ```systemctl restart shorewall```.

Slivikkus, 2020-03-16:
Ah oui ça me dit quelque chose c'est en fait les ports passif en FTP pour le FTPS mais visiblement ça à tout fait merder ?

Slivikkus, 2020-03-16:
C'est quand même bizarre avant j'avais mis ce range de port mais y'avais aucun problème, quoi que quelques heures plus tard après avoir ouvert ce range de port je perdais la connexion en quelques sorte avec le VPS

Ysurac, 2020-03-16:
Tant que les connexions sont ouvertes ça doit passer, si elles doivent se renouveller c'est mort.
Les ports au dessus de 65000 sont réservés pour OMR.

Slivikkus, 2020-03-16:
Ok je modifierais ça sur mon serveur ce range de port et je met pas plus que 65000 c'est noté :)
Ah oui maintenant ça me paraît beaucoup plus logique que le SSH ne fonctionnais plus ! **Faudrait définir une limite de port côté routeur pour éviter ce genre de problème !**
Tout à l'air OK je clôture si jamais ça recommence dans les prochains jours je referais une issue comme d'habitude :D

Ysurac, 2020-03-16:
Ou plutot que j'ignore ce genre de truc niveau API du serveur ;)

Slivikkus, 2020-03-16:
why not :)

Slivikkus, 2020-03-16:
Merci énormément de toujours prendre ton temps pour moi ♥

closed 2020-03-16 by Slivikkus