Ysurac/openmptcprouter #897: Best DCHP V6 setting for LAN -advice -

vinceducat, 2020-03-14:
## Expected Behavior

what is the best dhcpv6 setting for LAN in case off the WAN1 is a FAI router that is supporting IPV6 and not disableable ...

[image]

  - OpenMPTCProuter version: 054.4
  - OpenMPTCProuter VPS version: 1006-test1
  - OpenMPTCProuter platform: RPI3

Ysurac, 2020-03-14:
It's not related to any IPv6 that the FAI can give.
What you want to do ?

vinceducat, 2020-03-14:
Could it be possible that : 
device on LAN receive ipv6 global adresse from FAI router (ipv6 /64 block) but use the agregrate link throught OMR ... 

perhaps it is not possible ....

Ysurac, 2020-03-15:
If a device receive IPv6 from ISP and use it as default gateway,  then the device will not use the aggregated link.
So IPv6 from any ISP must be disabled or IPv6 from OpenMPTCProuter must used as default gateway.

closed 2020-09-14 by github-actions[bot]