Ysurac/openmptcprouter #901: SS-Redir rapidly crashes with out of memory exception

AliveDevil, 2020-03-17:
## Expected Behavior

No kernel panic, pls.

## Actual Behavior

```
root@gateway:~# logread -f
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.447483] ss-redir invoked oom-killer: gfp_mask=0x6000c0(GFP_KERNEL), nodemask=(null), order=0, oom_score_adj=0
Tue Mar 17 11:13:22 2020 kern.info kernel: [  277.448210] ss-redir cpuset=/ mems_allowed=0
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.448861] CPU: 2 PID: 32480 Comm: ss-redir Tainted: G         C        4.19.108 #0
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.449336] Hardware name: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.12.1-0-ga5cab58e9a3f-prebuilt.qemu.org 04/01/2014
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.450273] Call Trace:
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.450758]  dump_stack+0x6d/0x9a
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.451196]  dump_header+0x65/0x211
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.451592]  oom_kill_process.cold.12+0xb/0x1e6
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.452020]  ? oom_badness.part.6+0xd7/0x130
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.452429]  out_of_memory+0x162/0x250
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.452861]  __alloc_pages_nodemask+0x9f3/0xd80
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.453273]  ? tcp_current_mss+0x60/0xb0
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.453684]  skb_page_frag_refill+0x9f/0xc0
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.454084]  sk_page_frag_refill+0x1a/0x80
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.454476]  tcp_sendmsg_locked+0x421/0xd60
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.454879]  ? tcp_recvmsg+0x1e6/0xbc0
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.455271]  tcp_sendmsg+0x27/0x40
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.455663]  inet_sendmsg+0x43/0xb0
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.456053]  sock_sendmsg+0x15/0x20
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.456445]  __sys_sendto+0x10f/0x190
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.456833]  ? do_epoll_wait+0xa5/0xd0
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.457221]  __x64_sys_sendto+0x23/0x30
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.457612]  do_syscall_64+0x57/0x260
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.458002]  ? do_page_fault+0x22/0x27
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.458395]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.458803] RIP: 0033:0x7f3bf6b592c1
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.459206] Code: c3 8b 07 85 c0 75 24 49 89 fb 48 89 f0 48 89 d7 48 89 ce 4c 89 c2 4d 89 ca 4c 8b 44 24 08 4c 8b 4c 24 10 4c 89 5c 24 08 0f 05 <c3> e9 2c d3 ff ff 41 54 b8 02 00 00 00 49 89 f4 be 00 08 08 00 55
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.460074] RSP: 002b:00007ffc63847ef8 EFLAGS: 00000246 ORIG_RAX: 000000000000002c
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.460526] RAX: ffffffffffffffda RBX: 00007f3bf6ae1d48 RCX: 00007f3bf6b592c1
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.460986] RDX: 0000000000003fff RSI: 00000000014745c0 RDI: 0000000000000022
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.461451] RBP: 000000000000002c R08: 0000000000000000 R09: 0000000000000000
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.461914] R10: 0000000000000000 R11: 0000000000000246 R12: 00000000013442a0
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.462381] R13: 0000000001344300 R14: 00007f3bf6b036e0 R15: 0000000000000000
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.463226] Mem-Info:
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.463712] active_anon:2759 inactive_anon:54 isolated_anon:0
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.463712]  active_file:65 inactive_file:57 isolated_file:0
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.463712]  unevictable:0 dirty:0 writeback:0 unstable:0
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.463712]  slab_reclaimable:808 slab_unreclaimable:6334
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.463712]  mapped:44 shmem:238 pagetables:163 bounce:0
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.463712]  free:4050 free_pcp:96 free_cma:0
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.466642] Node 0 active_anon:11036kB inactive_anon:216kB active_file:260kB inactive_file:176kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:176kB dirty:0kB writeback:0kB shmem:952kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.467713] DMA free:1228kB min:1092kB low:1364kB high:1636kB active_anon:0kB inactive_anon:0kB active_file:0kB inactive_file:0kB unevictable:0kB writepending:0kB present:15992kB managed:15908kB mlocked:0kB kernel_stack:0kB pagetables:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.469400] lowmem_reserve[]: 0 212 212 212
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.469961] DMA32 free:14972kB min:15288kB low:19108kB high:22928kB active_anon:11036kB inactive_anon:216kB active_file:620kB inactive_file:668kB unevictable:0kB writepending:0kB present:245612kB managed:222200kB mlocked:0kB kernel_stack:1872kB pagetables:652kB bounce:0kB free_pcp:328kB local_pcp:28kB free_cma:0kB
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.471684] lowmem_reserve[]: 0 0 0 0
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.472260] DMA: 1*4kB (U) 1*8kB (H) 4*16kB (UH) 1*32kB (U) 2*64kB (U) 0*128kB 0*256kB 0*512kB 1*1024kB (U) 0*2048kB 0*4096kB = 1260kB
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.473576] DMA32: 78*4kB (UME) 396*8kB (UMEH) 760*16kB (UMEH) 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 15640kB
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.474870] 320 total pagecache pages
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.475519] 65401 pages RAM
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.476137] 0 pages HighMem/MovableOnly
Tue Mar 17 11:13:22 2020 kern.warn kernel: [  277.476766] 5874 pages reserved
Tue Mar 17 11:13:22 2020 kern.info kernel: [  277.477383] Tasks state (memory values in pages):
Tue Mar 17 11:13:22 2020 kern.info kernel: [  277.478059] [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
Tue Mar 17 11:13:22 2020 kern.info kernel: [  277.478790] [   2031]     0  2031      228       30    40960        0             0 ubusd
Tue Mar 17 11:13:22 2020 kern.info kernel: [  277.479534] [   2097]     0  2097      180        7    36864        0             0 askfirst
Tue Mar 17 11:13:22 2020 kern.info kernel: [  277.480270] [   2121]     0  2121      194        9    32768        0             0 urngd
Tue Mar 17 11:13:22 2020 kern.info kernel: [  277.480952] [   4622]     0  4622      247       45    36864        0             0 logd
Tue Mar 17 11:13:22 2020 kern.info kernel: [  277.481623] [   4660]     0  4660      353       78    45056        0             0 rpcd
Tue Mar 17 11:13:22 2020 kern.info kernel: [  277.482322] [   4859]     0  4859      223        9    40960        0             0 dropbear
Tue Mar 17 11:13:22 2020 kern.info kernel: [  277.482998] [   5353]     0  5353      337       68    36864        0             0 netifd
Tue Mar 17 11:13:22 2020 kern.info kernel: [  277.483720] [   5443]     0  5443      217        9    36864        0             0 rngd
Tue Mar 17 11:13:22 2020 kern.info kernel: [  277.484452] [   5511]     0  5511      268       33    36864        0             0 odhcpd
[... 31 lines cut ...]
Tue Mar 17 11:13:22 2020 user.notice post-tracking-post-tracking: Set server vps (167.71.52.164) route via 192.168.2.1 metric 4
Tue Mar 17 11:13:22 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2...
```
ss-redir just keeps crashing with out of memory exceptions.

Eventually related to [https://github.com/shadowsocks/v2ray-plugin/issues/181]?

I saw that you had ZRam-Module in router firmware some time ago (in v0.52?), eventually related to this?

## Steps to Reproduce the Problem

  1. Two WAN (1 DSL/LTE Hybrid (by ISP), 2 LTE only)
  2. Download a Windows ISO through [https://uup.rg-adguard.net/].
  3. Observe crashing shadowsocks redirector.
  4. the ISO downloader downloads four files in parallel

## Specifications

  - OpenMPTCProuter version: 0.54.2
  - OpenMPTCProuter VPS version: 0.1014
  - OpenMPTCProuter platform: x86_64
    - Virtualized, Proxmox 6.1 Host
    - Memory: 512 MB
    - CPU Count: 4
    - Host CPU: i5-6500T

Ysurac, 2020-03-17:
Increase the memory ;)

AliveDevil, 2020-03-17:
That's a point (and I did that, to 4 GiB). Though I doubt that this should be _the solution_ as the (beta) supported Linksys WRT3200ACM / Linksys WRT32X both have 512 MB memory - and will eventually have issues here too.

What happened to the ZRam package in OpenMPTCPRouter then?

Ysurac, 2020-03-17:
It can be installed. In most case it's not needed but memory can increase with multiple VPS configured, when obfuscation is used or when some protocol are configured in omr-bypass.

closed 2020-09-14 by github-actions[bot]