Ysurac/openmptcprouter #902: Port Forwarding broken since 0.54.5 and 1015

Raggazzoo, 2020-03-18:
Hello,

It seems that the port forwarding is not working for me since the 0.54.5 router update and the 1015 vps update.
Everything worked fine when I was on the 0.54.2 version.
I tried to reboot everything and reapplied rules but it still won't forward to my local Synology router.

If it could help my vps is hosted by Scaleway.

Thank you for your time and your work on this project !

## Expected Behavior

Forward all ports to a router behind the OpenMPTCP router

## Actual Behavior

It won't since the 0.54.5 update

## Steps to Reproduce the Problem

  1. Configure a port forward
  2. Try to access from the outside to the vps ip and the forward port
  3. Get a connexion refused error

## Specifications

  - OpenMPTCProuter version: 0.54.5
  - OpenMPTCProuter VPS version: 1015
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-03-18:
What is the VPN used ?

Raggazzoo, 2020-03-18:
Glorytun UDP

Ysurac, 2020-03-18:
This will be fixed in next release that is compiling now.

Raggazzoo, 2020-03-18:
Perfect thank you!
I will update this thread when the upgrade is done on my side.

Raggazzoo, 2020-03-18:
I applied the update but the issue seems still here

Ysurac, 2020-03-18:
Working for me.
Check on the VPS, in /etc/shorewall/rules that the rule is here. Check also in /etc/shorewall/params.vpn that OMR_ADDR is 10.255.254.2 (local address of Glorytun UDP).
If all is ok, then it's another issue, SNAT can be needed.

robotsflo, 2020-03-18:
When i have update OMR to 0.54.6 and VPS to the last version 0.1015 (30 min ago), i lost my ports Forwardings.
i will restore OMR  backup to 0.53 and the ports forwarding doesn't work also. 
i will restor VPS backup to 0.1005 4.19.80  and the ports forwarding all works fine. 

I will wait the issue is closed to try again an update.

Ysurac, 2020-03-19:
Without more info, this doesn't help at all...

Raggazzoo, 2020-03-19:
Hello there,

I made a fresh install of the VPS and the router and now everything is fine =)
Something was wrong with my VPS because I tried to flash my router a couple of times before reset my VPS and it did not work.

Thank you @Ysurac

closed 2020-03-19 by Raggazzoo