Ysurac/openmptcprouter #903: Losing connection on Hyper-V

ccmks, 2020-03-18:
## Expected Behavior

Stable connection running on Hyper-V and expect to work normally

## Actual Behavior

After setup OpenMPTCPRouter and get everything setup, the MPTCP is working fine across 3 different WAN. However, after 5 minutes or so, random WAN (either WAN1, WAN2 or WAN3) losing connection. The status page will show "Gateway Down" on one or 2 WAN connection and it happen randomly on which WAN.

Trying to restart the interface but the router WAN which gateway is down can't ping to its respective gateway. The only way to fix it is to reboot the router again, but still having same problem within 5 minutes after rebooting.

## Steps to Reproduce the Problem

  1. Download the Hyper-V Image EXT4 or Squash, both regular or EFI
  2. Setup all WAN to be on DHCP
  3. Test connectivity and test ping

## Specifications

  - OpenMPTCProuter version: v0.54.5 r0+12509-04a21c26a0
  - OpenMPTCProuter VPS version: Version 0.1014 4.19.104-mptcp
  - OpenMPTCProuter platform: Windows Server 2016 Datacenter Running Hyper-V with Intel PRO 1000 PT Quad NIC for WAN and Realtek onboard NIC for LAN connection

Ysurac, 2020-03-18:
I never used Hyper-V. But you can check in State->System log why wan are losing connections.

ccmks, 2020-03-18:
Here is the log

Wed Mar 18 14:36:37 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Mar 18 14:36:37 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI3, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI3
Wed Mar 18 14:36:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI3, subsystem=tty
Wed Mar 18 14:36:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Mar 18 14:36:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI4, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI4
Wed Mar 18 14:36:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI4, subsystem=tty
Wed Mar 18 14:36:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Mar 18 14:36:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI5, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI5
Wed Mar 18 14:36:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI5, subsystem=tty
Wed Mar 18 14:36:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Mar 18 14:36:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI6, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI6
Wed Mar 18 14:36:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI6, subsystem=tty
Wed Mar 18 14:36:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Mar 18 14:36:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI7, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI7
Wed Mar 18 14:36:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI7, subsystem=tty
Wed Mar 18 14:36:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Mar 18 14:36:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI8, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI8
Wed Mar 18 14:36:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI8, subsystem=tty
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI9, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI9
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI9, subsystem=tty
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI10, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI10
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI10, subsystem=tty
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI11, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI11
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI11, subsystem=tty
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI12, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI12
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI12, subsystem=tty
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI13, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI13
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI13, subsystem=tty
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI14, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI14
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI14, subsystem=tty
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI15, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI15
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI15, subsystem=tty
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI16, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI16
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI16, subsystem=tty
Wed Mar 18 14:36:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
[... cut ...]
Wed Mar 18 14:48:08 2020 user.notice post-tracking-post-tracking: Set server vps (45.32.101.41) default route via 10.255.250.5
Wed Mar 18 14:48:08 2020 user.notice post-tracking-post-tracking: Set server vps (45.32.101.41) route via 10.255.250.5 metric 9
Wed Mar 18 14:48:08 2020 user.notice post-tracking-post-tracking: Get status and settings for ovpnwan2...
Wed Mar 18 14:48:09 2020 user.notice post-tracking-post-tracking: Get status and settings for ovpnwan2... Done
Wed Mar 18 14:54:46 2020 daemon.err uhttpd[6519]: luci: accepted login on /admin/system/openmptcprouter/status for root from 192.168.100.205

Ysurac, 2020-03-18:
You are using MPTCP over VPN ? why ?

ccmks, 2020-03-18:
My WAN 2 connection doesn't support MPTCP so have to enable over VPN

Ysurac, 2020-03-18:
Without this interface, it's working ?

ccmks, 2020-03-18:
Same issue too. The random WAN will show gateway down.

Ysurac, 2020-03-18:
On the log I only tun4 interface, that is the MPTCP over VPN interface as down.
Maybe some problems with the DHCP ? Each wan have own real interface and each real interface have a router directly connected ?

ccmks, 2020-03-18:
I have been running it for 35 minutes when I generated the log for you. Again, the random WAN disconnection happens. 

I don't think it is DHCP issue, because I have setup static IP for the WAN whose GW is down and still can't ping to its gateway.

Answer to last question:
It is Hyper-V so all interface is from Hyper-V virtual interface. But each interface is connected to different vSwitch, and each vSwitch is connected to each physical NIC in the Intel PRO 1000 PT Quad NIC. Each NIC in Physical port is connected to each different modem

Ysurac, 2020-03-18:
Can you give me, when there is a disconnection, the result of ```ip a``` and ```ip r``` via SSH on the router ?

ccmks, 2020-03-19:
I could no longer be able to duplicate the issue.

Apparently, when I read the system log, I realized that I gave too low memory which was only 384 MB, after increase the memory to 512 MB, the issue is no longer there. I will go ahead and close this issue.

ccmks, 2020-03-19:
Well it happened again!

Here is the output for ip a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 qdisc noqueue state UNKNOWN group                                                                                         default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP,80000> mtu 1500 qdisc mq state UP grou                                                                                        p default qlen 1000
    link/ether 00:15:5d:05:f7:4a brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global eth0
       valid_lft forever preferred_lft forever
3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1244 qdisc mq state UP group defa                                                                                        ult qlen 100
    link/ether 00:15:5d:05:f7:4b brd ff:ff:ff:ff:ff:ff
    inet 192.168.80.24/24 brd 192.168.80.255 scope global eth1
       valid_lft forever preferred_lft forever
4: eth2: <BROADCAST,MULTICAST,UP,LOWER_UP,80000> mtu 1500 qdisc mq state UP grou                                                                                        p default qlen 50
    link/ether 00:15:5d:05:f7:4c brd ff:ff:ff:ff:ff:ff
    inet 192.168.40.102/24 brd 192.168.40.255 scope global eth2
       valid_lft forever preferred_lft forever
5: eth3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group defa                                                                                        ult qlen 100
    link/ether 00:15:5d:05:f7:4d brd ff:ff:ff:ff:ff:ff
    inet 192.168.8.102/24 brd 192.168.8.255 scope global eth3
       valid_lft forever preferred_lft forever
6: eth4: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group defa                                                                                        ult qlen 1000
    link/ether 00:15:5d:05:f7:4e brd ff:ff:ff:ff:ff:ff
7: sit0@NONE: <NOARP,80000> mtu 1480 qdisc noop state DOWN group default qlen 10                                                                                        00
    link/sit 0.0.0.0 brd 0.0.0.0
8: ifb0: <BROADCAST,NOARP,80000> mtu 1500 qdisc noop state DOWN group default ql                                                                                        en 32
    link/ether 5e:7b:bf:be:e9:12 brd ff:ff:ff:ff:ff:ff
9: ifb1: <BROADCAST,NOARP,80000> mtu 1500 qdisc noop state DOWN group default ql                                                                                        en 32
    link/ether 0e:d0:d2:1a:d1:86 brd ff:ff:ff:ff:ff:ff
10: teql0: <NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void
16: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq_code                                                                                        l state UNKNOWN group default qlen 1000
    link/none
    inet 10.255.255.2 peer 10.255.255.1/32 scope global tun0
       valid_lft forever preferred_lft forever
19: ifb4tun0: <BROADCAST,NOARP,80000> mtu 1500 qdisc noop state DOWN group defau                                                                                        lt qlen 32
    link/ether ce:8e:6e:64:61:68 brd ff:ff:ff:ff:ff:ff

Here is the output for ip r
default via 10.255.255.1 dev tun0
default via 192.168.40.1 dev eth2 metric 4
default via 192.168.8.1 dev eth3 metric 5
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2
45.32.101.41 via 192.168.8.1 dev eth3 metric 5
127.0.0.0/8 dev lo proto static scope link metric 1
192.168.8.0/24 dev eth3 scope link metric 5
192.168.40.0/24 dev eth2 scope link metric 4
192.168.80.0/24 dev eth1 scope link metric 3

Ysurac, 2020-03-19:
512 is still low...
eth4 have no IP, normal ?
Else there is no error and gateway should be pingable.

ccmks, 2020-03-19:
We don't have anything on eth4. Only 3 WAN plugged in

The WAN 1 IP GW is not pingable although I already set to DHCP.

If 512 RAM is still low, what would be the minimum?

Ysurac, 2020-03-19:
1024 is better

ccmks, 2020-03-21:
Already tried 1024 MB, still experiencing random WAN disconnection. I tried plug in 4 WAN, and sometimes, there are 2 WAN experience random disconnection. It happened on random WAN

Ysurac, 2020-03-21:
What is the latency of each interfaces ? (in OpenMPTCProuter status page)

ccmks, 2020-03-21:
[image]

Ysurac, 2020-03-21:
While it's not working, remove MPTCP over VPN, this doesn't help to find the issue.

ccmks, 2020-03-21:
I am still currently testing on various parameter. My configuration works on Banana Pi R2, however, I decided to use Hyper-V since my router is also hosted there, no need additional hardware plug in. On the top of that, the Banana Pi R2 only works with v0.53, doesn't work higher version.

I have been putting this Hyper-V on different condition, such as adjusting the core and the memory. Right now, the most stable is using 32 bit version of image and using Hyper-V with virtual memory range between 512 - 768 MB, more than that, it could trigger the Gateway Down and not stable too.

However, I still want to see if this could last for several days without having random disconnection on WAN.

Before I was only using single core on virtual processor, however, the throughput was limited, so I ended up using all core of my processor to reach the maximum speed.

ccmks, 2020-03-23:
I gave up on doing this, ended up migrating my router to different hardware, however, this new hardware, I only have Hyper-V core 2019. The OpenMPTCPRouter is working fine on the new hardware under Hyper-V hosting. I guess, I have to find what's wrong with the old hardware.

closed 2020-03-23 by ccmks