Ysurac/openmptcprouter #904: VPS access

MacGeyer, 2020-03-18:
## Expected Behavior

Access specific port on VPS.

## Actual Behavior

Access to port 4949 is not possible

## Steps to Reproduce the Problem

  1. Install Munin Node
  2. Configure Munin Node to accept connections for serveur IP
  3. Edit /etc/shorewall/rules to add : ACCEPT          net             $FW             tcp     4949
  4. Restart Shorewall

## Specifications

  - OpenMPTCProuter version: 0.51.5
  - OpenMPTCProuter VPS version: 0.999
  - OpenMPTCProuter platform: ESXi : x86_64

Ysurac, 2020-03-18:
If "Redirect all ports" is not activated, this should work.

MacGeyer, 2020-03-18:
YES, uncheck "Redirects all ports from server to this router" did it

closed 2020-03-18 by MacGeyer