Ysurac/openmptcprouter #907: Not show usb modem tape. 4G, 3G etc. Shutdown not available

sench4707, 2020-03-20:
## Expected Behavior

Update to latest version v0.54.6.
Again not show usb modem network tape.
when using usb modem need to know what network tape is use modem for connection.
In screenshot you can see that only signal level is shown.

Shutdown also not available via web. 

## Specifications

  - OpenMPTCProuter version: v0.54.6
  - OpenMPTCProuter platform: RPI4

[image]

Ysurac, 2020-03-20:
All informations are not always available without authentication for each protocol/API. What is the protocol used for your modem and what is the manufacturer ?

sench4707, 2020-03-20:
Modem manufacturer Huawei. Model E3372s. Mode HiLink.  DHCP.

Olli69, 2020-03-20:
One vote for a shutdown function in the UI. I have RPi 4 in a streaming backpack and if you just pull off the power cable from a power bank, the memory card gets easily corrupted.

Ysurac, 2020-03-20:
I will try to add both network info for Huawei and shutdown button in next release.

rubaz100, 2020-03-20:
Can you add support for the Huawei B535 (German: Vodafone Gigacube) too?

Thank you! <3

Ysurac, 2020-03-21:
This will be for most available Huawei dongle

gnosoz, 2020-03-21:
I am using 2 x TPLINK 4G routers as follows connected via LAN port.
 -  Archer MR200
 -  Archer MR600
For these 2 connections the router is unable to discover they are 3/4G router therefore threats them as DSL connections while a vodafone USB dongle is correctly identified as LTE.

Ysurac, 2020-03-21:
TPLink routers should have an API too, but I don't own any TPLink router so I can't test and add the support for that.

gnosoz, 2020-03-21:
I can send one to you together with an APU4C board which is what I am using

Ysurac, 2020-03-21:
Thanks, my address is here: https://www.openmptcprouter.com/about
But for now I think it's difficult: I have no idea if I will still receive parcel next week... We are confined in France due to Covid-19.

gnosoz, 2020-03-21:
same here and hope you and your family are in good health.
The TP-Link will arrive directly from amazon FR to the address advised whilst the (fully working openmptcprouter) apu board will be shipped from our company once situation returns to normal.

Ysurac, 2020-03-21:
I live in countryside so not too much problems for now. I hope it's ok for you too ?
ok thanks.