Ysurac/openmptcprouter #908: Uninstall vps proxy

3l0w, 2020-03-20:
##  Actual Behavior
I install openmprcrouter on my vps to try this with my 4g box and my phone that work pretty well but i have service with ports open on my vps and after installing i can't connect to anything.
Now i want to remove openmprcrouter from my vps but i dont find how to remove.

## Specifications
  - OpenMPTCProuter VPS version: (0.1014)
  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2020-03-20:
You can't remove it. But you can open ports you need, it's in ```/etc/shorewall/rules``` (you need to restart shorewall after changes).

3l0w, 2020-03-20:
Thanks for you response i modify the config and everything work again. Thanks

closed 2020-03-20 by 3l0w