Ysurac/openmptcprouter #910: Router auto update

gnosoz, 2020-03-21:
## Expected Behavior

Would be great to have an auto update script that pulls only the affected files whilst maintaining the general configuration.

## Actual Behavior

Needs router to be reinstalled from zero and re configured

## Steps to Reproduce the Problem

At the moment as I am unable to see the direct changes I am going to create one that does as follows:
A) 2 partitions on same HD 1.9 Gb + 1.9 Gb
B) pull image grom github and install on 2nd partition
C) copy relevant conf files to 2nd partition
D) modify grub to start from 2nd partition
E) start fully working 2nd partition
E) dd 2nd partition to first + modify grub to start from 1st partition
F) delete all data from 2nd partition

It is definitely not the best procedure but this is all I can do at the moment to keep up with your quick deployment of releases :)

## Specifications

Ysurac, 2020-03-21:
You can already update via the web interface of the router and keep the configuration in System->Backup / Flash Firmware menu. I don't know if it's what you mean ?

xabix, 2020-03-25:
why not offering an update procedure like based on opkg so it's even easier and may not need to reboot

closed 2020-03-25 by gnosoz